freebsd/drm-kmod #115: GNOME extremely laggy (basically unusable) on amdgpu unless AccelMethod is None

neelchauhan, 2021-09-22:
**Describe the bug**

Certain apps like GNOME and Wine extremely laggy (basically unusable) on amdgpu unless AccelMethod is None in `xorg.conf`.

**FreeBSD version**

    FreeBSD envy.neelc.org 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n249422-e51aabf8cbb: Thu Sep 16 07:54:14 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

    hostb0@pci0:0:0:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir/Cezanne Root Complex'
        class      = bridge
        subclass   = HOST-PCI
    none0@pci0:0:0:2:    class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir/Cezanne IOMMU'
        class      = base peripheral
        subclass   = IOMMU
    hostb1@pci0:0:1:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib1@pci0:0:1:3:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir/Cezanne PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    hostb2@pci0:0:2:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib2@pci0:0:2:1:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir/Cezanne PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    pcib3@pci0:0:2:3:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir/Cezanne PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    hostb3@pci0:0:8:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib4@pci0:0:8:1:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Internal PCIe GPP Bridge to Bus'
        class      = bridge
        subclass   = PCI-PCI
    intsmb0@pci0:0:20:0:    class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'FCH SMBus Controller'
        class      = serial bus
        subclass   = SMBus
    isab0@pci0:0:20:3:    class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'FCH LPC Bridge'
        class      = bridge
        subclass   = PCI-ISA
    hostb4@pci0:0:24:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1448 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Device 24: Function 0'
        class      = bridge
        subclass   = HOST-PCI
[... 79 lines cut ...]
        subclass   = HDA
    none4@pci0:4:0:7:    class=0x118000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e4 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Raven/Raven2/Renoir Sensor Fusion Hub'
        class      = dasp

**DRM KMOD version**

    drm-devel-kmod 5.5.19.g20210909

This also happens on drm-kmod 5.4 and 5.6-wip, on both HP and Gateway laptops.

**To Reproduce**

Take an amdgpu-based PC, and install GNOME. You need to use gnome-shell in .xinitrc or .xsession as gdm isn't working correctly as of now.

You will see extreme lag where GNOME is unusable. Other desktops like KDE5/KF5 and XFCE4 work fine.

**Screenshots**

N/A

**Additional context**

I am using a HP Envy 13-ay0021nr. This also happened on a Gateway laptop using amdgpu (forgot which model)

evadot, 2021-09-23:
Is that when using xf86-video-amdgpu ?
Could you test mesa-devel too, that could be mesa-related.

evadot, 2021-09-23:
This could also be something like https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=251836 where we don't support some stuff and mesa fallback to llvmpipe.

evadot, 2022-05-20:
Is it still happening with latest drm-kmod ?
Also you could test https://reviews.freebsd.org/D35269

grahamperrin, 2022-08-06:
Also, FreeBSD bug [257810 – x11/gnome3(-lite) & x11/gdm lags significantly on amdgpu](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=257810) (possible duplicate of 253746).

evadot, 2024-04-17:
Closing as no news from OP.

closed 2024-04-17 by evadot