freebsd/drm-kmod #116: Panic in dma-buf bits

kevans91, 2021-10-05:
A user on IRC reported the following panic while casually browsing the internet:

```
[14424] panic: mi_switch: switch in a critical section
[14424] cpuid = 0
[14424] time = 1632782330
[14424] __HardenedBSD_version = 1300061 __FreeBSD_version = 1300515
[14424] version = FreeBSD 13.0-STABLE-HBSD #203  hardened/13-stable/master-n190565-9ec32da6572: Sun Sep 26 12:13:09 EDT 2021
[14424]     root@ci-01.md.hardenedbsd.org:/usr/obj/src/13-stable/amd64.amd64/sys/HARDENEDBSD
[14424] KDB: stack backtrace:
[14424] #0 0xffffffff80c6ea0b at kdb_backtrace+0x6b
[14424] #1 0xffffffff80c24bcc at vpanic+0x18c
[14424] #2 0xffffffff80c24993 at panic+0x43
[14424] #3 0xffffffff80c315ea at mi_switch+0x19a
[14424] #4 0xffffffff80c0260c at __mtx_lock_sleep+0x1cc
[14424] #5 0xffffffff80c023e8 at __mtx_lock_flags+0xa8
[14424] #6 0xffffffff82cf0639 at dma_fence_signal+0x29
[14424] #7 0xffffffff82cf1761 at dma_resv_add_shared_fence+0xb1
[14424] #8 0xffffffff82cffa09 at ttm_eu_fence_buffer_objects+0x79
[14424] #9 0xffffffff82bb3d82 at radeon_cs_parser_fini+0x52
[14424] #10 0xffffffff82bb3934 at radeon_cs_ioctl+0x914
[14424] #11 0xffffffff82c961ee at drm_ioctl_kernel+0x7e
[14424] #12 0xffffffff82c9661c at drm_ioctl+0x37c
[14424] #13 0xffffffff80e6c71d at linux_file_ioctl+0x33d
[14424] #14 0xffffffff80c90a34 at kern_ioctl+0x204
[14424] #15 0xffffffff80c90784 at sys_ioctl+0x124
[14424] #16 0xffffffff810cf83a at amd64_syscall+0x13a
[14424] #17 0xffffffff810a0f0b at fast_syscall_common+0xf8
```

The relevant parts really just being frame 7 and up in the above trace. See:

https://github.com/freebsd/drm-kmod/blob/f0cca94f3ff7a2b7301b3638cc19c0c7971512e4/drivers/dma-buf/dma-resv.c#L237 -> enter a critical section, call dma_fence_is_signaled() 8 lines down
https://github.com/freebsd/drm-kmod/blob/f0cca94f3ff7a2b7301b3638cc19c0c7971512e4/linuxkpi/gplv2/include/linux/dma-fence.h#L422 -> dma_fence_is_signaled() calls dma_fence_signal(), still in a critical section
https://github.com/freebsd/drm-kmod/blob/f0cca94f3ff7a2b7301b3638cc19c0c7971512e4/drivers/dma-buf/dma-fence.c#L188 -> dma_fence_signal() tries to grab a spinlock (MTX_DEF mutex)

kib notes that the critical section is likely just to satisfy the requirements for rcu_deref_protected(), which may not be as heavy here; I won't have time to look at this any further for a while.

wulf7, 2021-10-10:
I hope 90ab37d244bb fixes it

grahamperrin, 2022-01-13:
https://github.com/freebsd/drm-kmod/issues/116#issue-1015861677

> `[14424] panic: mi_switch: switch in a critical section`

Is this related?

FreeBSD bug [261166 – graphics/drm-devel-kmod kernel panics: mi_switch: switch in a critical section](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=261166)

grahamperrin, 2022-01-19:
A commit references this bug:

https://github.com/freebsd/freebsd-src/commit/02ea6033020e11afec6472bf560b0ddebd0fa97a

<https://cgit.freebsd.org/src/commit/?id=02ea6033020e11afec6472bf560b0ddebd0fa97a>

```text
author    Vladimir Kondratyev <wulf@𡀦…>	2022-01-18 20:14:12 +0000
committer Vladimir Kondratyev <wulf@…>	2022-01-18 20:14:12 +0000
commit    02ea6033020e11afec6472bf560b0ddebd0fa97a 
```
LinuxKPI: Allow spin_lock_irqsave to be called within a critical section with spinning on spin_trylock. dma-buf part of drm-kmod depends on this property and absence of it support results in "mi_switch: switch in a critical section" assertions [1][2].

[1] https://github.com/freebsd/drm-kmod/issues/116
[2] https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=261166

MFC after:	1 week
Reviewed by:	manu
Differential Revision:	https://reviews.freebsd.org/D33887

grahamperrin, 2022-01-19:
Also MFC after one week, with reference to the same FreeBSD bug 261166: 

https://github.com/freebsd/freebsd-src/commit/89889ab470b9f1a1cd36913dd219b78efbf484df

grahamperrin, 2022-01-19:
@kevans91 @wulf7 

As fixes for this issue are in https://github.com/freebsd/freebsd-src/ – and, I guess, <https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=261166> will be _closed fixed_ after merges from `main` to `stable/13`, should we: 

* close this issue now; or 
* close after the merges?

evadot, 2022-01-27:
That should be fixed, if it's not please open a new issue

closed 2022-01-27 by evadot