freebsd/drm-kmod #117: radeonkms loading results in black screen with RS780L

mibifuzzi, 2021-10-05:
**Describe the bug**
I cannot load radeokms on my machine. Whenever I load it, the screen turns black. I tried loading it via kldlist in `/etc/rc.conf` or via `kldload`. I tried the binary package from pkg and building the module by hand via the ports collection. I also tried with `hw.syscons.disable=1` or without it in `/boot/loader.conf`.

**FreeBSD version**
FreeBSD marc 13.0-RELEASE-p4 FreeBSD 13.0-RELEASE-p4 #0: Tue Aug 24 07:33:27 UTC 2021     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>

  hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x9600 subvendor=0x1043 subdevice=0x8388
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'RS780 Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1043 device=0x9602 subvendor=0x1043 subdevice=0x8388
    vendor     = 'ASUSTeK Computer Inc.'
    device     = 'AMD RS780/RS880 PCI to PCI bridge (int gfx)'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:7:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x9607 subvendor=0x1043 subdevice=0x8388
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'RS780/RS880 PCI to PCI bridge (PCIE port 3)'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:10:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x9609 subvendor=0x1043 subdevice=0x8388
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'RS780/RS880 PCI to PCI bridge (PCIE port 5)'
    class      = bridge
    subclass   = PCI-PCI
ahci0@pci0:0:17:0:	class=0x010601 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4391 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0/SB8x0/SB9x0 SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
ohci0@pci0:0:18:0:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4397 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0/SB8x0/SB9x0 USB OHCI0 Controller'
    class      = serial bus
    subclass   = USB
ohci1@pci0:0:18:1:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4398 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0 USB OHCI1 Controller'
    class      = serial bus
    subclass   = USB
ehci0@pci0:0:18:2:	class=0x0c0320 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4396 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0/SB8x0/SB9x0 USB EHCI Controller'
    class      = serial bus
    subclass   = USB
ohci2@pci0:0:19:0:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4397 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0/SB8x0/SB9x0 USB OHCI0 Controller'
    class      = serial bus
    subclass   = USB
ohci3@pci0:0:19:1:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4398 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0 USB OHCI1 Controller'
    class      = serial bus
    subclass   = USB
ehci1@pci0:0:19:2:	class=0x0c0320 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4396 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0/SB8x0/SB9x0 USB EHCI Controller'
    class      = serial bus
    subclass   = USB
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x3c hdr=0x00 vendor=0x1002 device=0x4385 subvendor=0x1043 subdevice=0x8389
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SBx00 SMBus Controller'
    class      = serial bus
[... 121 lines cut ...]
  Oct  5 16:41:20 marc kernel: [drm] radeon: irq initialized.
  Oct  5 16:41:20 marc kernel: [drm] ring test on 0 succeeded in 1 usecs
  Oct  5 16:41:20 marc kernel: [drm] ib test on ring 0 succeeded in 0 usecs
  Oct  5 16:41:20 marc kernel: [drm] Connector VGA-1: get mode from tunables:
  Oct  5 16:41:20 marc kernel: [drm]   - kern.vt.fb.modes.VGA-1
  Oct  5 16:41:20 marc kernel: [drm]   - kern.vt.fb.default_mode
  Oct  5 16:41:20 marc kernel: [drm] Connector HDMI-A-1: get mode from tunables:
  Oct  5 16:41:20 marc kernel: [drm]   - kern.vt.fb.modes.HDMI-A-1
  Oct  5 16:41:20 marc kernel: [drm]   - kern.vt.fb.default_mode
  Oct  5 16:41:20 marc kernel: [drm] Radeon Display Connectors
  Oct  5 16:41:20 marc kernel: [drm] Connector 0:
  Oct  5 16:41:20 marc kernel: [drm]   VGA-1
  Oct  5 16:41:20 marc kernel: [drm]   DDC: 0x7e40 0x7e40 0x7e44 0x7e44 0x7e48 0x7e48 0x7e4c 0x7e4c
  Oct  5 16:41:20 marc kernel: [drm]   Encoders:
  Oct  5 16:41:20 marc kernel: [drm]     CRT1: INTERNAL_KLDSCP_DAC1
  Oct  5 16:41:20 marc kernel: [drm] Connector 1:
  Oct  5 16:41:20 marc kernel: [drm]   HDMI-A-1
  Oct  5 16:41:20 marc kernel: [drm]   HPD3
  Oct  5 16:41:20 marc kernel: [drm]   DDC: 0x7e50 0x7e50 0x7e54 0x7e54 0x7e58 0x7e58 0x7e5c 0x7e5c
  Oct  5 16:41:20 marc kernel: [drm]   Encoders:
  Oct  5 16:41:20 marc kernel: [drm]     DFP3: INTERNAL_KLDSCP_LVTMA
  Oct  5 16:41:20 marc kernel: [drm] Cannot find any crtc or sizes
  Oct  5 16:41:20 marc kernel: [drm] Initialized radeon 2.50.0 20080528 for drmn0 on minor 0
  Oct  5 16:41:21 marc kernel: [drm] Cannot find any crtc or sizes
</details>

evadot, 2021-10-06:
Does the workaround in #108 works ?

grahamperrin, 2021-10-09:
> `vgapci0@pci0:1:5:0: class=0x030000 rev=0x00 hdr=0x00 vendor=0x1002 device=0x9616 subvendor=0x1043`
> `subdevice=0x8388`
> `vendor = 'Advanced Micro Devices, Inc. [AMD/ATI]'`
> `device = 'RS780L [Radeon 3000]'`
> `class = display`
> `subclass = VGA`

For what it's worth, with quite different Radeon hardware ("Sumo [Radeon 6550d]"): 

"… the absolute path does work, contrary to …"

– <https://forums.FreeBSD.org/threads/81015/post-535736>

grahamperrin, 2021-10-09:
https://github.com/freebsd/drm-kmod/issues/117#issuecomment-935889491

> … after drm was loaded, I tried two things: starting X and loading radeonkms in addition. Both ways, the blackout happens again.

Does _temporary_ installation of [x11-drivers/xf86-video-ati](https://www.freshports.org/x11-drivers/xf86-video-ati/) make a difference in this case?

The driver may, may not be recommended for your hardware but still, I'd like to know the outcome. Thanks.

mibifuzzi, 2021-10-12:
Thanks for staying with me.

> As root, or as a normal user?

After loading `drm` and not `radeonkms`, without having installed `xf86-video-ati`, I get "no screens found" when starting X as both, either root or an unprivileged user.

Installing `xf86-video-ati` and starting X gives me the blank screen for both users, still without `radeonkms` loaded. But now, `kldstat` gives:

<details>
<summary>kldstat</summary>

```
  Id Refs Address                Size Name
  1   96 0xffffffff80200000  1f11f28 kernel
  2    1 0xffffffff82113000    1abe8 geom_eli.ko
  3    1 0xffffffff8212e000     b7b8 opensolaris.ko
  4    1 0xffffffff8213a000   67feb0 zfs.ko
  5    3 0xffffffff82920000    7f548 drm.ko
  6    3 0xffffffff829a0000     cbc8 linuxkpi_gplv2.ko
  7    1 0xffffffff829ad000     2328 lindebugfs.ko
  8    1 0xffffffff829b0000     3378 acpi_wmi.ko
  9    1 0xffffffff829b4000     3218 intpm.ko
  10    1 0xffffffff829b8000     2180 smbus.ko
  11    1 0xffffffff829bb000     4b60 ng_ubt.ko
  12    6 0xffffffff829c0000     aac8 netgraph.ko
  13    2 0xffffffff829cb000     a238 ng_hci.ko
  14    3 0xffffffff829d6000     25a8 ng_bluetooth.ko
  15    1 0xffffffff829d9000     2340 uhid.ko
  16    1 0xffffffff829dc000     4350 ums.ko
  17    1 0xffffffff829e1000     3380 usbhid.ko
  18    1 0xffffffff829e5000     31f8 hidbus.ko
  19    1 0xffffffff829e9000     3320 wmt.ko
  20    1 0xffffffff829ed000     e250 ng_l2cap.ko
  21    1 0xffffffff829fc000    1ce48 ng_btsocket.ko
  22    1 0xffffffff82a19000     3980 ng_socket.ko
  23    1 0xffffffff82a1d000     2a08 mac_ntpd.ko
  24    1 0xffffffff82a20000   14fc70 radeonkms.ko
  25    1 0xffffffff82b70000     e778 ttm.ko
  26    1 0xffffffff82b7f000     29d8 radeon_RS780_pfp_bin.ko
  27    1 0xffffffff82b82000     74d8 radeon_RS780_me_bin.ko
  28    1 0xffffffff82b8a000     2cd8 radeon_R600_rlc_bin.ko
```

</details>