freebsd/drm-kmod #124: Backlight on Intel TigerLake

neelchauhan, 2021-10-16:
**Describe the bug**

When I attempt to use the backlight on a HP Spectre x360 14-ea0023dx, it does not work. It stays at 90% brightness.

I tried `acpi_video` and the `backlight` command to no avail.

**NOTE:** This isn't related to my earlier TigerLake issues. Weird that I went in a full circle, sold a TigerLake laptop, bought an AMD Ryzen/Renoir laptop, and now went back to a TigerLake laptop.

**FreeBSD version**

    FreeBSD concorde.local 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n250051-04c91ac48ad: Thu Oct 14 08:03:39 UTC 2021     root@localhost.localdomain:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

    hostb0@pci0:0:0:0:    class=0x060000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a14 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = '11th Gen Core Processor Host Bridge/DRAM Registers'
        class      = bridge
        subclass   = HOST-PCI
    vgapci0@pci0:0:2:0:    class=0x030000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a49 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'TigerLake-LP GT2 [Iris Xe Graphics]'
        class      = display
        subclass   = VGA
    none0@pci0:0:4:0:    class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a03 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'TigerLake-LP Dynamic Tuning Processor Participant'
        class      = dasp
    pcib1@pci0:0:7:0:    class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a23 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    pcib2@pci0:0:7:1:    class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a25 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    none1@pci0:0:8:0:    class=0x088000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a11 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'GNA Scoring Accelerator module'
        class      = base peripheral
    xhci0@pci0:0:13:0:    class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a13 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 USB Controller'
        class      = serial bus
        subclass   = USB
    none2@pci0:0:13:2:    class=0x0c0340 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a1b subvendor=0x2222 subdevice=0x1111
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 NHI'
        class      = serial bus
        subclass   = USB
    pcib3@pci0:0:14:0:    class=0x010400 rev=0x00 hdr=0x00 vendor=0x8086 device=0x9a0b subvendor=0x8086 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = 'Volume Management Device NVMe RAID Controller'
        class      = mass storage
        subclass   = RAID
    none3@pci0:0:18:0:    class=0x070000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0fc subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Integrated Sensor Hub'
        class      = simple comms
        subclass   = UART
    xhci1@pci0:0:20:0:    class=0x0c0330 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0ed subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller'
        class      = serial bus
        subclass   = USB
    none4@pci0:0:20:2:    class=0x050000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0ef subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Shared SRAM'
        class      = memory
        subclass   = RAM
    none5@pci0:0:20:3:    class=0x028000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0f0 subvendor=0x8086 subdevice=0x0074
        vendor     = 'Intel Corporation'
        device     = 'Wi-Fi 6 AX201'
        class      = network
    ig4iic0@pci0:0:21:0:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e8 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Serial IO I2C Controller'
        class      = serial bus
    ig4iic1@pci0:0:21:1:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e9 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Serial IO I2C Controller'
        class      = serial bus
    none6@pci0:0:22:0:    class=0x078000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e0 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Management Engine Interface'
        class      = simple comms
    pcib6@pci0:0:28:0:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0bd subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tigerlake PCH-LP PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    none7@pci0:0:29:0:    class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x09ab subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        class      = base peripheral
    isab0@pci0:0:31:0:    class=0x060100 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa082 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP LPC Controller'
        class      = bridge
        subclass   = PCI-ISA
    hdac0@pci0:0:31:3:    class=0x040100 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0c8 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Smart Sound Technology Audio Controller'
        class      = multimedia
        subclass   = audio
    ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0a3 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP SMBus Controller'
        class      = serial bus
        subclass   = SMBus
    none8@pci0:0:31:5:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0a4 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP SPI Controller'
        class      = serial bus
    pcib4@pci65532:224:29:0:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0b0 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    pcib5@pci65532:224:29:2:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0b2 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        class      = bridge
        subclass   = PCI-PCI
    nvme0@pci65532:225:0:0:    class=0x010802 rev=0x03 hdr=0x00 vendor=0x8086 device=0x0975 subvendor=0x8086 subdevice=0x8410
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM
    nvme1@pci65532:226:0:0:    class=0x010802 rev=0x00 hdr=0x00 vendor=0x8086 device=0x0975 subvendor=0x8086 subdevice=0x8510
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM
    rtsx0@pci0:87:0:0:    class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x103c subdevice=0x87f7
        vendor     = 'Realtek Semiconductor Co., Ltd.'
        device     = 'RTS525A PCI Express Card Reader'

**DRM KMOD version**

Happens on `5.6-stable` and `master` via Git.

**To Reproduce**
Steps to reproduce the behavior:

Load drm-kmod on Intel TigerLake, attempt to use the backlight via `acpi_video` or the `backlight` command. It doesn't work.

I remember my ill-fated `5.7-wip` branch did not show this issue, but that was on the 13" Spectre x360 and I'm right now on the 14" x360.

**Screenshots**

N/A

**Additional context**

My ill-fated `5.7-wip` branch does not show this issue, but on a different laptop.

evadot, 2021-10-16:
Can you set hw.dri.__drm_debug=0x4 in loader.conf (or at loader prompt) to have more debug messages and post them please ?

evadot, 2021-10-16:
I'm wondering too if backlight needs the DMC firmware, currently the gpu-firmware-kmod package doesn't install the tigerlake firmware, it's fixed upstream in https://github.com/freebsd/drm-kmod-firmware/ and in https://reviews.freebsd.org/D32497
Since DMC manage power state if could also do some backlight related stuff.

neelchauhan, 2021-10-16:
Setting the backlight via `xrandr` works fine, but it gets reset in certain places such as GNOME "Night Light" or playing YouTube via Firefox.

unrelentingtech, 2021-11-23:
@neelchauhan ping, have you looked at the links above?

neelchauhan, 2021-12-23:
Sorry for the extreme delay. I didn't get time to check this out until now.

I did check the above links, but just now.

The mentioned tweak does work, but on FreeBSD, I have to patch the file itself:

    diff --git a/drivers/gpu/drm/i915/i915_params.h b/drivers/gpu/drm/i915/i915_params.h
    index 630155fcc..433878a97 100644
    --- a/drivers/gpu/drm/i915/i915_params.h
    +++ b/drivers/gpu/drm/i915/i915_params.h
    @@ -70,7 +70,7 @@ struct drm_printer;
     	param(unsigned int, reset, 3, 0600) \
     	param(unsigned int, inject_probe_failure, 0, 0600) \
     	param(int, fastboot, -1, 0600) \
    -	param(int, enable_dpcd_backlight, -1, 0600) \
    +	param(int, enable_dpcd_backlight, 1, 0600) \
     	param(char *, force_probe, CONFIG_DRM_I915_FORCE_PROBE, 0400) \
     	param(unsigned long, fake_lmem_start, 0, 0400) \
     	/* leave bools at the end to not create holes */ \

evadot, 2022-03-30:
That should work by setting hw.i915kms.enable_dpcd_backlight=1 in loader.conf
If we have more reports we can switch the default in i915_params.h or someone with this hardware will have to dig deeper

evadot, 2022-05-20:
No news from OP, closing now. If this still doesn't please re-open with more info.

closed 2022-05-20 by evadot