freebsd/drm-kmod #128: drm/i915/execlists: Track inflight CCID causes a boot panic in 5.7-wip

khodek, 2021-11-10:
Hello,

I've found that commit ee508d53580a182e34bc78df67e67e63aa75e219 panics after mode setting for me. Anybody else willing to test this? I'm on Kaby Lake. The code seems kinda related to my other problems on stable versions.

```
__curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
55              __asm("movq %%gs:%P1,%0" : "=r" (td) : "n" (offsetof(struct pcpu,
(kgdb) #0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
#1  doadump (textdump=<optimized out>)
    at /usr/src/sys/kern/kern_shutdown.c:399
#2  0xffffffff80c28c36 in kern_reboot (howto=260)
    at /usr/src/sys/kern/kern_shutdown.c:487
#3  0xffffffff80c290a6 in vpanic (fmt=0xffffffff811b0151 "%s",
    ap=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:920
#4  0xffffffff80c28ea3 in panic (fmt=<unavailable>)
    at /usr/src/sys/kern/kern_shutdown.c:844
#5  0xffffffff810b51d7 in trap_fatal (frame=0xfffffe00107bbc50,
    eva=18446741878757404736) at /usr/src/sys/amd64/amd64/trap.c:944
#6  0xffffffff810b522f in trap_pfault (frame=frame@entry=0xfffffe00107bbc50,
    usermode=false, signo=<optimized out>, signo@entry=0x0,
    ucode=<optimized out>, ucode@entry=0x0)
    at /usr/src/sys/amd64/amd64/trap.c:763
#7  0xffffffff810b48aa in trap (frame=0xfffffe00107bbc50)
    at /usr/src/sys/amd64/amd64/trap.c:441
#8  <signal handler called>
#9  0xffffffff82f50600 in execlists_schedule_out ()
   from /boot/modules/i915kms.ko
#10 0xffffffff82f50aba in process_csb () from /boot/modules/i915kms.ko
#11 0xffffffff82f4d6d9 in execlists_submission_tasklet ()
   from /boot/modules/i915kms.ko
#12 0xffffffff80e6d6a0 in tasklet_handler (arg=0xfffffe00d2a83000)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_tasklet.c:88
#13 0xffffffff80c7634d in gtaskqueue_run_locked (
    queue=queue@entry=0xfffff80003619900)
    at /usr/src/sys/kern/subr_gtaskqueue.c:371
#14 0xffffffff80c75fa3 in gtaskqueue_thread_loop (arg=<optimized out>,
    arg@entry=0xfffffe0011f67050) at /usr/src/sys/kern/subr_gtaskqueue.c:547
#15 0xffffffff80be5c7a in fork_exit (
    callout=0xffffffff80c75ee0 <gtaskqueue_thread_loop>,
    arg=0xfffffe0011f67050, frame=0xfffffe00107bbf40)
    at /usr/src/sys/kern/kern_fork.c:1092
#16 <signal handler called>
#17 mi_startup () at /usr/src/sys/kern/init_main.c:322
Backtrace stopped: Cannot access memory at address 0x11
```

khodek, 2021-11-10:
To clarify, the previous commit is okay.

I'm trying to track down a mysterious regression where I can easily trigger panics with Vulkan. I tested several FreeBSD versions too without resolution, so I'm up for whatever.

wulf7, 2021-11-10:
It looks like mismerge happened in one of previous commits. Test updated branch.

khodek, 2021-11-11:
That did the trick, thanks.

closed 2021-11-11 by khodek

khodek, 2021-11-11:
Maybe fixed. I was able to launch 3 affected games successfully whereas it was a certain panic or crash before.
https://gitlab.freedesktop.org/mesa/mesa/-/compare/38f0b36f1ac...6eb86efe915
drm-kmod 5.7-wip at e3d23858bf541e9db546539a15e1c01ac2a38fa4 
14.0-CURRENT #0 main-n250599-439c3d95638