freebsd/drm-kmod #129: 5.7-wip: High levels of graphics corruption on TigerLake on OpenGL & SNA

neelchauhan, 2021-11-11:
**Describe the bug**

On an HP Spectre x360 14-ea1023dx (TigerLake) running 14-CURRENT and `5.7-wip`, there are high levels of graphics corruption on OpenGL when running with `sna` acceleration. This was seen on both the Windows and Linux versions of [Toontown Rewritten](http://www.toon.town)

**FreeBSD version**

    FreeBSD concorde.local 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n250301-4827bf76bce: Thu Oct 28 07:16:45 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Details**

    hostb0@pci0:0:0:0:    class=0x060000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a14 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = '11th Gen Core Processor Host Bridge/DRAM Registers'
        class      = bridge
        subclass   = HOST-PCI
    vgapci0@pci0:0:2:0:    class=0x030000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a49 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'TigerLake-LP GT2 [Iris Xe Graphics]'
        class      = display
        subclass   = VGA
    none0@pci0:0:4:0:    class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a03 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'TigerLake-LP Dynamic Tuning Processor Participant'
        class      = dasp
    pcib1@pci0:0:7:0:    class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a23 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    pcib2@pci0:0:7:1:    class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a25 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    none1@pci0:0:8:0:    class=0x088000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a11 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'GNA Scoring Accelerator module'
        class      = base peripheral
    xhci0@pci0:0:13:0:    class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a13 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 USB Controller'
        class      = serial bus
        subclass   = USB
    none2@pci0:0:13:2:    class=0x0c0340 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a1b subvendor=0x2222 subdevice=0x1111
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 NHI'
        class      = serial bus
        subclass   = USB
    pcib3@pci0:0:14:0:    class=0x010400 rev=0x00 hdr=0x00 vendor=0x8086 device=0x9a0b subvendor=0x8086 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = 'Volume Management Device NVMe RAID Controller'
        class      = mass storage
        subclass   = RAID
    none3@pci0:0:18:0:    class=0x070000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0fc subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Integrated Sensor Hub'
        class      = simple comms
        subclass   = UART
    xhci1@pci0:0:20:0:    class=0x0c0330 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0ed subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller'
        class      = serial bus
        subclass   = USB
    none4@pci0:0:20:2:    class=0x050000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0ef subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Shared SRAM'
        class      = memory
        subclass   = RAM
    none5@pci0:0:20:3:    class=0x028000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0f0 subvendor=0x8086 subdevice=0x0074
        vendor     = 'Intel Corporation'
        device     = 'Wi-Fi 6 AX201'
        class      = network
    ig4iic0@pci0:0:21:0:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e8 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Serial IO I2C Controller'
        class      = serial bus
    ig4iic1@pci0:0:21:1:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e9 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Serial IO I2C Controller'
        class      = serial bus
    none6@pci0:0:22:0:    class=0x078000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e0 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Management Engine Interface'
        class      = simple comms
    pcib6@pci0:0:28:0:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0bd subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tigerlake PCH-LP PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    none7@pci0:0:29:0:    class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x09ab subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        class      = base peripheral
    isab0@pci0:0:31:0:    class=0x060100 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa082 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP LPC Controller'
        class      = bridge
        subclass   = PCI-ISA
    hdac0@pci0:0:31:3:    class=0x040100 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0c8 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Smart Sound Technology Audio Controller'
        class      = multimedia
        subclass   = audio
    ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0a3 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP SMBus Controller'
        class      = serial bus
        subclass   = SMBus
    none8@pci0:0:31:5:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0a4 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP SPI Controller'
        class      = serial bus
    pcib4@pci65532:224:29:0:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0b0 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    pcib5@pci65532:224:29:2:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0b2 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        class      = bridge
        subclass   = PCI-PCI
    nvme0@pci65532:225:0:0:    class=0x010802 rev=0x03 hdr=0x00 vendor=0x8086 device=0x0975 subvendor=0x8086 subdevice=0x8410
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM
    nvme1@pci65532:226:0:0:    class=0x010802 rev=0x00 hdr=0x00 vendor=0x8086 device=0x0975 subvendor=0x8086 subdevice=0x8510
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM
    rtsx0@pci0:87:0:0:    class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x103c subdevice=0x87f7
        vendor     = 'Realtek Semiconductor Co., Ltd.'
        device     = 'RTS525A PCI Express Card Reader'

**DRM KMOD version**
Paste the output of `pkg query -x '%n %v' 'drm.*kmod'`

`5.7-wip` as of November 10th at 17:10 PM PST

**To Reproduce**

Take a TigerLake system, use Toontown Rewritten (or any OpenGL app), you should see it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

Some levels of graphics corruption is shown here. It's hard to capture it as a picture.

**Additional context**

 * This also happened on an older unrelated/rejected `5.7-wip` branch on a 13" HP Spectre (also TigerLake), but it's less severe in the current `5.7-wip`

`5.6-stable` and `5.7-wip` with `uxa` does not have this issue.

unrelentingtech, 2021-11-23:
Is sna/uxa stuff even supposed to work with such new hardware? xf86-video-intel is deprecated, just use xf86-video-modesetting. (These “accelerations” are really for xorg 2d things but they can have unpredictable effects on everything somehow)

neelchauhan, 2021-12-06:
To use SNA, I need to do this in `/etc/X11/xorg.conf`:

wulf7, 2021-12-06:
Are you still using one month old version of 5.7-wip branch?

unrelentingtech, 2021-12-06:
Soooo have you tried the xf86 `modesetting` driver? Or wayland ;)

neelchauhan, 2021-12-08:
I am using the current `5.7-wip` branch with the `modesetting` driver.

neelchauhan, 2021-12-08:
I could also test my WhiskeyLake laptop to see if it works.

wulf7, 2021-12-08:
How can I run `Toontown Rewritten` to test it?
If I run installer on our default centos7-based linuxolator, it reports about missing PyQt4.QtCore
```
% ./Launcher 
Traceback (most recent call last):
  File "PyInstaller-3.6-py2.7.egg/PyInstaller/loader/rthooks/pyi_rth_qt4plugins.py", line 45, in <module>
ImportError: No module named PyQt4.QtCore
[2976] Failed to execute script pyi_rth_qt4plugins
```

wulf7, 2021-12-08:
Also if it is regression after `5.6-stable` you can bisect commits to find guilty

neelchauhan, 2021-12-08:
Toontown Rewritten requires the Ubuntu-based Linuxulator. The CentOS 7-based Linuxulator doesn't work, or even CentOS 7 itself.

neelchauhan, 2021-12-08:
I have an Ubuntu Linuxulator installer here: https://github.com/neelchauhan/linux-browser-installer

My installer is a forked version built to unbreak 14-CURRENT.

You can also install Ubuntu in the Linuxulator manually here: https://forums.freebsd.org/threads/linuxulator-how-to-run-google-chrome-linux-binary-on-freebsd.77559/

wulf7, 2021-12-09:
Thank you for the hints. Now I am able to run it.
I see no corruption on SkyLake with SNA. TigerLake - specific bug?

neelchauhan, 2021-12-09:
I believe it's a TigerLake-specific bug since my WhiskeyLake laptop didn't show corruption either.

evadot, 2022-06-01:
@neelchauhan can you give an update on this ? Is this still happening with 5.10 ? Also you didn't answered @unrelentingtech question, what about modesetting or wayland ?

neelchauhan, 2022-06-01:
It doesn't happen on 5.10 (or 5.9) on modesetting. I haven't tried Wayland since Toontown Rewritten (for Linux) is an X11 app, and I haven't gotten GNOME working on Wayland yet.

Toontown Rewritten has some graphics corruption issues on the Linuxulator on both TigerLake and WhiskeyLake. Nowadays I use the Wine version since it has no corruption whatsoever.

closed 2022-06-01 by neelchauhan