freebsd/drm-kmod #131: Black screen in xorg on 965GM

dmitryluhtionov, 2021-11-30:
**Describe the bug**
When I starting xorg - I see simple black screen, but DPMS working (backlight).
Native i915kms in 12.2-STABLE working, but with GLX loading error in Xorg.0.log
VT console steel working.

**FreeBSD version**
12.2-STABLE i386, 13.0-STABLE i386, 13.0-STABLE amd64

**PCI Info**

<details>
  <summary> pciconf -lv </summary>
 
>    vgapci0@pci0:0:2:0:     class=0x030000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x2a02 subvendor=0x17aa subdevice=0x20b5
>    vendor     = 'Intel Corporation'
>   device     = 'Mobile GM965/GL960 Integrated Graphics Controller (primary)'
>  class      = display
>    subclass   = VGA

</details>

**DRM KMOD version**

Paste the output of `pkg query -x '%n %v' 'drm.*kmod'`
drm-fbsd12.0-kmod 4.16.g20201016 on 12-STABLE

drm-fbsd13-kmod 5.4.144.g20211013 on 13-STABLE

**To Reproduce**
Install drm-kmod from packages
Try any driver: vesa, modesetting, intel

**Screenshots**
Sibmple black screen in Xorg.

**Additional context**
[dmesg.txt](https://github.com/freebsd/drm-kmod/files/7627007/dmesg.txt)
[Xorg.0.log](https://github.com/freebsd/drm-kmod/files/7627009/Xorg.0.log)

evadot, 2021-12-02:
12.X isn't supported here, this repo is only for FreeBSD >= 13
I strongly suggest that you update to 13 as drm isn't really maintained for 12

dmitryluhtionov, 2021-12-02:
> 12.X isn't supported here, this repo is only for FreeBSD >= 13 I strongly suggest that you update to 13 as drm isn't really maintained for 12

FreeBSD R61 13.0-STABLE FreeBSD 13.0-STABLE #0 stable/13-n248228-354988ca3f9: Thu Nov 25 02:31:45 UTC 2021     root@releng3.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

evadot, 2021-12-02:
Sorry I've missed this info

dmitryluhtionov, 2021-12-02:
How can I help in finding the problem? 
The console continues to work fine. Only xorg produce black screen

dmitryluhtionov, 2021-12-02:
Hm
I trying run `startx` directly from console
And I see in /var/log/messages
Dec  2 10:00:29 R61 devd[574]: notify_clients: send() failed; dropping unresponsive client
Dec  2 10:00:29 R61 syslogd: last message repeated 1 times
Dec  2 10:01:07 R61 kernel: pid 1187 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:08 R61 kernel: pid 1188 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:09 R61 kernel: pid 1189 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:10 R61 kernel: pid 1190 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:12 R61 kernel: pid 1191 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:13 R61 kernel: pid 1192 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:14 R61 kernel: pid 1194 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:15 R61 kernel: pid 1195 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:16 R61 kernel: pid 1196 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:17 R61 kernel: pid 1197 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:19 R61 kernel: pid 1198 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:20 R61 kernel: pid 1199 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:21 R61 kernel: pid 1200 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:21 R61 kernel: pid 1201 (icewm), jid 0, uid 1001: exited on signal 11 (core dumped)
Dec  2 10:01:27 R61 devd[574]: check_clients:  dropping disconnected client

Before this also `slim` core dumped

I managed to run successfully `twm`

dmitryluhtionov, 2021-12-02:
mitya@R61:/# lldb --core slim.core `which slim`
(lldb) target create "/usr/local/bin/slim" --core "slim.core"
Core file '/slim.core' (x86_64) was loaded.

(lldb) bt
* thread #1, name = 'slim', stop reason = signal SIGSEGV
  * frame #0: 0x00000008004d5114 libXft.so.2`XftDrawGlyphs + 36
    frame #1: 0x00000008004d5585 libXft.so.2`XftDrawString16 + 309
    frame #2: 0x0000000800720928 libslim.so.1.3.6`Panel::OnExpose(this=0x00000008010a8000) at panel.cpp:0
    frame #3: 0x0000000800722dfb libslim.so.1.3.6`Panel::EventHandler(this=0x00000008010a8000, curfield=<unavailable>) at panel.cpp:465:3
    frame #4: 0x0000000801858fac slim`conv(num_msg=<unavailable>, msg=<unavailable>, resp=<unavailable>, appdata_ptr=<unavailable>) at app.cpp:64:12
    frame #5: 0x00000008002b63fa libpam.so.6`pam_vprompt + 266
    frame #6: 0x00000008002b589f libpam.so.6`pam_prompt + 127
    frame #7: 0x00000008002b53af libpam.so.6`pam_get_user + 303
(lldb)

dmitryluhtionov, 2021-12-02:
```
mitya@R61:/home/mitya# lldb --core icewm.core `which icewm`
(lldb) target create "/usr/local/bin/icewm" --core "icewm.core"
Core file '/home/mitya/icewm.core' (x86_64) was loaded.

(lldb) bt
* thread #1, name = 'icewm', stop reason = signal SIGSEGV
  * frame #0: 0x000000000030d05e icewm`YClock::calcWidth(this=0x00000008012f2f00, s="23:48:59", count=8) at aclock.cc:444:27
    frame #1: 0x000000000030cced icewm`YClock::autoSize(this=0x00000008012f2f00) at aclock.cc:85:13
    frame #2: 0x000000000030cb3f icewm`YClock::YClock(this=0x00000008012f2f00, smActionListener=0x00007fffffffe970, iapp=0x00000008012f2dd0, aParent=0x00000008012f2c00) at aclock.cc:58:5
    frame #3: 0x000000000027166e icewm`TaskBar::initApplets(this=0x00000008012f2c00) at wmtaskbar.cc:276:22
    frame #4: 0x00000000002711fc icewm`TaskBar::TaskBar(this=0x00000008012f2c00, app=0x00007fffffffe7e0, aParent=0x0000000801282800, wmActionListener=0x00007fffffffe960, smActionListener=0x00007fffffffe970) at wmtaskbar.cc:171:5
    frame #5: 0x00000000002b84bb icewm`YWMApp::createTaskBar(this=0x00007fffffffe7e0) at wmapp.cc:1786:23
    frame #6: 0x00000000002b83ab icewm`YWMApp::handleIdle(this=0x00007fffffffe7e0) at wmapp.cc:1443:9
    frame #7: 0x0000000000368930 icewm`YApplication::mainLoop(this=0x00007fffffffe7e0) at yapp.cc:242:24
    frame #8: 0x00000000002b8177 icewm`YWMApp::mainLoop(this=0x00007fffffffe7e0) at wmapp.cc:1393:21
    frame #9: 0x00000000002b92e6 icewm`main(argc=1, argv=0x00007fffffffead8) at wmapp.cc:1781:16
    frame #10: 0x00000000002604f0 icewm`_start(ap=<unavailable>, cleanup=<unavailable>) at crt1_c.c:75:7
(lldb)
```

dmitryluhtionov, 2021-12-02:
I think, i915kms report wrong screen/buffer size or similar
Where can i watch it?

closed 2021-12-13 by dmitryluhtionov