freebsd/drm-kmod #133: kodi cec drm-kmod

halfagascan, 2021-12-22:
Clean install of FreeBSD 13.0-RELEASE-p4 amd64.
Intel(R) Core(TM) i5-4250U CPU @ 1.30GHz NUC
Pulse eight usb adaptor.
Base installed, desktop-installer installed, kodi installed, some errors see bug PR 260490,
Kodi starts and runs, plays iptv channels, no issues. Just need to use a usb mouse to control.
As part of kodi install, libcec-6.0.2, gets installed.
As user:
echo 'scan' | cec-client -s -d 1
ERROR:   [            3553]     error opening serial port '/dev/ttyU0': Permission denied

ls -all /dev/ttyU0 crw-------  1 root  wheel  0x182 Dec 19 09:09 /dev/ttyU0

id uid=1001(jerry) gid=1001(jerry) groups=1001(jerry),0(wheel)

as root:
echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
ERROR:   [           19932]     failed to ping the adapter 3 times in a row. closing the connection.
Connection lost - trying to reconnect

requesting CEC bus information ...
ERROR:   [           40898]     failed to ping the adapter 3 times in a row. closing the connection.
Connection lost - trying to reconnect

CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   unknown
power status:  on
language:      ???

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Unknown
osd string:    Recorder 1
CEC version:   unknown
power status:  unknown
language:      eng

Part of cec debug, much to long to post here:

CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   unknown
power status:  unknown
language:      ???

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng
currently active source: unknown (-1)
Illegal instruction (core dumped)

 ls -all -rw-------   1 root  wheel  38088704 Dec 19 09:45 cec-client-6.0.2.core

And cec never connects to or sends to the tv
Some internet searching seems to suggest a bad hdmi cable, this same set works fine on linux.
Can anyone provide more troubleshooting advice or a fix?

Maybe this is related?
https://github.com/Pulse-Eight/libcec/commit/8d5379dce05f646975635d0f83dbe0d69a2e5d20
Also filed this :Bug 260579

evadot, 2021-12-22:
We don't have CEC support in the compat code so no native CEC can be used for now.
Your adapter seems to expose a uart<->usb interface so maybe try to add your user into the dialer group.
Closing this issue as it have nothing to do with DRM.

closed 2021-12-22 by evadot

halfagascan, 2021-12-22:
>no native CEC can be used for now<
Says to me, CEC will not work, is that correct?
Added to dialer, same as before
thanks

evadot, 2021-12-22:
Native CEC will not work, you're trying to use an adapter that breaks out the CEC line of the HDMI cable to USB, this might work without a lot of work I think but this is out of scope of this project.

halfagascan, 2021-12-22:
I have filed this bug Bug 260579] multimedia/libcec, that the correct place?
Thanks for out of scope answers :)