freebsd/drm-kmod #134: Cannot use VAAPI on headless amdgpu devices

agrajag9, 2021-12-27:
**Describe the bug**
VAAPI calls may never return if there's no display attached. This is at least true for the WX 2100.

There's an option in [`amdgpu_drv.c`](https://github.com/freebsd/drm-kmod/blob/drm_v5.4.144_3/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c#L453-L463) but I wasn't able to get it working.

**FreeBSD version**
stable/13-n248443-c89c8b894cf GENERIC amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
vgapci0@pci0:3:0:0:     class=0x030000 rev=0x00 hdr=0x00 vendor=0x1002 device=0x6995 subvendor=0x1028 subdevice=0x0b0c
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Lexa XT [Radeon PRO WX 2100]'
    class      = display
    subclass   = VGA
```
</details>

**DRM KMOD version**
```
drm-fbsd13-kmod 5.4.144.g20211013
drm-kmod g20190710_1
```
**To Reproduce**

1. Detach all displays from amdgpu device
2. Run `LIBVA_TRACE=1 LIBVA_DRIVER_NAME=radeonsi vainfo`

unrelentingtech, 2022-01-21:
> There's an option in amdgpu_drv.c but I wasn't able to get it working.

It's under `#ifdef __linux__` so we don't have it.

I have no idea why that is the case, this was added in a [really old commit](https://github.com/freebsd/drm-kmod/commit/941d050cfe4358f422c6e7d9af470d50c21a6411) and nothing that actually *reads* the value is ifdef'd, it's literally only the sysctl itself.

Please do try removing the ifdef and using the functionality.

unrelentingtech, 2022-02-11:
With https://reviews.freebsd.org/D34248 + https://reviews.freebsd.org/D34252 + #146 it will work.

agrajag9, 2022-02-20:
Wonderful! Sorry for not trying the quick fix on my end, both $HOME and $WORK have been oversaturated with more important stuff, but glad to see we've gotten there anyways!

closed 2022-02-20 by agrajag9