freebsd/drm-kmod #139: rocketlake processor (i9-11900k) graphics not detected

yuripv, 2022-01-24:
**Describe the bug**
RocketLake processor (i9-11900k) graphics not detected.  Adding the pci id under the TigerLake section in include/drm/i915_pciids.h, as well as "porting" the RocketLake defines in i915_pciids.h and i915_pci.c seems to help to some extent, but I only get 2560x1440 resolution as a result having 4K display connected via DisplayPort -- checked using latest Ubuntu LiveCD that it is NOT the problem with GPU/cable/monitor, have the correct 3840x2160 resolution there.

**FreeBSD version**
FreeBSD saturn.home.arpa 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n252579-5025e850132d: Sun Jan 23 22:27:45 MSK 2022     yuri@saturn.home.arpa:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
hostb0@pci0:0:0:0:      class=0x060000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x4c43 subvendor=0x1458 subdevice=0x5000
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x4c8a subvendor=0x1458 subdevice=0xd000
    vendor     = 'Intel Corporation'
    device     = 'RocketLake-S GT1 [UHD Graphics 750]'
    class      = display
    subclass   = VGA
pcib1@pci0:0:6:0:       class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x4c09 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = PCI-PCI
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43ed subvendor=0x1458 subdevice=0x5007
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H USB 3.2 Gen 2x1 xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none0@pci0:0:20:2:      class=0x050000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43ef subvendor=0x0000 subdevice=0x0000
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H Shared SRAM'
    class      = memory
    subclass   = RAM
none1@pci0:0:22:0:      class=0x078000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43e0 subvendor=0x1458 subdevice=0x1c3a
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H Management Engine Interface'
    class      = simple comms
ahci0@pci0:0:23:0:      class=0x010601 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43d2 subvendor=0x1458 subdevice=0xb005
    vendor     = 'Intel Corporation'
    class      = mass storage
    subclass   = SATA
pcib2@pci0:0:28:0:      class=0x060400 rev=0x11 hdr=0x01 vendor=0x8086 device=0x43b8 subvendor=0x1458 subdevice=0x5001
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:4:      class=0x060400 rev=0x11 hdr=0x01 vendor=0x8086 device=0x43bc subvendor=0x1458 subdevice=0x5001
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:28:6:      class=0x060400 rev=0x11 hdr=0x01 vendor=0x8086 device=0x43be subvendor=0x1458 subdevice=0x5001
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:29:0:      class=0x060400 rev=0x11 hdr=0x01 vendor=0x8086 device=0x43b0 subvendor=0x1458 subdevice=0x5001
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:      class=0x060100 rev=0x11 hdr=0x00 vendor=0x8086 device=0x4385 subvendor=0x1458 subdevice=0x5001
    vendor     = 'Intel Corporation'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:      class=0x040300 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43c8 subvendor=0x1458 subdevice=0xa0d5
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H HD Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43a3 subvendor=0x1458 subdevice=0x5001
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H SMBus Controller'
    class      = serial bus
    subclass   = SMBus
none2@pci0:0:31:5:      class=0x0c8000 rev=0x11 hdr=0x00 vendor=0x8086 device=0x43a4 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Intel Corporation'
    device     = 'Tiger Lake-H SPI Controller'
    class      = serial bus
nvme0@pci0:1:0:0:       class=0x010802 rev=0x01 hdr=0x00 vendor=0x1987 device=0x5013 subvendor=0x1987 subdevice=0x5013
    vendor     = 'Phison Electronics Corporation'
    device     = 'PS5013 E13 NVMe Controller'
    class      = mass storage
    subclass   = NVM
aq0@pci0:3:0:0: class=0x020000 rev=0x02 hdr=0x00 vendor=0x1d6a device=0xd107 subvendor=0x1458 subdevice=0xe000
    vendor     = 'Aquantia Corp.'
    device     = 'AQC107 NBase-T/IEEE 802.3bz Ethernet Controller [AQtion]'
    class      = network
    subclass   = ethernet
none3@pci0:4:0:0:       class=0x028000 rev=0x1a hdr=0x00 vendor=0x8086 device=0x2725 subvendor=0x8086 subdevice=0x0024
    vendor     = 'Intel Corporation'
    device     = 'Wi-Fi 6 AX210/AX211/AX411 160MHz'
    class      = network
nvme1@pci0:5:0:0:       class=0x010802 rev=0x01 hdr=0x00 vendor=0x1987 device=0x5016 subvendor=0x1987 subdevice=0x5016
    vendor     = 'Phison Electronics Corporation'
    device     = 'E16 PCIe4 NVMe Controller'
    class      = mass storage
    subclass   = NVM
</details>

**DRM KMOD version**
drm-devel-kmod 5.5.19.g20211230

yuripv, 2022-01-24:
I wonder if backporting the rocketlake support to current drm-kmod's master would work (as in accepted and useful for others), or should I wait until the newer version proper is imported?

yuripv, 2022-01-24:
Any help required to make that happen?

evadot, 2022-01-26:
> Any help required to make that happen?

I think that @amshafer started to do a 5.8 update but don't know more.
I need to take time to update to 5.10 (at least) soon anyway.

VicoChern, 2022-02-16:
@evadot
5.15 LTS has been released, why can't we bypass 5.8, 5.9, .. and move the work to 5.15 directly?

evadot, 2022-02-16:
We can't bypass anything, to update to 5.15 we need to go thought 5.8 5.9 etc ...
I have 5.8 in WIP, will do 5.9 and 5.10 just after.

VicoChern, 2022-03-01:
@evadot ,
I do the git graph for the repo, the porting work looks like cherry pick each commit in Linux drm, so we can't bypass any Linux versions. Am I right?

evadot, 2022-03-30:
That should be ok in master (tracking v5.10 now) but as I don't have the hardware this wasn't tested.

evadot, 2022-05-19:
Closing, if it's not working please open a new issue with a full report.

closed 2022-05-19 by evadot