freebsd/drm-kmod #142: can not build on FreeBSD 14.0-CURRENT

x-shayan, 2022-02-02:
Hi. thanks for your job.  I'm not sure that I'm asking in right place or not, cause I can't find your E-Mail.
well, I wanna get my AMD Vega 6 (Renior) to work. I searched in FreeBSD forums, someone told that I should compile drm-kmod onto FreeBSD 14.0-CURRENT. I'm trying to build v5.6-stable and v5.7-stable because the drm-current-kmod port from FreeBSD has kernel panic for me. 

but I'm getting the following error: 

[image]

Sorry, I have not GUI for copying pciconf or uname -a. but It is FreeBSD-14.0-CURRENT-amd64-20220127 build. 
I did it:

git clone -b 5.6-stable https://github.com/freebsd/drm-kmod
cd drm-kmod
make 
make install

wulf7, 2022-02-02:
You should install drm-devel-kmod port rather then drm-current-kmod to get v5.7

x-shayan, 2022-02-02:
oh, sorry I wrote wrong. the drm-current-kmod port can't detect my gpu.
drm-devel-kmod can, but gives me kernel panic.

evadot, 2022-02-02:
5.5 and 5.6 branch aren't supported anymore.
Try master or 5.7
And if you also have a panic with thoses open a issue with all the details.
5.7-stable is know to compile fine on current.
Thanks.

x-shayan, 2022-02-02:
thanks for your quick reply.
I compiled and installed 5.7-branch. when I run startxfce4 or startx, it panics again.

My Device:
Lenovo Ideapad Gaming 3 15ARH05 
AMD Ryzen 5 4600H with AMD Vega 6 (Renoir) Graphics and GTX-1650Ti

kldstat:
[image]

pciconf of my gpu:
[image]

dmesg: 

[image]

[image]
[image]

panic details:

[image]

FreeBSD 14.0-CURRENT
I hope you help me and I'm sorry about that I can't send text logs

x-shayan, 2022-02-02:
I suprised. It works now! cool! thank you very much. 
another little thing that, why amdgpu.ko can't automatically load at boot? 
I have kld_list="amdgpu" in /etc/rc.conf

closed 2022-02-03 by evadot