freebsd/drm-kmod #143: 5.7-stable not building on 13.0-RELEASE-p7

fellmoon, 2022-02-09:
**Describe the bug**
I'm trying to build 5.7-stable branch on FreeBSD 13.0-RELEASE-p7 to test if drm-kmod works with ryzen 5 pro 5650u but it fails after a couple of seconds due to a missing file

**FreeBSD version**
FreeBSD 13.0-RELEASE-p7

**DRM KMOD version**
5.7-stable from github

**To Reproduce**
git clone https://github.com/freebsd/drm-kmod.git /root/drm-kmod
git checkout 5.7-stable
make

produces the following error after a few seconds:

/root/drm-kmod/linuxkpi/bsd/include/linux/kernel.h:15:10: fatal error: 'asm/cpufeature.h' file not found
#include <asm/cpufeature.h>

*** Error code 1

Stop.
make: stopped in /root/drm/kmod

**Additional context**
The Ryzen 5 pro 5650(cezanne, device ID 0x1638) of my Thinkpad p14s gen 2 amd seems not to be supported with drm-kmod from pkg or from ports so I thought I try the version from git before opening an issue about not supported amdgpu but as it looks I'm not reaching that point due to the error above.
I'd like to point out that I have very limited experience with FreeBSD and building software so maybe I've just overlooked something.

wulf7, 2022-02-11:
5.7-stable is not supposed to be built on 13.0-RELEASE due to missing LinuxKPI bits. Use recent 14-CURRENT or 13-STABLE

fellmoon, 2022-02-13:
okay, builds on 13-STABLE without further issues, will open another issue for the rest.

closed 2022-02-13 by fellmoon