freebsd/drm-kmod #144: drmn0: GPU HANG: ecode 9:1:0x00000000, hang on rcs0

Jehops, 2022-02-10:
After upgrading a Skylake system (Intel HD 530 video) from 13.0-R to main-n253041-ff5025d10450, the system frequently panics.  The panics seem to occur randomly when browsing the web.
 
`uname -a`

```
FreeBSD ala.ftfl.ca 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n253041-ff5025d10450: Wed Feb  9 13:01:21 AST 2022     root@ala.ftfl.ca:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64
```

`pciconf -vl`

```
jrm@ala ~ % pciconf -lv
hostb0@pci0:0:0:0:      class=0x060000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x190f subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x06 hdr=0x00 vendor=0x8086 device=0x1912 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics 530'
    class      = display
    subclass   = VGA
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa12f subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
none0@pci0:0:22:0:      class=0x078000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa13a subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family MEI Controller'
    class      = simple comms
ahci0@pci0:0:23:0:      class=0x010601 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa102 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = 'Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
pcib1@pci0:0:28:0:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa116 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:      class=0x060100 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa148 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = 'B150 Chipset LPC/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none1@pci0:0:31:2:      class=0x058000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa121 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Power Management Controller'
    class      = memory
hdac0@pci0:0:31:3:      class=0x040300 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa170 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family HD Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa123 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family SMBus'
    class      = serial bus
    subclass   = SMBus
em0@pci0:0:31:6:        class=0x020000 rev=0x31 hdr=0x00 vendor=0x8086 device=0x15b8 subvendor=0x17aa subdevice=0x30d2
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (2) I219-V'
    class      = network
    subclass   = ethernet
iwm0@pci0:1:0:0:        class=0x028000 rev=0x99 hdr=0x00 vendor=0x8086 device=0x3166 subvendor=0x8086 subdevice=0x4210
    vendor     = 'Intel Corporation'
    device     = 'Dual Band Wireless-AC 3165 Plus Bluetooth'
    class      = network
```

`pkg query -x '%n %v' 'drm.*kmod'`

```
drm-current-kmod 5.4.144.g20220128
drm-kmod g20190710_1
```

`Screenshot`
[image]

evadot, 2022-02-10:
The panic isn't related to drm, the drm_modeset_is_locked warnings are just side effect of switching vt when a panic occurs.
I suggest that you create a PR on bugs.freebsd.org with all the info that you have.
Cheers,

closed 2022-02-10 by evadot

evadot, 2022-02-10:
See this bug report : https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=261790
That seems to match what you have here.