freebsd/drm-kmod #149: Kernel panic: page fault (version 5.4.144.g20220128)

0mp, 2022-02-14:
**Describe the bug**
My kernel panics randomly after a recent upgrade FreeBSD 14.0-CURRENT upgrade.

**FreeBSD version**
The current version I'm running is: `FreeBSD 14.0-CURRENT #21 main-n253122-70e9f880d88e: Sun Feb 13 12:10:38 CET 2022`.

The previous version I was running which was pretty much rock-stable was compiled from the main branch on 2022-01-17.

**PCI Info**

<details>
  <summary>pciconf -lv</summary>

```
hostb0@pci0:0:0:0:      class=0x060000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x5914 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x17aa subdevice=0x225e
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:       class=0x118000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
none1@pci0:0:8:0:       class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d2f subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:  class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d31 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Thermal subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:    class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d60 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
none2@pci0:0:22:0:      class=0x078000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3a subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP CSME HECI'
    class      = simple comms
pcib1@pci0:0:28:0:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d10 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:6:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d16 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:29:0:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d18 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:29:2:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d1a subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:      class=0x060100 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d4e subvendor=0x17aa subdevice=0x225d
[... 212 lines cut ...]
    data=<optimized out>, cred=<optimized out>, td=0xfffffe01467b51e0)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:1648
#21 0xffffffff80c7f062 in fo_ioctl (fp=<optimized out>, com=3222299841, 
    data=0x10000, active_cred=0xfffffe0146886af0, td=0xfffffe01467b51e0)
    at /usr/src/sys/sys/file.h:364
#22 kern_ioctl (td=<optimized out>, td@entry=0xfffffe01467b51e0, 
    fd=<optimized out>, com=com@entry=3222299841, 
    data=0x10000 <error: Cannot access memory at address 0x10000>, 
    data@entry=0xfffffe0146886d50 "\005")
    at /usr/src/sys/kern/sys_generic.c:803
#23 0xffffffff80c7edb6 in sys_ioctl (td=0xfffffe01467b51e0, 
    uap=0xfffffe01467b55d0)
    at /usr/src/sys/kern/sys_generic.c:711
#24 0xffffffff810e738e in syscallenter (td=<optimized out>)
    at /usr/src/sys/amd64/amd64/../../kern/subr_syscall.c:189
#25 amd64_syscall (td=0xfffffe01467b51e0, traced=0)
    at /usr/src/sys/amd64/amd64/trap.c:1191
#26 <signal handler called>
#27 0x000000910eafa7da in ?? ()
Backtrace stopped: Cannot access memory at address 0x910940f348
(kgdb) 

```

</details>

dumbbell, 2022-02-20:
Do you still have that panic?

I got it a few times as well (using Firefox like you, visiting Google Maps triggered it very easily for instance). However, I didn't get it after upgrading or recompiling the kernel modules (I don't remember which fixed it exactly).

We probably hit an incompatibility because we are using a kernel module compiled against an old kernel source and running it on a newer kernel built after commit freebsd/freebsd-src@513c7a6e0cd8c13f765fda6fa1b2de53c1973753.

0mp, 2022-02-20:
> Do you still have that panic?
> 

No, but I'm running an older drm-kmod version now: `drm-current-kmod-5.4.144.g20211230`. 

> I got it a few times as well (using Firefox like you, visiting Google Maps triggered it very easily for instance). However, I didn't get it after upgrading or recompiling the kernel modules (I don't [...]

I'm not sure. The commit you linked to is from 2022-02-11 and I've got a build from 2022-02-13, which panics with `drm-current-kmod-5.4.144.g20220128`. I recompiled the module with the right kernel sources.

evadot, 2022-05-20:
@0mp is it still happening for you ?

evadot, 2022-06-20:
No answer, closing.

closed 2022-06-20 by evadot

0mp, 2022-06-20:
I'm running `FreeBSD 14.0-CURRENT #25 main-n255989-00b0158d2ca6: Sun Jun  5` and the following drm packages:

```
$ pkg info -g drm-*
drm-510-kmod-5.10.113
drm-kmod-20220501
```

I do not experience any kernel panics currently.