freebsd/drm-kmod #150: FreeBSD Kmod Cannot find sys/param.h

Montana, 2022-02-15:
Hey folks,

This is a sample file entitled `hello.c` 

```c
#include <sys/param.h>
#include <sys/module.h>
#include <sys/kernel.h>
#include <sys/systm.h>

static int
hello_modevent(module_t mod __unused, int event, void *arg __unused)
{
    int error = 0;
    switch (event)
    {
        case MOD_LOAD:
            uprintf("Hello, world\n");
        case MOD_UNLOAD:
            uprintf("GoodBye, cruel world");
        default:
             error = EOPNETSUPP;
    }

return (error);
}

static moduledata_t hello_mod = {
   "hello",
   hello_modevent,
   NULL
}

DECLARE_MODULE(hello, hello_mod, SI_SUB_DRIVERS, SI_ORDER_MIDDLE); 
```
It's not compiling through the `makefile`, if I remember correctly I need to have the kernel source installed if you want to build modules correct? 

Cheers,
Montana Mendy

Spineswitch, 2022-02-15:
@Montana,

You need to have the kernel source installed if you want to build modules.

Montana, 2022-02-15:
Hi @Spineswitch,

Not sure if you read any of my post but I said exactly what you said. It turned out there were some Syntax errors, I'll post the working version here. 

```c
#include <sys/param.h>
#include <sys/module.h>
#include <sys/kernel.h>
#include <sys/systm.h>
#include <sys/errno.h>

static int
hello_modevent(module_t mod __unused, int event, void *arg __unused)
{
    int error = 0;
    switch (event)
    {
        case MOD_LOAD:
            uprintf("Hello, world\n");
        case MOD_UNLOAD:
            uprintf("GoodBye, cruel world");
        default:
             error = EOPNOTSUPP;
    }

return (error);
}

static moduledata_t hello_mod = {
   "hello",
   hello_modevent,
   NULL
};

DECLARE_MODULE(hello, hello_mod, SI_SUB_DRIVERS, SI_ORDER_MIDDLE);
```

This compiled with the `makefile` and all is solved.

Cheers,
Montana Mendy

closed 2022-02-15 by evadot

Montana, 2022-02-15:
Thanks for the information @evadot, 

I'll send an email if I run into anymore roadblocks. 

Cheers, 
Montana Mendy