freebsd/drm-kmod #153: i915kms: heartbeat panic

kevans91, 2022-03-08:
**Describe the bug**
Regular panic as a result of heartbeat, both in xfce and just at a console. It looks a bit like it panicked while handling a GPU hang (see end).

**FreeBSD version**
FreeBSD 5678114cd8b3

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
> hostb0@pci0:0:0:0:	class=0x060000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x1237 subvendor=0x0000 subdevice=0x0000
>     vendor     = 'Intel Corporation'
>     device     = '440FX - 82441FX PMC [Natoma]'
>     class      = bridge
>     subclass   = HOST-PCI
> isab0@pci0:0:1:0:	class=0x060100 rev=0x00 hdr=0x00 vendor=0x8086 device=0x7000 subvendor=0x0000 subdevice=0x0000
>     vendor     = 'Intel Corporation'
>     device     = '82371SB PIIX3 ISA [Natoma/Triton II]'
>     class      = bridge
>     subclass   = PCI-ISA
> atapci0@pci0:0:1:1:	class=0x01018a rev=0x01 hdr=0x00 vendor=0x8086 device=0x7111 subvendor=0x0000 subdevice=0x0000
>     vendor     = 'Intel Corporation'
>     device     = '82371AB/EB/MB PIIX4 IDE'
>     class      = mass storage
>     subclass   = ATA
> vgapci0@pci0:0:2:0:	class=0x030000 rev=0x00 hdr=0x00 vendor=0x15ad device=0x0405 subvendor=0x15ad subdevice=0x0405
>     vendor     = 'VMware'
>     device     = 'SVGA II Adapter'
>     class      = display
>     subclass   = VGA
> em0@pci0:0:3:0:	class=0x020000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x100e subvendor=0x8086 subdevice=0x001e
>     vendor     = 'Intel Corporation'
>     device     = '82540EM Gigabit Ethernet Controller'
>     class      = network
>     subclass   = ethernet
> none0@pci0:0:4:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x80ee device=0xcafe subvendor=0x0000 subdevice=0x0000
>     vendor     = 'InnoTek Systemberatung GmbH'
>     device     = 'VirtualBox Guest Service'
>     class      = base peripheral
> pcm0@pci0:0:5:0:	class=0x040100 rev=0x01 hdr=0x00 vendor=0x8086 device=0x2415 subvendor=0x8086 subdevice=0x0000
>     vendor     = 'Intel Corporation'
>     device     = '82801AA AC'97 Audio Controller'
>     class      = multimedia
>     subclass   = audio
> ohci0@pci0:0:6:0:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x106b device=0x003f subvendor=0x0000 subdevice=0x0000
>     vendor     = 'Apple Inc.'
>     device     = 'KeyLargo/Intrepid USB'
>     class      = serial bus
>     subclass   = USB
> intsmb0@pci0:0:7:0:	class=0x068000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x7113 subvendor=0x0000 subdevice=0x0000
>     vendor     = 'Intel Corporation'
>     device     = '82371AB/EB/MB PIIX4 ACPI'
>     class      = bridge
> 

</details>

**DRM KMOD version**
5.6-stable @ d0593ef234c6ef

**To Reproduce**
Inconsistent. but often under some kind of load

**Additional context**
Core info:
```
Unread portion of the kernel message buffer:
drmn0: GPU HANG: ecode 6:1:bbd53fff, in chrome [101247]
drmn0: Resetting chip for stopped heartbeat on rcs0
drmn0: chrome[101247] context reset due to GPU hang
drmn0: GPU HANG: ecode 6:1:bde7ffff, in MainThread [100851]
Kernel page fault with the following non-sleepable locks held:
exclusive rw kernel vm object (kernel vm object) r = 0 (0xffffffff81fb6668) locked @ /usr/src/sys/vm/vm_kern.c:582
stack backtrace:
#0 0xffffffff80c7d791 at witness_debugger+0x71
#1 0xffffffff80c7e8ea at witness_warn+0x3ea
#2 0xffffffff810eda76 at trap_pfault+0x86
#3 0xffffffff810be9d8 at calltrap+0x8
#4 0xffffffff80f84add at kmem_free+0x2d
#5 0xffffffff82f66bcd at __i915_gpu_coredump_free+0x12d
#6 0xffffffff82f3bb39 at intel_gt_handle_error+0xa9
#7 0xffffffff82f29553 at heartbeat+0x113
#8 0xffffffff80e8d702 at linux_work_fn+0xe2
#9 0xffffffff80c6f89a at taskqueue_run_locked+0xaa
#10 0xffffffff80c70922 at taskqueue_thread_loop+0xc2
#11 0xffffffff80bc32a0 at fork_exit+0x80
#12 0xffffffff810bfa4e at fork_trampoline+0xe

Fatal trap 12: page fault while in kernel mode
cpuid = 3; apic id = 03
fault virtual address   = 0x61
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff80f84978
stack pointer           = 0x28:0xfffffe00835bab90
frame pointer           = 0x28:0xfffffe00835babc0
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = interrupt enabled, resume, IOPL = 0
current process         = 0 (linuxkpi_short_wq_0)
trap number             = 12
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /root/drm-kmod/drivers/gpu/drm/drm_atomic_helper.c:616
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /root/drm-kmod/drivers/gpu/drm/drm_atomic_helper.c:616
WARNING !drm_modeset_is_locked(&dev->mode_config.connection_mutex) failed at /root/drm-kmod/drivers/gpu/drm/drm_atomic_helper.c:661
                 
<4>WARN_ON(!mutex_is_locked(&fbc->lock))WARN_ON(!mutex_is_locked(&fbc->lock))WARN_ON(!mutex_is_locked(&fbc->lock))WARN_ON(!mutex_is_locked(&fbc->lock))
panic: page fault
cpuid = 3
time = 1646781454
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe00835ba950
vpanic() at vpanic+0x17f/frame 0xfffffe00835ba9a0
panic() at panic+0x43/frame 0xfffffe00835baa00
trap_fatal() at trap_fatal+0x385/frame 0xfffffe00835baa60
trap_pfault() at trap_pfault+0xab/frame 0xfffffe00835baac0
calltrap() at calltrap+0x8/frame 0xfffffe00835baac0
--- trap 0xc, rip = 0xffffffff80f84978, rsp = 0xfffffe00835bab90, rbp = 0xfffffe00835babc0 ---
_kmem_unback() at _kmem_unback+0x78/frame 0xfffffe00835babc0
kmem_free() at kmem_free+0x2d/frame 0xfffffe00835babe0
__i915_gpu_coredump_free() at __i915_gpu_coredump_free+0x12d/frame 0xfffffe00835bac30
intel_gt_handle_error() at intel_gt_handle_error+0xa9/frame 0xfffffe00835badb0
heartbeat() at heartbeat+0x113/frame 0xfffffe00835badf0
linux_work_fn() at linux_work_fn+0xe2/frame 0xfffffe00835bae40
taskqueue_run_locked() at taskqueue_run_locked+0xaa/frame 0xfffffe00835baec0
taskqueue_thread_loop() at taskqueue_thread_loop+0xc2/frame 0xfffffe00835baef0
fork_exit() at fork_exit+0x80/frame 0xfffffe00835baf30
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe00835baf30
```

evadot, 2022-03-14:
We probably shouldn't panic here but from experience when we reach "Resetting chip for stopped heartbeat on rcs0" there isn't much to do.
It's either an hardware problem or something fixed in later linux version.
Mind trying to update to master (5.8) and see if that fixes it ?

kevans91, 2022-04-21:
I tried the current master (as of ~a day or two ago), and I think we can just close it. I still get some weird rendering artifacts at times, but I've experienced a GPU hang (presumably, it stopped rendering) now that didn't result in a panic.

closed 2022-04-21 by kevans91