freebsd/drm-kmod #154: drm.ko from 5.9-wip panicks on load

vishwin, 2022-03-21:
**Describe the bug**
`drm.ko` from 5.9-wip panicks on load with `mutex Giant not owned at /usr/src/sys/kern/subr_bus.c:874`

**FreeBSD version**
`FreeBSD ardmore 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n253876-1e9ce60a6d7: Mon Mar 21 01:15:30 EDT 2022     root@ardmore:/usr/obj/usr/src/amd64.amd64/sys/ARDMORE-DEBUG  amd64`

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0:      class=0x060000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x1604 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Broadwell-U Host Bridge -OPI'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x1616 subvendor=0x17aa subdevice=0x2225
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics 5500'
    class      = display
    subclass   = VGA
hdac0@pci0:0:3:0:       class=0x040300 rev=0x09 hdr=0x00 vendor=0x8086 device=0x160c subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Broadwell-U Audio Controller'
    class      = multimedia
    subclass   = HDA
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9cb1 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP USB xHCI Controller'
    class      = serial bus
    subclass   = USB
none0@pci0:0:22:0:      class=0x078000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9cba subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP MEI Controller'
    class      = simple comms
em0@pci0:0:25:0:        class=0x020000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x15a3 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (3) I218-V'
    class      = network
    subclass   = ethernet
hdac1@pci0:0:27:0:      class=0x040300 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca0 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
pcib1@pci0:0:28:0:      class=0x060400 rev=0xe3 hdr=0x01 vendor=0x8086 device=0x9c9a subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:1:      class=0x060400 rev=0xe3 hdr=0x01 vendor=0x8086 device=0x9c94 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:4:      class=0x060400 rev=0xe3 hdr=0x01 vendor=0x8086 device=0x9c98 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
ehci0@pci0:0:29:0:      class=0x0c0320 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca6 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP USB EHCI Controller'
    class      = serial bus
    subclass   = USB
isab0@pci0:0:31:0:      class=0x060100 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9cc3 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP LPC Controller'
    class      = bridge
    subclass   = PCI-ISA
ahci0@pci0:0:31:2:      class=0x010601 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9c83 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
ichsmb0@pci0:0:31:3:    class=0x0c0500 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca2 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP SMBus Controller'
    class      = serial bus
    subclass   = SMBus
pchtherm0@pci0:0:31:6:  class=0x118000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca4 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP Thermal Management Controller'
    class      = dasp
rtsx0@pci0:2:0:0:       class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x5227 subvendor=0x17aa subdevice=0x2223
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS5227 PCI Express Card Reader'
iwm0@pci0:3:0:0:        class=0x028000 rev=0x59 hdr=0x00 vendor=0x8086 device=0x095b subvendor=0x8086 subdevice=0x5210
    vendor     = 'Intel Corporation'
    device     = 'Wireless 7265'
    class      = network
vgapci1@pci0:8:0:0:     class=0x030200 rev=0xa2 hdr=0x00 vendor=0x10de device=0x137a subvendor=0x17aa subdevice=0x2225
    vendor     = 'NVIDIA Corporation'
    device     = 'GM108GLM [Quadro K620M / Quadro M500M]'
    class      = display
    subclass   = 3D
```

</details>

**DRM KMOD version**
`drm-devel-kmod 5.9.g20220318`, from ae722b137024d7cf240471836d48de6067aa87b6

**To Reproduce**
`kldload drm`

**Screenshots**
N/A

**Additional context**
`core.txt` incoming?

vishwin, 2022-03-21:
<details>
<summary>core.txt</summary>

```
ardmore dumped core - see /var/crash/vmcore.7

Mon 21 Mar 2022 01:37:04 EDT

FreeBSD ardmore 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n253876-1e9ce60a6d7: Mon Mar 21 01:15:30 EDT 2022     root@ardmore:/usr/obj/usr/src/amd64.amd64/sys/ARDMORE-DEBUG  amd64

panic: mutex Giant not owned at /usr/src/sys/kern/subr_bus.c:874

GNU gdb (GDB) 11.2 [GDB v11.2 for FreeBSD]
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-portbld-freebsd14.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /boot/kernel/kernel...
Reading symbols from /usr/lib/debug//boot/kernel/kernel.debug...

Unread portion of the kernel message buffer:
---<<BOOT>>---
Copyright (c) 1992-2022 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 14.0-CURRENT #0 main-n253876-1e9ce60a6d7: Mon Mar 21 01:15:30 EDT 2022
    root@ardmore:/usr/obj/usr/src/amd64.amd64/sys/ARDMORE-DEBUG amd64
FreeBSD clang version 13.0.0 (git@github.com:llvm/llvm-project.git llvmorg-13.0.0-0-gd7b669b3a303)
WARNING: WITNESS option enabled, expect reduced performance.
VT(efifb): resolution 2880x1620
CPU: Intel(R) Core(TM) i7-5500U CPU @ 2.40GHz (2394.59-MHz K8-class CPU)
  Origin="GenuineIntel"  Id=0x306d4  Family=0x6  Model=0x3d  Stepping=4
  Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
  Features2=0x7ffafbbf<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,EST,TM2,SSSE3,SDBG,FMA,CX16,xTPR,PDCM,PCID,SSE4.1,SSE4.2,x2APIC,MOVBE,POPCNT,TSCDLT,AESNI,XSAVE,OSXSAVE,AVX,F16C,RDRAND>
  AMD Features=0x2c100800<SYSCALL,NX,Page1GB,RDTSCP,LM>
  AMD Features2=0x121<LAHF,ABM,Prefetch>
  Structured Extended Features=0x21c27ab<FSGSBASE,TSCADJ,BMI1,AVX2,SMEP,BMI2,ERMS,INVPCID,NFPUSG,RDSEED,ADX,SMAP,PROCTRACE>
  Structured Extended Features3=0x9c000600<MCUOPT,MD_CLEAR,IBPB,STIBP,L1DFL,SSBD>
  XSAVE Features=0x1<XSAVEOPT>
  VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID
  TSC: P-state invariant, performance statistics
real memory  = 8589934592 (8192 MB)
avail memory = 8144629760 (7767 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <LENOVO TP-N11  >
FreeBSD/SMP: Multiprocessor System Detected: 4 CPUs
FreeBSD/SMP: 1 package(s) x 2 core(s) x 2 hardware threads
random: registering fast source Intel Secure Key RNG
random: fast provider: "Intel Secure Key RNG"
random: unblocking device.
Security policy loaded: MAC/ntpd (mac_ntpd)
ioapic0 <Version 2.0> irqs 0-39
Launching APs: 1 2 3
TCP Hpts created 4 swi interrupt threads and bound 0 to cpus
HPTS is in INVARIANT mode!!
Cuse v0.1.36 @ /dev/cuse
random: entropy device external interface
kbd1 at kbdmux0
efirtc0: <EFI Realtime Clock>
[... 1979 lines cut ...]
device	uether
device	urndis
device	lagg
device	uhid
device	ums
device	snd_uaudio
device	pf
device	pflog
device	smbus
device	smb
device	ichsmb
device	pchtherm
device	acpi_wmi
device	nmdm
device	ucom
device	uplcom
device	uslcom
device	backlight

------------------------------------------------------------------------
ddb capture buffer

```

</details>

evadot, 2022-03-21:
I can't reproduce here on apollolake and skylake, and based on the panic message I don't see how it can comes from DRM (nor linuxkpi we just call bus_generic_attach)

vishwin, 2022-05-15:
Still happening on 5.10. No such panic on 5.4 (or the previous 5.5). Looks like the difference maker is lkpi I2C stuff

evadot, 2022-06-06:
I've changed a bit some stuff in linux_i2c recently, might worth the try.

vishwin, 2022-06-07:
Same panic as of freebsd/freebsd-src@cd7e11f78de (next commit is D35403 and its `__FreeBSD_version` bump, but holding off due to resulting expensive rebuild), now with additional VBIOS error message:
```
<118>Loading kernel modules:
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
<5>[drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
<6>[drm] Got stolen memory base 0x9e000000, size 0x2000000
drmn0: [drm] Failed to find VBIOS tables (VBT)
lkpi_iic0: <LinuxKPI I2C> on drmn0
iicbus0: <Philips I2C bus> on lkpi_iic0
iic0: <I2C generic I/O> on iicbus0
lkpi_iic1: <LinuxKPI I2C> on drmn0
iicbus1: <Philips I2C bus> on lkpi_iic1
iic1: <I2C generic I/O> on iicbus1
lkpi_iic2: <LinuxKPI I2C> on drmn0
iicbus2: <Philips I2C bus> on lkpi_iic2
iic2: <I2C generic I/O> on iicbus2
lkpi_iic3: <LinuxKPI I2C> on drmn0
iicbus3: <Philips I2C bus> on lkpi_iic3
iic3: <I2C generic I/O> on iicbus3
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
lkpi_iic4: <LinuxKPI I2C> on drm2
iicbus4: <Philips I2C bus> on lkpi_iic4
iic4: <I2C generic I/O> on iicbus4
lkpi_iic5: <LinuxKPI I2C> on drm3
iicbus5: <Philips I2C bus> on lkpi_iic5
iic5: <I2C generic I/O> on iicbus5
lkpi_iic6: <LinuxKPI I2C> on drm5
iicbus6: <Philips I2C bus> on lkpi_iic6
iic6: <I2C generic I/O> on iicbus6
<6>[drm] Initialized i915 1.6.0 20200917 for drmn0 on minor 0
panic: mutex Giant not owned at /usr/src/sys/kern/subr_bus.c:874
cpuid = 1
time = 1654618030
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe00852aea50
vpanic() at vpanic+0x151/frame 0xfffffe00852aeaa0
panic() at panic+0x43/frame 0xfffffe00852aeb00
__mtx_assert() at __mtx_assert+0xb9/frame 0xfffffe00852aeb10
device_probe_and_attach() at device_probe_and_attach+0x2a/frame 0xfffffe00852aeb40
bus_generic_attach() at bus_generic_attach+0x18/frame 0xfffffe00852aeb60
lkpi_i2c_add_adapter() at lkpi_i2c_add_adapter+0x7f/frame 0xfffffe00852aeb80
drm_dp_port_set_pdt() at drm_dp_port_set_pdt+0x1f8/frame 0xfffffe00852aebd0
drm_dp_send_link_address() at drm_dp_send_link_address+0x89a/frame 0xfffffe00852aed00
drm_dp_check_and_send_link_address() at drm_dp_check_and_send_link_address+0x27/frame 0xfffffe00852aed40
drm_dp_mst_link_probe_work() at drm_dp_mst_link_probe_work+0x271/frame 0xfffffe00852aedf0
linux_work_fn() at linux_work_fn+0xe8/frame 0xfffffe00852aee40
taskqueue_run_locked() at taskqueue_run_locked+0xb0/frame 0xfffffe00852aeec0
taskqueue_thread_loop() at taskqueue_thread_loop+0xd2/frame 0xfffffe00852aeef0
fork_exit() at fork_exit+0x85/frame 0xfffffe00852aef30
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe00852aef30
--- trap 0x268a3e4e, rip = 0x34e3db98054fb880, rsp = 0x2cefc3940f63a08c, rbp = 0xf4b33282bee1a627 ---
Uptime: 20s
Dumping 526 out of 8060 MB:..4%..13%..22%..31%..43%..52%..61%..73%..83%..92%

__curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
55		__asm("movq %%gs:%P1,%0" : "=r" (td) : "n" (offsetof(struct pcpu,
(kgdb) #0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
#1  dump_savectx () at /usr/src/sys/kern/kern_shutdown.c:401
#2  0xffffffff80aaf6e5 in dumpsys (di=0x0)
    at /usr/src/sys/x86/include/dump.h:87
#3  doadump (textdump=1) at /usr/src/sys/kern/kern_shutdown.c:430
#4  kern_reboot (howto=260) at /usr/src/sys/kern/kern_shutdown.c:537
#5  0xffffffff80aafb1e in vpanic (fmt=<optimized out>, 
    ap=ap@entry=0xfffffe00852aeae0) at /usr/src/sys/kern/kern_shutdown.c:975
#6  0xffffffff80aaf883 in panic (fmt=<unavailable>)
    at /usr/src/sys/kern/kern_shutdown.c:899
#7  0xffffffff80a8a729 in __mtx_assert (c=<optimized out>, 
    what=<optimized out>, file=<unavailable>, line=<unavailable>)
    at /usr/src/sys/kern/kern_mutex.c:1092
#8  0xffffffff80aec40a in bus_topo_assert ()
    at /usr/src/sys/kern/subr_bus.c:874
#9  device_probe_and_attach (dev=dev@entry=0xfffff80005b95100)
    at /usr/src/sys/kern/subr_bus.c:2984
#10 0xffffffff80aedac8 in bus_generic_attach (dev=<optimized out>)
    at /usr/src/sys/kern/subr_bus.c:3801
#11 0xffffffff80db41df in lkpi_i2c_add_adapter (adapter=0xfffff8000584cc38)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_i2c.c:224
#12 0xffffffff83501318 in drm_dp_port_set_pdt () from /boot/modules/drm.ko
#13 0xffffffff83500aea in drm_dp_send_link_address ()
   from /boot/modules/drm.ko
#14 0xffffffff83500187 in drm_dp_check_and_send_link_address ()
   from /boot/modules/drm.ko
#15 0xffffffff834fd931 in drm_dp_mst_link_probe_work ()
   from /boot/modules/drm.ko
#16 0xffffffff80dc2668 in linux_work_fn (context=0xfffff8000584cc38, 
    pending=<optimized out>)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_work.c:299
#17 0xffffffff80b15080 in taskqueue_run_locked (
    queue=queue@entry=0xfffff800011db100)
    at /usr/src/sys/kern/subr_taskqueue.c:514
#18 0xffffffff80b16142 in taskqueue_thread_loop (arg=<optimized out>)
    at /usr/src/sys/kern/subr_taskqueue.c:826
#19 0xffffffff80a664d5 in fork_exit (
    callout=0xffffffff80b16070 <taskqueue_thread_loop>, 
    arg=0xfffff80001651d40, frame=0xfffffe00852aef40)
    at /usr/src/sys/kern/kern_fork.c:1102
#20 <signal handler called>
#21 0x34e3db98054fb880 in ?? ()
Backtrace stopped: Cannot access memory at address 0x2cefc3940f63a08c
(kgdb) 
```

evadot, 2022-06-14:
Does https://reviews.freebsd.org/D35478 helps you ?
I honestly don't understand why it is not needed for me but ...

vishwin, 2022-06-17:
After D35478, so far so good with only the laptop screen. Will need to continue testing with multiple (high pixel density) monitor setup when I have access to it again in a couple days

vishwin, 2022-06-22:
Unfortunately the dual high pixel density screen setup resulted in a complete black screen hang. Input devices (keyboard included) stuck, hard reset needed. No log messages whatsoever this time.

Interestingly, both laptop screen only (which itself is a high pixel density screen) and laptop + standard pixel density monitor do not exhibit the hang. This setup does not seem to cause a framebuffer video mode change whereas dual high pixel density does.

evadot, 2022-06-22:
Can you share more info on your setup ? (Which screens are connected and how)

vishwin, 2022-06-22:
Built-in laptop screen (high pixel density) is eDP, high pixel density external monitor is DisplayPort, standard pixel density external monitor is VGA. All video I/O goes through Intel only

evadot, 2022-06-22:
And eDP + VGA work, edp + VGA + DP doesn't ?
What about eDP + DP ?

vishwin, 2022-06-22:
Don't have all three at my disposal at the same time, the problem configuration is eDP + DP.

evadot, 2022-06-23:
Does it hangs too if you plug the DP screen after loading the module ? Or do you have some panic that could help finding what's going on ?

vishwin, 2022-06-24:
The DP monitor lives on a docking station so I will see what happens when I hotplug that

vishwin, 2022-06-24:
No hang on hotplug.

evadot, 2022-06-24:
> No hang on hotplug.

No hang and work ? No hang and panic ?

vishwin, 2022-06-24:
No hang and works when hotplugged after loading the module.

vishwin, 2022-06-24:
Relevant output when hotplugged:
```
Jun 23 22:49:50 ardmore kernel: fbd0 on drmn0
Jun 23 22:49:50 ardmore kernel: device_attach: fbd0 attach returned 6
Jun 23 22:49:50 ardmore kernel: lkpi_iic7: <LinuxKPI I2C> on drmn0
Jun 23 22:49:50 ardmore kernel: iicbus7: <Philips I2C bus> on lkpi_iic7
Jun 23 22:49:50 ardmore kernel: iic7: <I2C generic I/O> on iicbus7
Jun 23 22:49:50 ardmore kernel: ugen0.4: <LENOVO Lenovo ThinkPad Dock> at usbus0
Jun 23 22:49:50 ardmore kernel: uhub3 on uhub0
Jun 23 22:49:50 ardmore kernel: uhub3: <LENOVO Lenovo ThinkPad Dock, class 9/0, rev 3.00/50.41, addr 3> on usbus0
Jun 23 22:49:50 ardmore kernel: uhub3: 4 ports with 3 removable, self powered
Jun 23 22:49:50 ardmore kernel: fbd0 on drmn0
Jun 23 22:49:50 ardmore kernel: device_attach: fbd0 attach returned 6
Jun 23 22:49:50 ardmore kernel: lkpi_iic8: <LinuxKPI I2C> on drmn0
Jun 23 22:49:50 ardmore kernel: iicbus8: <Philips I2C bus> on lkpi_iic8
Jun 23 22:49:50 ardmore kernel: iic8: <I2C generic I/O> on iicbus8
```

evadot, 2022-06-29:
> 

Not sure I understand, if you load it from single user mode it works but if you load it in multi user (either by hand or via kld_list) it hang ?

vishwin, 2022-07-02:
Fully works when loaded via `kld_list` now, not sure why it didn't before

closed 2022-07-02 by vishwin