freebsd/drm-kmod #155: AMD Ryzen 7 5800U support

ivan-volnov, 2022-03-27:
**Describe the bug**
Hi there!

The new Ryzen 7 5800U doesn't work on 13.0-RELEASE with drm-fbsd13-kmod. Hope I will be fixed on 13.1-RELEASE.

Can help with testing and patching but don't want to upgrade the whole system.

**FreeBSD version**

```
FreeBSD hostname 13.0-RELEASE-p8 FreeBSD 13.0-RELEASE-p8 #0: Tue Mar 15 09:36:28 UTC 2022     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64
```

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0: class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x1022 subdevice=0x1630
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2: class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x1022 subdevice=0x1631
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0: class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1: class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1633 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:3: class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0: class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:2:1: class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:2:2: class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:2:3: class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0: class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib6@pci0:0:8:1: class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
[... 114 lines cut ...]
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
```
</details>

**DRM KMOD version**
```
drm-fbsd13-kmod 5.4.144.g20220128
drm-kmod g20190710_1
```

**To Reproduce**
Try to launch a wayland DM dwl:

```
❯ dwl
00:00:10.505 [backend/backend.c:217] Found 0 GPUs, cannot create backend
00:00:10.505 [backend/backend.c:386] Failed to open any DRM device
couldn't create backend
```

**Additional context**

Found [here](https://forums.freebsd.org/threads/freebsd-13-1-in-beta.84389/page-3#post-560594) a tread about the Cezanne core cpu. They say that drm-devel-kmod can be built on 13.1-BETA1, but Xorg 1.20.14 (also 1.20.13) crashes while accessing amdgpu.

pointcheck, 2022-04-13:
Xorg 1.20.14 still does not work on Cezenne CPUs, it crashes while, presumably, accessing AMDGPU. Also GPU offloading to NVIDIA does not work, I fail to make it detect more than one providers. 

Using FreeBSD 13.1-RC1, drm-devel-kmod 5.7.19.g20220223, NVIDIA driver 510.60.02.

[drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x17AA:0x3A5D 0xC6).
nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  510.60.02  Wed Mar 16 11:03:12 UTC 2022

ivan-volnov, 2022-04-29:
Updated to FreeBSD 13.5-RC1. And added a second monitor via DP.

I experience reboots on the DP monitor start.

- On driver load during reboot
- On wayland session create
- On wake up from sleep

Some of dmesg hope it helps:

```
VT: Replacing driver "efifb" with new "fb".
start FB_INFO:
type=11 height=1440 width=2560 depth=32
pbase=0x1010c8d000 vbase=0xfffff81010c8d000
name=drmn0 flags=0x0 stride=10240 bpp=32
end FB_INFO
drmn0: ring gfx uses VM inv eng 0 on hub 0
drmn0: ring comp_1.0.0 uses VM inv eng 1 on hub 0
drmn0: ring comp_1.1.0 uses VM inv eng 4 on hub 0
drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
drmn0: ring sdma0 uses VM inv eng 0 on hub 1
drmn0: ring vcn_dec uses VM inv eng 1 on hub 1
drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 1
drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 1
drmn0: ring jpeg_dec uses VM inv eng 6 on hub 1
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] Initialized amdgpu 3.37.0 20150101 for drmn0 on minor 0
acpi_wmi0: <ACPI-WMI mapping> on acpi0
acpi_wmi0: cannot find EC device
acpi_wmi0: Embedded MOF found
ACPI: \134AOD.WQBA: 1 arguments were passed to a non-method ACPI object (Buffer) (20201113/nsarguments-361)
driver bug: Unable to set devclass (class: ppc devname: (unknown))
intsmb0: <AMD FCH SMBus Controller> at device 20.0 on pci0
smbus0: <System Management Bus> on intsmb0
lo0: link state changed to UP
re0: link state changed to DOWN
uhid0 on uhub1
uhid0: <vendor 0x0c45 FC660M, class 0/0, rev 2.00/0.01, addr 1> on usbus1
ums0 on uhub1
ums0: <ROCCAT ROCCAT Kone Pure Military, class 0/0, rev 2.00/2.00, addr 2> on usbus1
ums0: 5 buttons and [XYZT] coordinates ID=1
pflog0: promiscuous mode enabled
re0: link state changed to UP
Security policy loaded: MAC/ntpd (mac_ntpd)
[drm] Fence fallback timer expired on ring gfx
[drm] Fence fallback timer expired on ring sdma0
[drm ERROR :drm_atomic_helper_wait_for_flip_done] [CRTC:62:crtc-3] flip_done timed out
[drm ERROR :drm_atomic_helper_wait_for_flip_done] [CRTC:62:crtc-3] flip_done timed out
WARNING atomic_read(&vblank->refcount) == 0 failed at /usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.7.19_3/drivers/gpu/drm/drm_vblank.c:1143
[drm] Fence fallback timer expired on ring gfx
[drm] Fence fallback timer expired on ring gfx
```

xmhd, 2022-05-04:
I have a 5850u (which is the same cpu, just the 'pro' version) and it is working nicely (with gpu acceleration) on 14.0-CURRENT from April 7th 2022 along with drm-devel-kmod.

This morning I tried 13.1-RC5 and drm-510-kmod, which unfortunately do not work. I am able to 'kldload amdgpu' which compared to the same command on the aforementioned working setup, seems to work-ish, and when running 'startx' afterwards, the xorg server does not start.

I will provide more information later today when I am off work.

xmhd, 2022-05-06:
so I done some digging and found the following missing pieces:

- firmware for 5000 series/cezanne is currently missing, see: [https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=263793](url)
- however, as cezanne is a modified renoir, they work (mostly) with renoir firmware ...
- drm-devel-kmod (v5.7) which works for my 5850u has the following commit adding the relevant device ID's [https://github.com/freebsd/drm-kmod/commit/44a2eaf653479067ddee2c2db79e68993ec5b4b8](url)
- v5.4 and v5.4 do not have these device ID's added

I will test this out and report back.

xmhd, 2022-05-06:
I will try that @wulf7 thanks for the advice

pointcheck, 2022-05-06:
Hi all.

I have just tried 5.10-lts, it does not load on Cezanne CPU at all, reports 'amdgpu/green_sardine_sdma.bin' is missing, see my dmesg output below. So, 5.7-stable branch is the only that loads but still does not work - Xorg crashes at AMDGPU access. 

I have been investigating this issue for quite some time and came to conclusion that the problem is not in DRM driver, neither in Xorg, but in /lib/libthr.so.3 or even in /lib/libc.so.7 library which comes with FreeBSD/LLVM. In other words it is a bug in FreeBSD.

drmn1: <drmn> on vgapci1
vgapci1: child drmn1 requested pci_enable_io
vgapci1: child drmn1 requested pci_enable_io
[drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x17AA:0x3A5D 0xC6).
drmn1: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
[drm] register mmio base: 0xD1400000
[drm] register mmio size: 524288
[drm] add ip block number 0 <soc15_common>
[drm] add ip block number 1 <gmc_v9_0>
[drm] add ip block number 2 <vega10_ih>
[drm] add ip block number 3 <psp>
[drm] add ip block number 4 <smu>
[drm] add ip block number 5 <gfx_v9_0>
[drm] add ip block number 6 <sdma_v4_0>
[drm] add ip block number 7 <dm>
[drm] add ip block number 8 <vcn_v2_0>
[drm] add ip block number 9 <jpeg_v2_0>
drmn1: Fetched VBIOS from VFCT
drmn1: could not load firmware image 'amdgpu/green_sardine_sdma.bin'
[drm ERROR :sdma_v4_0_init_microcode] sdma_v4_0: Failed to load firmware "amdgpu/green_sardine_sdma.bin"
[drm ERROR :sdma_v4_0_early_init] Failed to load sdma firmware!
[drm] VCN decode is enabled in VM mode
[drm] VCN encode is enabled in VM mode
[drm] JPEG decode is enabled in VM mode
[drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
drmn1: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
drmn1: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF
drmn1: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
[drm] Detected VRAM RAM=4096M, BAR=4096M
[drm] RAM width 128bits DDR4
[drm] amdgpu: 4096M of VRAM memory ready
[drm] amdgpu: 4096M of GTT memory ready.
[drm] GART: num cpu pages 262144, num gpu pages 262144
[drm] PCIE GART of 1024M enabled (table at 0x000000F400900000).
drmn1: could not load firmware image 'amdgpu/green_sardine_asd.bin'
drmn1: fail to initialize asd microcode
[drm ERROR :psp_sw_init] Failed to load psp firmware!
[drm ERROR :amdgpu_device_ip_init] sw_init of IP block <psp> failed -2
drmn1: amdgpu_device_ip_init failed
drmn1: Fatal error during GPU init
drmn1: amdgpu: finishing device.
[drm] amdgpu: ttm finalized
device_attach: drmn1 attach returned 2

xmhd, 2022-05-06:
> Hi all.
> 
> I have just tried 5.10-lts, it does not load on Cezanne CPU at all, reports 'amdgpu/green_sardine_sdma.bin' is missing, see my dmesg output below. So, 5.7-stable branch is the only that loads but still does not work - Xorg crashes at AMDGPU access.
> 
> I have been investigating this issue for quite some time and came to conclusion that the problem is not in DRM driver, neither in Xorg, but in /lib/libthr.so.3 or even in /lib/libc.so.7 library which comes with FreeBSD/LLVM. In other words it is a bug in FreeBSD.
> 
> drmn1: on vgapci1 vgapci1: child drmn1 requested pci_enable_io vgapci1: child drmn1 requested pci_enable_io [drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x17AA:0x3A5D 0xC6). drmn1: Trusted Memory Zone (TMZ) feature disabled as experimental (default) [drm] register mmio base: 0xD1400000 [drm] register mmio size: 524288 [drm] add ip block number 0 <soc15_common> [drm] add ip block number 1 <gmc_v9_0> [drm] add ip block number 2 <vega10_ih> [drm] add ip block number 3 [drm] add ip block number 4 [drm] add ip block number 5 <gfx_v9_0> [drm] add ip block number 6 <sdma_v4_0> [drm] add ip block number 7 [drm] add ip block number 8 <vcn_v2_0> [drm] add ip block number 9 <jpeg_v2_0> drmn1: Fetched VBIOS from VFCT drmn1: could not load firmware image 'amdgpu/green_sardine_sdma.bin' [drm ERROR :sdma_v4_0_init_microcode] sdma_v4_0: Failed to load firmware "amdgpu/green_sardine_sdma.bin" [drm ERROR :sdma_v4_0_early_init] Failed to load sdma firmware! [drm] VCN decode is enabled in VM mode [drm] VCN encode is enabled in VM mode [drm] JPEG decode is enabled in VM mode [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit drmn1: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used) drmn1: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF drmn1: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF [drm] Detected VRAM RAM=4096M, BAR=4096M [drm] RAM width 128bits DDR4 [drm] amdgpu: 4096M of VRAM memory ready [drm] amdgpu: 4096M of GTT memory ready. [drm] GART: num cpu pages 262144, num gpu pages 262144 [drm] PCIE GART of 1024M enabled (table at 0x000000F400900000). drmn1: could not load firmware image 'amdgpu/green_sardine_asd.bin' drmn1: fail to initialize asd microcode [drm ERROR :psp_sw_init] Failed to load psp firmware! [drm ERROR :amdgpu_device_ip_init] sw_init of IP block failed -2 drmn1: amdgpu_device_ip_init failed drmn1: Fatal error during GPU init drmn1: amdgpu: finishing device. [drm] amdgpu: ttm finalized device_attach: drmn1 attach returned 2

Yes, even with the green sardine firmware missing the driver should still work (to a certain extent) with the renoir firmware as it had been doing so with the 5.7 branch.

I had last tested 5.7 branch with 14.0-CURRENT with the April 6th release of install media. So if it is broken in FreeBSD as you say, the change has happened somewhere in that time.

xmhd, 2022-05-06:
> Yes, even with the green sardine firmware missing the driver should still work (to a certain extent) with the renoir firmware as it had been doing so with the 5.7 branch.
>
> I had last tested 5.7 [...]

I had the same experience as you with 5.10-lts.

pointcheck, 2022-05-06:
Can you please show me your Xorg.0.log as well as /etc/X11/xorg.conf ? Thanks.

I have installed missing green_sardine firmware manually from /usr/ports/graphics/gpu-firmware-kmod/ (strangely it's just missing in Makefile), now 5.10-lts driver loads ok, modesetting works, yet same issue with Xorg using amdgpu.

xmhd, 2022-05-06:
> Can you please show me your Xorg.0.log as well as /etc/X11/xorg.conf ? Thanks.
>
> I have installed missing green_sardine firmware manually from /usr/ports/graphics/gpu-firmware-kmod/ (strangely it's [...]

I can try that tonight. If you don't mind, can you please provide the changes that you made?

I was looking at adding the firmware files to gpu-firmware-amd-kmod which doesn't look too difficult.

pointcheck, 2022-05-06:
I did not do any changes. I just installed missing firmware from their resting directories:

# cd /usr/ports/graphics/gpu-firmware-kmod && make install

# ls -d /usr/ports/graphics/gpu-firmware-kmod/work/drm-kmod-firmware-*/amdgpukmsfw/green_sardine_* | xargs -I % /bin/sh -c '(cd %; make install)'

xmhd, 2022-05-07:
> I did not do any changes. I just installed missing firmware from their resting directories:
>
> ```
> # cd /usr/ports/graphics/gpu-firmware-kmod && make install
>
> # ls -d [...]

I wasn't able to replicate an install of green_sardine_* firmware files with this method, sorry. While I have some experience with the amdgpu driver, my experience with drm-kmod + gpu-firmware-kmod is limited :-)

I would imagine that xf86-video-amdgpu being stuck at version 19.x is the reason behind the crashing that you are experiencing.

ivan-volnov, 2022-05-07:
Hi there!

Can anybody add a pull request to https://github.com/freebsd/drm-kmod-firmware to restore green_sardine support?

I opened an issue https://github.com/freebsd/drm-kmod-firmware/issues/21 for that.

xmhd, 2022-05-07:
> Hi there!
> 
> Can anybody add a pull request to https://github.com/freebsd/drm-kmod-firmware to restore green_sardine support?
> 
> I opened an issue [freebsd/drm-kmod-firmware#21](https://github.com/freebsd/drm-kmod-firmware/issues/21) for that.

I have done so. As mentioned in my PR, the next piece to be added is the relevant bits in the Makefile for the gpu-firmware-kmod port.

pointcheck, 2022-05-08:
> I would imagine that xf86-video-amdgpu being stuck at version 19.x is the reason behind the crashing that you are experiencing.

As I said I begin thinking that the problem is not in Xorg/xf86-video-amdgpu per se, but in libc library used in FreeBSD 13. I would like to see if this same issue persists in FreeBSD 14-CURRENT. Can you please send me your Xorg.0.log for investigation ?

xmhd, 2022-05-08:
> As I said I begin thinking that the problem is not in Xorg/xf86-video-amdgpu per se, but in libc library used in FreeBSD 13. I would like to see if this same issue persists in FreeBSD 14-CURRENT. [...]

Ok now I see what you're talking about. I tried out the 14.0-CURRENT image from April 14th 2022 and now I experience the same problem, even with v5.7 in use. So the breakage occured somewhere between April 7th and April 14th.

With regards to xorg.conf - I don't change any defaults. The only 'tweaks' I make is to /usr/local/share/X11/xorg.conf.d/10-amdgpu.conf where I enable TearFree for no screen tearing.

ivan-volnov, 2022-05-17:
The firmware is available now.

Updated to 13.1-RELEASE, installed gpu-firmware-amd-kmod-green-sardine and drm-510-kmod. No luck. Described the issue here #167 

Can anybody test it also?

evadot, 2022-05-20:
Closing this issue, #167 will track the problem now that we (should) have everything to support this GPU generation.

closed 2022-05-20 by evadot