freebsd/drm-kmod #156: Broken X server on resume from sleep FreeBSD 13.1RC2

dr3mro, 2022-04-08:
**Describe the bug**
Whenever I do zzz on my Lenovo Thinkpad x250 using freeBSD 13.1RC2 the Xserver crash then I have to either do zzz again or do kldunload i915kms && kldload i915kms then It will work fine until I do zzz again
this issue happens on lid close and was happening on RC1 when using picom GLX backend but now on RC2 it happens even if no applications are running and picom not running at all or running with xrender backend which was working on RC1

**FreeBSD version**
FreeBSD x250 13.1-RC2 FreeBSD 13.1-RC2 releng/13.1-n250104-c2e15a08fd9 GENERIC amd64

**PCI Info**

```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x1604 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Broadwell-U Host Bridge -OPI'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x1616 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics 5500'
    class      = display
    subclass   = VGA
hdac0@pci0:0:3:0:	class=0x040300 rev=0x09 hdr=0x00 vendor=0x8086 device=0x160c subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Broadwell-U Audio Controller'
    class      = multimedia
    subclass   = HDA
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9cb1 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP USB xHCI Controller'
    class      = serial bus
    subclass   = USB
none0@pci0:0:22:0:	class=0x078000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9cba subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP MEI Controller'
    class      = simple comms
em0@pci0:0:25:0:	class=0x020000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x15a2 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (3) I218-LM'
    class      = network
    subclass   = ethernet
hdac1@pci0:0:27:0:	class=0x040300 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca0 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
pcib1@pci0:0:28:0:	class=0x060400 rev=0xe3 hdr=0x01 vendor=0x8086 device=0x9c9a subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:1:	class=0x060400 rev=0xe3 hdr=0x01 vendor=0x8086 device=0x9c94 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
ehci0@pci0:0:29:0:	class=0x0c0320 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca6 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP USB EHCI Controller'
    class      = serial bus
    subclass   = USB
isab0@pci0:0:31:0:	class=0x060100 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9cc3 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP LPC Controller'
    class      = bridge
    subclass   = PCI-ISA
ahci0@pci0:0:31:2:	class=0x010601 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9c83 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
ichsmb0@pci0:0:31:3:	class=0x0c0500 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca2 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP SMBus Controller'
    class      = serial bus
    subclass   = SMBus
pchtherm0@pci0:0:31:6:	class=0x118000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x9ca4 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Intel Corporation'
    device     = 'Wildcat Point-LP Thermal Management Controller'
    class      = dasp
rtsx0@pci0:2:0:0:	class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x5227 subvendor=0x17aa subdevice=0x2226
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS5227 PCI Express Card Reader'
iwm0@pci0:3:0:0:	class=0x028000 rev=0x59 hdr=0x00 vendor=0x8086 device=0x095b subvendor=0x8086 subdevice=0x5210
    vendor     = 'Intel Corporation'
    device     = 'Wireless 7265'
    class      = network
```
**DRM KMOD version**
drm-fbsd13-kmod 5.4.144.g20220223
drm-kmod g20190710_1

**To Reproduce**
on 13.1Rc1 enabling picom or compton with glx backend 
on 13.1Rc2 independent of picom backend even if uninstalled
just enter sleep then wake up then x server crash

xorg.0.log
[Xorg.0.log](https://github.com/freebsd/drm-kmod/files/8454076/Xorg.0.log)

dr3mro, 2022-04-08:
I have found a work around which works even if PICOM with GLX backend is running

I have to switch VT to tty2 for example and login then zzz 

then it will resume to this tty and then I switch back with ctrl+alt+f9 then no crash 

may be it tries to switch to the tty too early before its ready!

evadot, 2022-05-09:
drm-fbsd13-kmod (which tracks 5.4) is known to have suspend/resume problems.
Try graphics/drm-510-kmod which compiles on 13.1

dr3mro, 2022-05-09:
I can confirm that fixes the issue for me

closed 2022-06-06 by evadot