freebsd/drm-kmod #158: Wiki request

thisplacestinksoffascism, 2022-04-11:
Hello,

Is it possible to get a basic Wiki page going? I am after: instructions on how to build drm-kmod master, or any branch of the git repository; links to appropriate Linux forums and code bases (I have no idea where to find the Linux development of DRM); any papers/articles on DRM hacking (I remember a TTM paper from way back, hopefullly documentation has improved since then); how to setup and configure a host to debug DRM code; how to configure and setup a host to dump and reboot on panic.

Some of this stuff could just point to the FreeBSD Handbook too. Ideally, this information would aim to provide the means for users to setup their host follow drm-kmod master, and provide more feedback, it would also provide a base to start from for would be kernel GPU hackers (who would want that really? :-D)

I would be happy to contribute to some of those pages if possible.

Thanks.

grahamperrin, 2022-05-02:
<https://wiki.freebsd.org/Graphics> ▶ drm-kmod

Another part of the page refers to `graphics/drm-devel-kmod`, this is outdated. 

> … no idea where to find the Linux development of DRM); …

Cross-reference: <https://old.reddit.com/r/freebsd/comments/ug8pot/-/i6zx1bp/?context=1>

> … how to build drm-kmod master, or any branch of the git repository; …

Rough notes to self: <https://github.com/freebsd/drm-kmod/issues/93#issuecomment-962622626>

thisplacestinksoffascism, 2022-05-04:
> https://wiki.freebsd.org/Graphics arrow_forward drm-kmod
> 
> Another part of the page refers to `graphics/drm-devel-kmod`, this is outdated.
> 
> > … no idea where to find the Linux development of DRM); …
> 
> Cross-reference: https://old.reddit.com/r/freebsd/comments/ug8pot/-/i6zx1bp/?context=1
> 
> > … how to build drm-kmod master, or any branch of the git repository; …
> 
> Rough notes to self: [#93 (comment)](https://github.com/freebsd/drm-kmod/issues/93#issuecomment-962622626)

Thank you so much.

evadot, 2022-05-20:
The FreeBSD wiki should be enough (even if ideally everything should be in the handbook).

closed 2022-05-20 by evadot