freebsd/drm-kmod #159: loading i915kms in ThinkPad X13 Intel Gen2 will hang up.

takawata, 2022-04-16:
**Describe the bug**
loading i915kms gets screen blackout. Try it from remote login, it probed that the initialization routine is hanging and did not return from kldload(8). 
typing ^Z did not return to prompt, and ^T shows
`
load: 0x43 cmd:kldload 1762 [running] 8.54r 0.00u 8.50s 58% 2296k
intel_ddi_compute_config+0x3f,intel_atomic_check+0x65a, drm_atomic_check_only+0x400,
drm_atomic_commit+0x13 intel_modeset_init+0x1e28 i915_driver_probe+0xe39 i915_probe+0x42 linux_pci_attach_device+0x43b device_attach+0x3c3 bus_generic_driver_added+0xa1 devclass_driver_added+0x39 devclass_add_driver+0x13d _linux_pci_register_driver+0xcc i915kms_evh+0x39 module_register_init+0xa4 linker_load_module+0xbad kern_kldload=0x16a sys_kldload+0x5b
`
**FreeBSD version**
Paste the output of `uname -a`
FreeBSD 14.0-CURRENT #12 takawata-wip-n254513-7a473c2ec3b9-dirty: Fri Apr  8 22:
37:25 JST 2022

**PCI Info**
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a49 subvendor=0x17aa subdevice=0x22d1
    vendor     = 'Intel Corporation'
    device     = 'TigerLake-LP GT2 [Iris Xe Graphics]'
    class      = display
    subclass   = VGA
**DRM KMOD version**
master

**To Reproduce**
loading i915kms.ko

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

takawata, 2022-04-18:
The dmesg when the module is loaded is as  follows.
> iic0: <I2C generic I/O> on iicbus0
iic1: <I2C generic I/O> on iicbus1
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
[drm] Got stolen memory base 0x0, size 0x0
[drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
lkpi_iic0: <LinuxKPI I2C> on drmn0
iicbus2: <Philips I2C bus> on lkpi_iic0
iic2: <I2C generic I/O> on iicbus2
lkpi_iic1: <LinuxKPI I2C> on drmn0
iicbus3: <Philips I2C bus> on lkpi_iic1
iic3: <I2C generic I/O> on iicbus3
lkpi_iic2: <LinuxKPI I2C> on drmn0

evadot, 2022-04-19:
Did it worked before ?
If yes could you bisect ?
Thanks

takawata, 2022-04-19:
Never. I tried drm-kmod-current ports and did not work , then I try the master .

evadot, 2022-04-19:
drmn drmn0: drm_WARN_ON(common_len <= 0) seems to be displayport related.
Can you try without external screen/dock etc ... ?

takawata, 2022-04-19:
I don't connect anything but ethernet cable when I got the message.

takawata, 2022-04-24:
Ignoreing DP initialization after common_len <= 0 then module loading success and seems to  X work, though non-X console resolution did not change as usual.

    diff --git a/drivers/gpu/drm/i915/display/intel_dp.c 
    b/drivers/gpu/drm/i915/display/intel_dp.c
    
    index 70831e738f..7ef06791e8 100644
    --- a/drivers/gpu/drm/i915/display/intel_dp.c
    +++ b/drivers/gpu/drm/i915/display/intel_dp.c
    @@ -2355,7 +2355,10 @@ intel_dp_compute_link_config(struct intel_encoder *encoder,
            common_len = intel_dp_common_len_rate_limit(intel_dp,
                                                       intel_dp->max_link_rate);
    -
    +       if(common_len <= 0)
    +       {
    +               return -EINVAL;
    +       }
           /* No common link rates between source and sink */        　　　　 
    drm_WARN_ON(encoder->base.dev, common_len <= 0);

[Xorg.0.log](https://github.com/freebsd/drm-kmod/files/8549135/Xorg.0.log)

takawata, 2023-01-09:
drm-510-kmod still hang up, This patch is still needed, it seems.

evadot, 2024-01-05:
Using 5.15-lts or 6.1-lts will fix this.

closed 2024-01-05 by evadot