freebsd/drm-kmod #161: (Thinkpad X13) AMD R7-4750U with drm-devel-kmod can't boot Gnome

ykla, 2022-04-25:
**Describe the bug**
Using drm-devel-kmod in FreeBSD 14, I can boot XFCE and KDE5 normal. But it can't boot Gnome 3.36 or 41. Launching gnome gets stuck in gdm, as does using other window managers. Using the command startx does not work either. Logging in with startx only shows a mouse and is not clickable.

**FreeBSD version**
FreeBSD mybsd 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n254961-b91a48693a5: Thu Apr 21 07:34:26 UTC 2022    

**PCI Info**

<details>
  <summary>pciconf -lv</summary>

root@mybsd:/home/freebsd # pciconf -lv
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:2:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:2:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:2:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:2:7:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib6@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x5082 subdevice=0x17aa
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
[... 120 lines cut ...]
    class      = multimedia
    subclass   = HDA
root@mybsd:/home/freebsd # 

</details>

**DRM KMOD version**
drm-devel-kmod 5.7.19.g20220223

**To Reproduce**
Install drm-devel-kmod, xorg and gnome, configuration rc.conf and fstab then reboot.

**Screenshots**
[image]
[image]
[image]

**Additional context**
I aslo trid drm-fbsd13-kmod in FreeBSD 13.1 rc4. But boot stop at kernel.

```
#cat  /boot/loader.conf
hw.amdgpu.exp_hw_support="1"
```
[image]

evadot, 2022-05-19:
Can you test again with drm-510-kmod please ?

ykla, 2022-05-20:
> 

drm-510-kmod also can't startx.

[image]

So I install gpu-firmware-kmod and xf86-video-amdgpu, but also can't startx.

[image]

ykla, 2022-06-01:
All works. Thanks.

>This part of the tutorial has been tested for renoir graphics cards.

To install the required drivers (all latest sources or from ports).

```
# pkg install drm-510-kmod gpu-firmware-kmod xf86-video-amdgpu
```

Create a new and edit xorg configuration file.
```
# ee /usr/local/etc/X11/xorg.conf.d/06-driver.conf
```

Add (**Note that the BusID should be changed to your own, use `pciconf -l` to see it**).

```
Section "Device"
    Option "DRI" "3"
    Option "AccelMethod" "exa"
    Option "MigrationHeuristic" "greedy
    Option "TearFree" "On"
    Driver "amdgpu"
    BusID "PCI:6:0:0"
```

closed 2022-06-01 by ykla