freebsd/drm-kmod #162: graphics/drm-510-kmod panic when loading amdgpu.ko

jownit, 2022-05-03:
**Describe the bug**
After installing graphics/drm-510-kmod the kernel panics when loading amdgpu.ko

**FreeBSD version**
14.0-CURRENT #23 main-n255143-271f6d52a60

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d0 subvendor=0x1022 subdevice=0x15d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d1 subvendor=0x1022 subdevice=0x15d1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:1:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:1:7:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15db subvendor=0x5126 subdevice=0x17aa
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb3@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 0'
    class      = bridge
    subclass   = HOST-PCI
[... 271 lines cut ...]
cpuid = 4
time = 1651491261
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe00b4765860
vpanic() at vpanic+0x17f/frame 0xfffffe00b47658b0
panic() at panic+0x43/frame 0xfffffe00b4765910
trap_fatal() at trap_fatal+0x385/frame 0xfffffe00b4765970
trap_pfault() at trap_pfault+0x4f/frame 0xfffffe00b47659d0
calltrap() at calltrap+0x8/frame 0xfffffe00b47659d0
--- trap 0xc, rip = 0xffffffff806702c6, rsp = 0xfffffe00b4765aa0, rbp = 0xfffffe00b4765ad0 ---
fbd_register() at fbd_register+0xe6/frame 0xfffffe00b4765ad0
device_attach() at device_attach+0x3c3/frame 0xfffffe00b4765b20
bus_generic_attach() at bus_generic_attach+0x4b/frame 0xfffffe00b4765b50
lkpi_i2c_add_adapter() at lkpi_i2c_add_adapter+0x5b/frame 0xfffffe00b4765b70
drm_dp_port_set_pdt() at drm_dp_port_set_pdt+0x1e9/frame 0xfffffe00b4765bc0
drm_dp_send_link_address() at drm_dp_send_link_address+0x89a/frame 0xfffffe00b4765cf0
drm_dp_check_and_send_link_address() at drm_dp_check_and_send_link_address+0x27/frame 0xfffffe00b4765d30
drm_dp_mst_link_probe_work() at drm_dp_mst_link_probe_work+0x268/frame 0xfffffe00b4765de0
linux_work_fn() at linux_work_fn+0xe3/frame 0xfffffe00b4765e40
taskqueue_run_locked() at taskqueue_run_locked+0x181/frame 0xfffffe00b4765ec0
taskqueue_thread_loop() at taskqueue_thread_loop+0xc2/frame 0xfffffe00b4765ef0
fork_exit() at fork_exit+0x7e/frame 0xfffffe00b4765f30
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe00b4765f30
--- trap 0, rip = 0, rsp = 0, rbp = 0 ---
KDB: enter: panic

jownit, 2022-05-03:
Additional info:
5.5 worked
5.7 paniced in the same way as 5.10

Now I built drm-54-kmod-5.4.144 on FreeBSD 14.0-CURRENT #23 main-n255143-271f6d52a60 and it works.

So it broke between 5.5 and 5.7.

evadot, 2022-05-09:
I don't understand why you have 'fbd0 on drmn0' printed.
Or more exactly why I don't have it ...
Is it GENERIC without any tuning ?

jownit, 2022-05-09:
GENERIC-NODEBUG

debdrup, 2022-05-13:
As a follow-up from #166 above, `core.txt` can be found [here](https://people.freebsd.org/~debdrup/core.txt.1).

evadot, 2022-05-13:
So, I believe this is happening when one have multiple screen with different resolution.
I'll try to debug that soon.

jownit, 2022-05-17:
> So, I believe this is happening when one have multiple screen with different resolution. I'll try to debug that soon.

This is true in my setup. The ThinkPad has an internal screen with 1920x1080, and there is a 4K monitor and a 1920x1200 monitor connected to the dock.

evadot, 2022-05-17:
Could you confirm that with only the laptop screen everything work ?
I'm not at home this week so testing multiple display is hard right now for me.

jownit, 2022-05-17:
> Could you confirm that with only the laptop screen everything work ? I'm not at home this week so testing multiple display is hard right now for me.

Confirmed.
Loading the module with the laptop undocked it works. Docking it immediately panics.
I also tested having it docked with the external monitors unplugged, it works. If I plug in a monitor in, it panics.

evadot, 2022-05-17:
Thanks for confirmation, this will help me to fix this quickly when I'm back :)

evadot, 2022-05-23:
@debdrup @jownit Are both your docks usb-c ?
I cannot reproduce here with multiple screen and different resolution on multiple machines.

jownit, 2022-05-23:
> > @debdrup @jownit Are both your docks usb-c ? I cannot reproduce here with multiple screen and different resolution on multiple machines.
> 
> Mine is. I could try plugging in a monitor directly in the HDMI port on the laptop and see what results I get.

I just tried that. With a 4K monitor plugged in to the laptops HDMI port, it works. Plugging in another monitor in the dock displayport (which is USB-C) panics.

So it seems the problem is with USB-C.
Loading drm-54-kmod works.

debdrup, 2022-05-23:
> @debdrup @jownit Are both your docks usb-c ? I cannot reproduce here with multiple screen and different resolution on multiple machines.

I'm using [this dock](https://support.lenovo.com/sa/en/solutions/pd500265-thinkpad-thunderbolt-3-dock-gen-2-overview-and-sevice-parts).

> I just tried that. With a 4K monitor plugged in to the laptops HDMI port, it works. Plugging in another monitor in the dock displayport (which is USB-C) panics.
>
> So it seems the problem is with [...]

Should I try to reproduce this too?
I'd need to get a USB-C to DisplayPort connector in order to try it, unless I can locate a spare HDMI cable that isn't embedded in a wall.

jownit, 2022-06-01:
> So, I can't reproduce with an USB-C HDMI dongle. I don't have a dock here.

Anything I can do to help debugging?

evadot, 2022-07-25:
Closing this one now, @debdrup please re-open if my patch didn't fixed this for you, what @jownit is seeing is another problem.