freebsd/drm-kmod #166: drm-510: Kernel panic on external display

debdrup, 2022-05-13:
**Describe the bug**
Using drm-510-kmod, I can reproducibly cause a panic(9) with one or more external monitors attached via DisplayPort.
Using drm-54-kmod, I can have multiple external monitors attached, and they show up correctly during device initialization:
```
[8.073977] <6>[drm]   - kern.vt.fb.modes.DP-1
[8.073981] <6>[drm]   - kern.vt.fb.default_mode
[8.073995] <6>[drm] Connector HDMI-A-1: get mode from tunables:
[8.074001] <6>[drm]   - kern.vt.fb.modes.HDMI-A-1
[8.074006] <6>[drm]   - kern.vt.fb.default_mode
[8.074025] <6>[drm] Connector DP-2: get mode from tunables:
[8.074030] <6>[drm]   - kern.vt.fb.modes.DP-2
[8.074035] <6>[drm]   - kern.vt.fb.default_mode
[8.074045] <6>[drm] Connector HDMI-A-2: get mode from tunables:
[8.074051] <6>[drm]   - kern.vt.fb.modes.HDMI-A-2
[8.074056] <6>[drm]   - kern.vt.fb.default_mode
```

**FreeBSD version**
```
FreeBSD 14.0-CURRENT-drm-510-kmod 14.0-CURRENT FreeBSD 14.0-CURRENT #1 main-n255577-586ed321068: Wed May 11 19:48:41 CEST 2022     debdrup@geroi:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
```

**PCI Info**
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x5914 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
none1@pci0:0:8:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d2f subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d31 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Thermal subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d60 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
none2@pci0:0:22:0:	class=0x078000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3a subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP CSME HECI'
    class      = simple comms
pcib1@pci0:0:28:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d10 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:3:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d13 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:4:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d14 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:28:6:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d16 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:29:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d18 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d4e subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point LPC Controller/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none3@pci0:0:31:2:	class=0x058000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d21 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PMC'
    class      = memory
hdac0@pci0:0:31:3:	class=0x040300 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d71 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP HD Audio'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d23 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP SMBus'
    class      = serial bus
    subclass   = SMBus
em0@pci0:0:31:6:	class=0x020000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x15d8 subvendor=0x17aa subdevice=0x2258
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (4) I219-V'
    class      = network
    subclass   = ethernet
none4@pci0:2:0:0:	class=0x0d4000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x7360 subvendor=0x8086 subdevice=0x0020
    vendor     = 'Intel Corporation'
    device     = 'XMM7360 LTE Advanced Modem'
    class      = wireless controller
    subclass   = cellular controller/modem
iwm0@pci0:61:0:0:	class=0x028000 rev=0x78 hdr=0x00 vendor=0x8086 device=0x24fd subvendor=0x8086 subdevice=0x0010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 8265 / 8275'
    class      = network
nvme0@pci0:62:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x144d device=0xa804 subvendor=0x144d subdevice=0xa801
    vendor     = 'Samsung Electronics Co Ltd'
    device     = 'NVMe SSD Controller SM961/PM961/SM963'
    class      = mass storage
    subclass   = NVM
```

**DRM KMOD version**
drm-54-kmod 5.4.144

**To Reproduce**
Either:
1. Boot system with i915kms in kld_list in /etc/rc.conf
2. Attach external monitor

Or:
1. Boot system without i915kms
2. Attach external monitor
3. kldload i915kms

**Screenshots**
[image]
[image]
[image]
[image]

**Additional context**
The drm-510-kmod port was built with debugging symbols.
I have a boot environment with the setup available, if additional information is required.

evadot, 2022-05-13:
Do you have the panic message ? Or even better the core.txt ?
Thanks.

debdrup, 2022-05-13:
After rootcausing, this issue has been identified as a duplicate of #162.

As such, I'll go post the core.txt there. :)

closed 2022-05-13 by debdrup