freebsd/drm-kmod #167: Kernel freeze on green_sardine

ivan-volnov, 2022-05-17:
**Describe the bug**
Got a Kernel freeze after trying to start wayland window manager sway on AMD Ryzen 7 5800U.

**FreeBSD version**
`FreeBSD hostname 13.1-RELEASE FreeBSD 13.1-RELEASE releng/13.1-n250148-fc952ac2212 GENERIC amd64`

**DRM KMOD version**

```
drm-510-kmod-5.10.113
Name           : drm-510-kmod
Version        : 5.10.113
Installed on   : Tue May 17 18:39:48 2022 CST
Origin         : graphics/drm-510-kmod
Architecture   : FreeBSD:13:amd64
Prefix         : /usr/local
Categories     : graphics kld
Licenses       : BSD2CLAUSE, MIT, GPLv2
Maintainer     : x11@FreeBSD.org
WWW            : https://github.com/freebsd/drm-kmod/
Comment        : DRM drivers modules
Annotations    :
	FreeBSD_version: 1301000
Flat size      : 12.6MiB
Description    :
amdgpu, i915, and radeon DRM drivers modules.
Currently corresponding to Linux 5.10 DRM.
This version is for FreeBSD 13.1 and above.

WWW: https://github.com/freebsd/drm-kmod/
```

**GPU Firmware version**

```
gpu-firmware-amd-kmod-green-sardine-20220511
Name           : gpu-firmware-amd-kmod-green-sardine
Version        : 20220511
Installed on   : Tue May 17 18:35:24 2022 CST
Origin         : graphics/gpu-firmware-amd-kmod
Architecture   : FreeBSD:13:amd64
Prefix         : /usr/local
Categories     : graphics kld
Licenses       : AMD
Maintainer     : x11@FreeBSD.org
WWW            : https://github.com/freebsd/drm-kmod-firmware
Comment        : Firmware modules for green_sardine AMD GPUs
Annotations    :
	FreeBSD_version: 1301000
	flavor         : green_sardine
Flat size      : 1.63MiB
Description    :
Firmware modules for the drm-kmod drivers.

Firmware files for AMD GPUs supported by the amdgpu driver.

WWW: https://github.com/freebsd/drm-kmod-firmware
```

**Logs**
```
May 17 18:42:18 hostname kernel: [drm] amdgpu kernel modesetting enabled.
May 17 18:42:18 hostname kernel: drmn0: <drmn> on vgapci0
May 17 18:42:18 hostname kernel: vgapci0: child drmn0 requested pci_enable_io
May 17 18:42:18 hostname syslogd: last message repeated 1 times
May 17 18:42:18 hostname kernel: [drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x1002:0x0123 0xC1).
May 17 18:42:18 hostname kernel: drmn0: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
May 17 18:42:18 hostname kernel: [drm] register mmio base: 0xFCB00000
May 17 18:42:18 hostname kernel: [drm] register mmio size: 524288
May 17 18:42:18 hostname kernel: [drm] add ip block number 0 <soc15_common>
[... 71 lines cut ...]
May 17 18:42:18 hostname kernel: drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
May 17 18:42:18 hostname kernel: drmn0: ring sdma0 uses VM inv eng 0 on hub 1
May 17 18:42:18 hostname kernel: drmn0: ring vcn_dec uses VM inv eng 1 on hub 1
May 17 18:42:18 hostname kernel: drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 1
May 17 18:42:18 hostname kernel: drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 1
May 17 18:42:18 hostname kernel: drmn0: ring jpeg_dec uses VM inv eng 6 on hub 1
May 17 18:42:18 hostname kernel: vgapci0: child drmn0 requested pci_get_powerstate
May 17 18:42:18 hostname kernel: sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
May 17 18:42:18 hostname kernel: [drm] Initialized amdgpu 3.40.0 20150101 for drmn0 on minor 0
May 17 18:42:18 hostname kernel: acpi_wmi0: <ACPI-WMI mapping> on acpi0
May 17 18:42:18 hostname kernel: acpi_wmi0: cannot find EC device
May 17 18:42:18 hostname kernel: acpi_wmi0: Embedded MOF found
May 17 18:42:18 hostname kernel: ACPI: \AOD.WQBA: 1 arguments were passed to a non-method ACPI object (Buffer) (20201113/nsarguments-361)
May 17 18:42:18 hostname kernel: driver bug: Unable to set devclass (class: ppc devname: (unknown))
May 17 18:42:18 hostname kernel: intsmb0: <AMD FCH SMBus Controller> at device 20.0 on pci0
May 17 18:42:18 hostname kernel: smbus0: <System Management Bus> on intsmb0
```

**Logs just before the freeze**
```
May 17 18:42:49 hostname kernel: drmn0: 0xfffff80003597400 pin failed
May 17 18:42:49 hostname kernel: [drm ERROR :dm_plane_helper_prepare_fb] Failed to pin framebuffer with error -12
May 17 18:42:49 hostname kernel: drmn0: 0xfffff8000300b400 pin failed
May 17 18:42:49 hostname kernel: [drm ERROR :dm_plane_helper_prepare_fb] Failed to pin framebuffer with error -12
```

ivan-volnov, 2022-05-17:
Tried to increase video ram size to 4G and 8G. Still no luck.

@evadot What can I do more?

ivan-volnov, 2022-05-19:
Here is a post with the same error:
https://forums.freebsd.org/threads/after-refactored-drm-kmod-gpu-firmware-kmod-cursor-is-now-invisible.85209/

evadot, 2022-05-19:
Does Xorg works ? Or is it only wayland/sway that cause the hang ?

ivan-volnov, 2022-05-19:
Installed xorg just to test. It works without custom config. But it doesn't loads amdgpu.

After adding a config section

```
Section "Device"
        Identifier      "Card0"
        Driver          "amdgpu"
EndSection
```

It turns off all monitors with no signal.

[xorg_amdgpu.log](https://github.com/freebsd/drm-kmod/files/8729269/xorg_amdgpu.log)

ivan-volnov, 2022-05-19:
Log without amdgpu
[xorg_no_config.log](https://github.com/freebsd/drm-kmod/files/8729293/xorg_no_config.log)

ivan-volnov, 2022-05-19:
I found the error in the logs listed above:

```
[  1165.221] (II) AMDGPU(0): Setting screen physical size to 1354 x 381
[  1165.295] (EE) AMDGPU(0): failed to set mode: Cannot allocate memory
[  1165.295] (WW) AMDGPU(0): Failed to set mode on CRTC 0
[  1165.390] (EE) AMDGPU(0): failed to set mode: Cannot allocate memory
[  1165.390] (WW) AMDGPU(0): Failed to set mode on CRTC 1
[  1165.422] (EE) AMDGPU(0): Failed to enable any CRTC
```

So, it seams like a memory allocation issue along with `Failed to pin framebuffer with error -12`

As I wrote above I tried increase GPU memory up to 8G. And it's very strange because it worked reasonably well with old Renoir driver and firmware on 512M of video ram.

pointcheck, 2022-05-20:
> It turns off all monitors with no signal.

> [  1164.996] (--) AMDGPU(0): Chipset: "Unknown AMD Radeon GPU" (ChipID = 0x1638)

According to your Xorg.log your amdgpu was not recognized. You should upgrade drm-kmod to at least 5.7.19 that's where chip id 0x1638 was introduced and also to latest x11-drivers/xf86-video-amdgpu. Right after that your Xorg will start crashing deep inside llibthr/libc soon as it accesses amdgpu. I was trying to trace this bug down, but have not yet succeeded.

evadot, 2022-05-20:
> 

This is just the decoding of the marketing name that fails here, I don't think that this is a problem.

ivan-volnov, 2022-05-20:
> You should upgrade drm-kmod to at least 5.7.19 that's where chip id 0x1638 was introduced and also to latest x11-drivers/xf86-video-amdgpu

As I mentioned in the header post I already have drm-510-kmod-5.10.113 installed.
The version of xf86-video-amdgpu is 19.1.0_2 from packages.

So, both wayland and xorg fail to start.

pointcheck, 2022-05-20:
Ivan, I'm using 5.10-lts branch, that's the only one that works for me. The latest from Github does not even compile, lots of error messages as below. I'm on 13.1-RELAESE now.

`error: unknown type name 'ktime_t'; did you mean 'time_t'?`

wulf7, 2022-05-20:
> Firmware didn't changed (we used to track 20210315 and still do) so the only changes are in drm-kmod from 5.7 to 5.10

But amdgpu.ko starts load different firmware after commit f315fe56451e35658ece3367773a2dee91481349 which is not included in 5.7

ivan-volnov, 2022-05-20:
@pointcheck Does it work for you or you still have an issue with xorg?

@evadot Can it be because of [this](https://github.com/freebsd/drm-kmod/blob/5.10-lts/drivers/gpu/drm/amd/amdgpu/amdgpu_display.c#L503-L538)?

> But amdgpu.ko starts load different firmware

I confirm. I use it now with old firmware and old drm-devel-kmod

```
drmn0: successfully loaded firmware image 'amdgpu/renoir_asd.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_pfp.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_me.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_ce.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_rlc.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_mec.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_mec2.bin'
```

It works, but not perfectly.

ivan-volnov, 2022-05-20:
Tracked from dmesg error `[drm ERROR :dm_plane_helper_prepare_fb] Failed to pin framebuffer with error -12`

[dm_plane_helper_prepare_fb](https://github.com/freebsd/drm-kmod/blob/5.10-lts/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c#L5908) to [amdgpu_bo_pin](https://github.com/freebsd/drm-kmod/blob/5.10-lts/drivers/gpu/drm/amd/amdgpu/amdgpu_object.c#L895-L980) and [amdgpu_display_supported_domains](https://github.com/freebsd/drm-kmod/blob/5.10-lts/drivers/gpu/drm/amd/amdgpu/amdgpu_display.c#L503-L538)

evadot, 2022-05-20:
> 

Can you try old firmware but newer drm-kmod ?
IIRC vcn is just for video decoding stuff

pointcheck, 2022-05-20:
> @pointcheck Does it work for you or you still have an issue with xorg?

Works only with modesetting driver. Using amdgpu makes Xorg to crash. Same Xorg on Linux works fine. It's so annoying, so I decided to dive into this issue by myself. I could trace it to some code in libc, but have no clear understanding of what's going on. It makes me thinking that libc in FreeBSD 13.x has a bug (not completely sure).

ivan-volnov, 2022-05-20:
> Can you try old firmware but newer drm-kmod ?

With drm-510-kmod and old firmware it can't load green_sardine firmware during boot

ivan-volnov, 2022-05-20:
@pointcheck Are you sure that the problem is in libc? It can be just an endpoint of the problem but not the actual bug. Just double check that you dig to the right way.

evadot, 2022-05-20:
> > Can you try old firmware but newer drm-kmod ?
> 
> With drm-510-kmod and old firmware it can't load green_sardine firmware during boot

?
What's the error exactly ?

evadot, 2022-05-20:
If you could test to update the firmware with the latest one I've pushed there : https://github.com/freebsd/drm-kmod-firmware/

ivan-volnov, 2022-05-20:
> What's the error exactly ?

```
May 20 21:29:01 hostname kernel: [drm] amdgpu kernel modesetting enabled.
May 20 21:29:01 hostname kernel: drmn0: <drmn> on vgapci0
May 20 21:29:01 hostname kernel: vgapci0: child drmn0 requested pci_enable_io
May 20 21:29:01 hostname syslogd: last message repeated 1 times
May 20 21:29:01 hostname kernel: [drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x1002:0x0123 0xC1).
May 20 21:29:01 hostname kernel: drmn0: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
May 20 21:29:01 hostname kernel: [drm] register mmio base: 0xFCB00000
May 20 21:29:01 hostname kernel: [drm] register mmio size: 524288
May 20 21:29:01 hostname kernel: [drm] add ip block number 0 <soc15_common>
May 20 21:29:01 hostname kernel: [drm] add ip block number 1 <gmc_v9_0>
May 20 21:29:01 hostname kernel: [drm] add ip block number 2 <vega10_ih>
May 20 21:29:01 hostname kernel: [drm] add ip block number 3 <psp>
May 20 21:29:01 hostname kernel: [drm] add ip block number 4 <smu>
May 20 21:29:01 hostname kernel: [drm] add ip block number 5 <gfx_v9_0>
May 20 21:29:01 hostname kernel: [drm] add ip block number 6 <sdma_v4_0>
May 20 21:29:01 hostname kernel: [drm] add ip block number 7 <dm>
May 20 21:29:01 hostname kernel: [drm] add ip block number 8 <vcn_v2_0>
May 20 21:29:01 hostname kernel: [drm] add ip block number 9 <jpeg_v2_0>
May 20 21:29:01 hostname kernel: drmn0: Fetched VBIOS from VFCT
May 20 21:29:01 hostname kernel: amdgpu: ATOM BIOS: 113-CEZANNE-018
May 20 21:29:01 hostname kernel: drmn0: could not load firmware image 'amdgpu/green_sardine_sdma.bin'
May 20 21:29:01 hostname kernel: [drm ERROR :sdma_v4_0_init_microcode] sdma_v4_0: Failed to load firmware "amdgpu/green_sardine_sdma.bin"
May 20 21:29:01 hostname kernel: [drm ERROR :sdma_v4_0_early_init] Failed to load sdma firmware!
May 20 21:29:01 hostname kernel: [drm] VCN decode is enabled in VM mode
May 20 21:29:01 hostname kernel: [drm] VCN encode is enabled in VM mode
May 20 21:29:01 hostname kernel: [drm] JPEG decode is enabled in VM mode
May 20 21:29:01 hostname kernel: [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
May 20 21:29:01 hostname kernel: drmn0: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
May 20 21:29:01 hostname kernel: drmn0: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF
May 20 21:29:01 hostname kernel: drmn0: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
May 20 21:29:01 hostname kernel: [drm] Detected VRAM RAM=4096M, BAR=4096M
May 20 21:29:01 hostname kernel: [drm] RAM width 128bits DDR4
May 20 21:29:01 hostname kernel: [TTM] Zone  kernel: Available graphics memory: 31386586 KiB
May 20 21:29:01 hostname kernel: [TTM] Zone   dma32: Available graphics memory: 2097152 KiB
May 20 21:29:01 hostname kernel: [TTM] Initializing pool allocator
May 20 21:29:01 hostname kernel: [drm] amdgpu: 4096M of VRAM memory ready
May 20 21:29:01 hostname kernel: [drm] amdgpu: 4096M of GTT memory ready.
May 20 21:29:01 hostname kernel: [drm] GART: num cpu pages 262144, num gpu pages 262144
May 20 21:29:01 hostname kernel: [drm] PCIE GART of 1024M enabled (table at 0x000000F400900000).
May 20 21:29:01 hostname kernel: drmn0: could not load firmware image 'amdgpu/green_sardine_asd.bin'
May 20 21:29:01 hostname kernel: drmn0: fail to initialize asd microcode
May 20 21:29:01 hostname kernel: [drm ERROR :psp_sw_init] Failed to load psp firmware!
May 20 21:29:01 hostname kernel: [drm ERROR :amdgpu_device_ip_init] sw_init of IP block <psp> failed -2
May 20 21:29:01 hostname kernel: drmn0: amdgpu_device_ip_init failed
May 20 21:29:01 hostname kernel: drmn0: Fatal error during GPU init
May 20 21:29:01 hostname kernel: drmn0: amdgpu: finishing device.
May 20 21:29:01 hostname kernel: [TTM] Finalizing pool allocator
May 20 21:29:01 hostname kernel: [TTM] Zone  kernel: Used memory at exit: 0 KiB
May 20 21:29:01 hostname kernel: [TTM] Zone   dma32: Used memory at exit: 0 KiB
May 20 21:29:01 hostname kernel: [drm] amdgpu: ttm finalized
May 20 21:29:01 hostname kernel: device_attach: drmn0 attach returned 2
```

evadot, 2022-05-20:
And firmware are present and correctly built ? (i.e. not built for an earlier version of FreeBSD ?)

ivan-volnov, 2022-05-20:
> And firmware are present and correctly built ? (i.e. not built for an earlier version of FreeBSD ?)

It was an old firmware from FreeBSD 13.1-RC5. Can't reproduce of building it.

> If you could test to update the firmware with the latest one I've pushed there : https://github.com/freebsd/drm-kmod-firmware/

Ok, I'll try

ivan-volnov, 2022-05-20:
> Reproduce what ? The problem ?

That old firmware was from packages while I tested 13.1-RC5. There is no such package anymore, but I saved zfs snapshot when I had it. I can't reinstall it.
It still works with old driver in combo.

> If you could test to update the firmware with the latest one I've pushed there : https://github.com/freebsd/drm-kmod-firmware/

Strange, it doesn't work for me.
Removed old firmware, git cloned and executed `make install` with no errors.
Is it correct installation procedure from git?

```
[drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x1002:0x0123 0xC1).
drmn0: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
[drm] register mmio base: 0xFCB00000
[drm] register mmio size: 524288
[drm] add ip block number 0 <soc15_common>
[drm] add ip block number 1 <gmc_v9_0>
[drm] add ip block number 2 <vega10_ih>
[drm] add ip block number 3 <psp>
[drm] add ip block number 4 <smu>
[drm] add ip block number 5 <gfx_v9_0>
[drm] add ip block number 6 <sdma_v4_0>
[drm] add ip block number 7 <dm>
[drm] add ip block number 8 <vcn_v2_0>
[drm] add ip block number 9 <jpeg_v2_0>
drmn0: Fetched VBIOS from VFCT
amdgpu: ATOM BIOS: 113-CEZANNE-018
drmn0: could not load firmware image 'amdgpu/green_sardine_sdma.bin'
[drm ERROR :sdma_v4_0_init_microcode] sdma_v4_0: Failed to load firmware "amdgpu/green_sardine_sdma.bin"
[drm ERROR :sdma_v4_0_early_init] Failed to load sdma firmware!
[drm] VCN decode is enabled in VM mode
[drm] VCN encode is enabled in VM mode
[drm] JPEG decode is enabled in VM mode
[drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
drmn0: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
drmn0: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF
drmn0: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
[drm] Detected VRAM RAM=4096M, BAR=4096M
[drm] RAM width 128bits DDR4
[TTM] Zone  kernel: Available graphics memory: 31386586 KiB
[TTM] Zone   dma32: Available graphics memory: 2097152 KiB
[TTM] Initializing pool allocator
[drm] amdgpu: 4096M of VRAM memory ready
[drm] amdgpu: 4096M of GTT memory ready.
[drm] GART: num cpu pages 262144, num gpu pages 262144
[drm] PCIE GART of 1024M enabled (table at 0x000000F400900000).
drmn0: could not load firmware image 'amdgpu/green_sardine_asd.bin'
drmn0: fail to initialize asd microcode
[drm ERROR :psp_sw_init] Failed to load psp firmware!
[drm ERROR :amdgpu_device_ip_init] sw_init of IP block <psp> failed -2
drmn0: amdgpu_device_ip_init failed
drmn0: Fatal error during GPU init
drmn0: amdgpu: finishing device.
[TTM] Finalizing pool allocator
[TTM] Zone  kernel: Used memory at exit: 0 KiB
[TTM] Zone   dma32: Used memory at exit: 0 KiB
[drm] amdgpu: ttm finalized
device_attach: drmn0 attach returned 2
```

evadot, 2022-05-20:
> 

Do you have any green_sardine firmware in /boot/modules ?

ivan-volnov, 2022-05-20:
> Do you have any green_sardine firmware in /boot/modules ?

Strange, but I don't
It installs a lot of firmwares but not green_sardine

ivan-volnov, 2022-05-20:
> Are the sources in /usr/src from 13.1 and/or current ?

```
❯ cd /usr/src && git status
On branch releng/13.1
```

ivan-volnov, 2022-05-20:
Ok. Tested the firmware. The new version from git works in the same way as the last version from ports.
To build it I moved green_sardine from 13.1 expression to a normal one in the Makefile.

The fresh firmware
```
[drm] amdgpu kernel modesetting enabled.
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] initializing kernel modesetting (RENOIR 0x1002:0x1638 0x1002:0x0123 0xC1).
drmn0: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
[drm] register mmio base: 0xFCB00000
[drm] register mmio size: 524288
[drm] add ip block number 0 <soc15_common>
[drm] add ip block number 1 <gmc_v9_0>
[drm] add ip block number 2 <vega10_ih>
[drm] add ip block number 3 <psp>
[drm] add ip block number 4 <smu>
[drm] add ip block number 5 <gfx_v9_0>
[drm] add ip block number 6 <sdma_v4_0>
[drm] add ip block number 7 <dm>
[drm] add ip block number 8 <vcn_v2_0>
[drm] add ip block number 9 <jpeg_v2_0>
drmn0: Fetched VBIOS from VFCT
amdgpu: ATOM BIOS: 113-CEZANNE-018
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_sdma.bin'
[drm] VCN decode is enabled in VM mode
[drm] VCN encode is enabled in VM mode
[drm] JPEG decode is enabled in VM mode
[drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
drmn0: VRAM: 4096M 0x000000F400000000 - 0x000000F4FFFFFFFF (4096M used)
drmn0: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF
drmn0: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
[drm] Detected VRAM RAM=4096M, BAR=4096M
[drm] RAM width 128bits DDR4
[TTM] Zone  kernel: Available graphics memory: 31386586 KiB
[TTM] Zone   dma32: Available graphics memory: 2097152 KiB
[TTM] Initializing pool allocator
[drm] amdgpu: 4096M of VRAM memory ready
[drm] amdgpu: 4096M of GTT memory ready.
[drm] GART: num cpu pages 262144, num gpu pages 262144
[drm] PCIE GART of 1024M enabled (table at 0x000000F400900000).
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_asd.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_ta.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_pfp.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_me.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_ce.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_rlc.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_mec.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_mec2.bin'
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_dmcub.bin'
[drm] Loading DMUB firmware via PSP: version=0x01010014
drmn0: successfully loaded firmware image 'amdgpu/green_sardine_vcn.bin'
[drm] Found VCN firmware Version ENC: 1.14 DEC: 5 VEP: 0 Revision: 20
drmn0: Will use PSP to load VCN firmware
[drm] reserve 0x400000 from 0xf4ff800000 for PSP TMR
drmn0: RAS: optional ras ta ucode is not available
drmn0: RAP: optional rap ta ucode is not available
drmn0: SMU is initialized successfully!
[drm] kiq ring mec 2 pipe 1 q 0
[drm] Display Core initialized with v3.2.104!
[drm] DMUB hardware initialized: version=0x01010014
[drm] VCN decode and encode initialized successfully(under DPG Mode).
[drm] JPEG decode initialized successfully.
drmn0: SE 1, SH per SE 1, CU per SH 8, active_cu_number 8
[drm] fb mappable at 0xF30CCF000
[drm] vram apper at 0xF30000000
[drm] size 14745600
[drm] fb depth is 24
[drm]    pitch is 10240
[... 147 lines cut ...]
sway log:

```
May 21 00:52:03 amdgpu: os_same_file_description couldn't determine if two DRM fds reference the same file description.
May 21 00:52:03 If they do, bad things may happen!
May 21 00:52:03 2022-05-21 00:52:03 - [main.c:579] Missing a required Wayland interface
May 21 00:52:03 00:00:00.608 [ERROR] [wlr] [backend/drm/atomic.c:38] connector DP-1: Atomic commit failed (modeset): Cannot allocate memory
May 21 00:52:03 00:00:00.608 [ERROR] [sway/config/output.c:501] Failed to commit output DP-1
May 21 00:52:03 00:00:00.622 [ERROR] [wlr] [backend/drm/atomic.c:38] connector HDMI-A-1: Atomic commit failed (modeset): Cannot allocate memory
May 21 00:52:03 00:00:00.622 [ERROR] [sway/config/output.c:501] Failed to commit output HDMI-A-1
May 21 00:52:03 00:00:00.644 [ERROR] [sway/config.c:682] Error on line 'focus output DP-1': Can't run this command while there's no outputs connected.
May 21 00:52:03 2022-05-21 00:52:03 - [Line 448] Failed to find session
May 21 00:52:03 2022-05-21 00:52:03 - [Line 279] Failed to parse get BlockInhibited property: Invalid argument
```

xorg log:

```
[    49.745] (II) AMDGPU(0): Setting screen physical size to 1354 x 381
[    49.830] (EE) AMDGPU(0): failed to set mode: Cannot allocate memory
[    49.830] (WW) AMDGPU(0): Failed to set mode on CRTC 0
[    49.915] (EE) AMDGPU(0): failed to set mode: Cannot allocate memory
[    49.915] (WW) AMDGPU(0): Failed to set mode on CRTC 1
[    49.946] (EE) AMDGPU(0): Failed to enable any CRTC
```

ivan-volnov, 2022-05-20:
Tried tune BIOS:

```
Enabled Above 4G Decoding
Enabled Re-Size BAR Support
Set VRAM to 16G
```

Still no luck. Undo.

evadot, 2022-05-20:
Ok so the problem is definitely in drm-kmod.
First it would be good to confirm that using the firmware present in ports and building drm-kmod from tag drm_v5.7.19_4 works (this is what drm-devel-kmod was).
Then you might want to try some tags in drm-kmod to reduce bisecting, drm_v5.8 and drm_v5.9 first.
Then bisect between those tags.

ivan-volnov, 2022-05-20:
@evadot Can you explain please how to install it? What port I need to choose or build it from git?

evadot, 2022-05-20:
No port, directly clone the repo here and git checkout <tag>

ivan-volnov, 2022-05-21:
@evadot Ok. Tested.

```
drm_v5.7.19_4       works with renoir firmware
drm_v5.8            doesn't build
drm_v5.9_2          driver doesn't load silently
drm_v5.10_5         driver doesn't load silently
drm_v5.10.113_1     loads ok but cannot allocate memory error
```

drm_v5.8 build error:

```
===> linuxkpi (all)
Warning: Object directory not changed from original /root/drm-kmod/linuxkpi
cc  -O2 -pipe '-DKBUILD_MODNAME="linuxkpi_gplv2"' -DLINUXKPI_VERSION=50000 -DCONFIG_DRM_AMDGPU_CIK -DCONFIG_DRM_AMDGPU_SI -DCONFIG_DRM_AMD_DC -DCONFIG_DRM_AMD_DC_FBC -DCONFIG_DRM_AMD_POWERPLAY -DCONFIG_DRM_I915_ALPHA_SUPPORT -DCONFIG_DRM_I915_FORCE_PROBE='"*"' -DCONFIG_DRM_I915_CAPTURE_ERROR -DCONFIG_DRM_I915_SPIN_REQUEST=5 -DCONFIG_DRM_I915_USERFAULT_AUTOSUSPEND=250 -DCONFIG_DRM_I915_STOP_TIMEOUT=100 -DCONFIG_DRM_I915_PREEMPT_TIMEOUT=640 -DCONFIG_DRM_I915_HEARTBEAT_INTERVAL=2500 -DCONFIG_DRM_I915_TIMESLICE_DURATION=1 -DCONFIG_DRM_I915_MAX_REQUEST_BUSYWAIT=8000 -DCONFIG_DRM_I915_FENCE_TIMEOUT=10000 -DCONFIG_DRM_MIPI_DSI -DCONFIG_DRM_PANEL_ORIENTATION_QUIRKS -DCONFIG_DRM_FBDEV_EMULATION -DCONFIG_DRM_FBDEV_OVERALLOC=100 -DCONFIG_ARCH_HAVE_NMI_SAFE_CMPXCHG -DCONFIG_BACKLIGHT_CLASS_DEVICE -DCONFIG_DMI -DCONFIG_FB -DCONFIG_MTRR -DCONFIG_PCI -DCONFIG_PM -DCONFIG_SMP -DCONFIG_ACPI -DCONFIG_ACPI_SLEEP -DCONFIG_X86 -DCONFIG_X86_PAT -DCONFIG_64BIT -DCONFIG_AS_MOVNTDQA -DCONFIG_COMPAT -DCONFIG_X86_64 -DCONFIG_DRM_AMD_DC_DCN -DCONFIG_DRM_AMD_DC_DCN1_0 -DCONFIG_DRM_AMD_DC_DCN1_01 -DCONFIG_DRM_AMD_DC_DCN2_0 -DCONFIG_DRM_AMD_DC_DCN2_1 -DCONFIG_DRM_AMD_DC_DSC_SUPPORT  -fno-strict-aliasing -Werror -D_KERNEL -DKLD_MODULE -nostdinc  -I/root/drm-kmod/include -I/root/drm-kmod/linuxkpi/dummy/include -I/root/drm-kmod/linuxkpi/gplv2/include -I/root/drm-kmod/linuxkpi/bsd/include -I/usr/src/sys/compat/linuxkpi/common/include -include /root/drm-kmod/linuxkpi/opt_global.h -I. -I/usr/src/sys -I/usr/src/sys/contrib/ck/include -fno-common  -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdebug-prefix-map=./machine=/usr/src/sys/amd64/include -fdebug-prefix-map=./x86=/usr/src/sys/x86/include     -MD  -MF.depend.linux_compat.o -MTlinux_compat.o -mcmodel=kernel -mno-red-zone -mno-mmx -mno-sse -msoft-float  -fno-asynchronous-unwind-tables -ffreestanding -fwrapv -fstack-protector -Wall -Wredundant-decls -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wcast-qual -Wundef -Wno-pointer-sign -D__printf__=__freebsd_kprintf__ -Wmissing-include-dirs -fdiagnostics-show-option -Wno-unknown-pragmas -Wno-error=tautological-compare -Wno-error=empty-body -Wno-error=parentheses-equality -Wno-error=unused-function -Wno-error=pointer-sign -Wno-error=shift-negative-value -Wno-address-of-packed-member -Wno-error=unused-but-set-variable -Wno-format-zero-length -Wno-pointer-arith   -mno-aes -mno-avx  -std=iso9899:1999 -c /root/drm-kmod/linuxkpi/gplv2/src/linux_compat.c -o linux_compat.o
*** Error code 1

Stop.
make[1]: stopped in /root/drm-kmod/linuxkpi
*** Error code 1

Stop.
make: stopped in /root/drm-kmod
```

`dmesg | grep drm` on drm_v5.9_2 and drm_v5.10_5

```
[drm] amdgpu kernel modesetting enabled.
```

ivan-volnov, 2022-05-21:
There are [981 commits between drm_v5.7.19_4 and drm_v5.8](https://github.com/freebsd/drm-kmod/compare/drm_v5.7.19_4...drm_v5.8) branches

pointcheck, 2022-05-21:
The master version from repository does not build.

```
rz@butterfly:~/drm-kmod % uname -r
13.1-RELEASE
```

```
rz@butterfly:~/drm-kmod % git status && git log -1
Refresh index: 100% (2391/2391), done.
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
commit 2f6f91e3bc68f1949eeb25020540c3106b8bc806 (HEAD -> master, origin/master, origin/HEAD)
Author: Vladimir Kondratyev <wulf@FreeBSD.org>
Date:   Tue May 17 14:58:06 2022 +0300

    LinuxKPI: Use linux/irq_work.h from base
```

```
rz@butterfly:~/drm-kmod % make
===> dmabuf (all)
Warning: Object directory not changed from original /usr/home/rz/drm-kmod/dmabuf
cc  -O2 -pipe '-DKBUILD_MODNAME="dmabuf"' -DLINUXKPI_VERSION=50000 -DCONFIG_DRM_AMDGPU_CIK -DCONFIG_DRM_AMDGPU_SI -DCONFIG_DRM_AMD_DC -DCONFIG_DRM_AMD_DC_SI -DCONFIG_DRM_I915_FORCE_PROBE='"*"' -DCONFIG_DRM_I915_CAPTURE_ERROR -DCONFIG_DRM_I915_USERFAULT_AUTOSUSPEND=250 -DCONFIG_DRM_I915_STOP_TIMEOUT=100 -DCONFIG_DRM_I915_PREEMPT_TIMEOUT=640 -DCONFIG_DRM_I915_HEARTBEAT_INTERVAL=2500 -DCONFIG_DRM_I915_TIMESLICE_DURATION=1 -DCONFIG_DRM_I915_MAX_REQUEST_BUSYWAIT=8000 -DCONFIG_DRM_I915_FENCE_TIMEOUT=10000 -DCONFIG_DRM_MIPI_DSI -DCONFIG_DRM_PANEL_ORIENTATION_QUIRKS -DCONFIG_DRM_FBDEV_EMULATION -DCONFIG_DRM_FBDEV_OVERALLOC=100 -DCONFIG_ARCH_HAVE_NMI_SAFE_CMPXCHG -DCONFIG_BACKLIGHT_CLASS_DEVICE -DCONFIG_DMI -DCONFIG_FB -DCONFIG_MTRR -DCONFIG_PCI -DCONFIG_PM -DCONFIG_SMP -DCONFIG_ACPI -DCONFIG_ACPI_SLEEP -DCONFIG_X86 -DCONFIG_X86_PAT -DCONFIG_64BIT -DCONFIG_AS_MOVNTDQA -DCONFIG_COMPAT -DCONFIG_X86_64 -DCONFIG_DRM_AMD_DC_DCN -DCONFIG_DRM_AMD_DC_DCN3_0  -fno-strict-aliasing -Werror -D_KERNEL -DKLD_MODULE -nostdinc  -I/usr/home/rz/drm-kmod/linuxkpi/gplv2/include -I/usr/home/rz/drm-kmod/linuxkpi/bsd/include -I/usr/src/sys/compat/linuxkpi/common/include -I/usr/home/rz/drm-kmod/linuxkpi/dummy/include -include /usr/home/rz/drm-kmod/dmabuf/opt_global.h -I. -I/usr/src/sys -I/usr/src/sys/contrib/ck/include -fno-common  -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdebug-prefix-map=./machine=/usr/src/sys/amd64/include -fdebug-prefix-map=./x86=/usr/src/sys/x86/include     -MD  -MF.depend.dma-buf.o -MTdma-buf.o -mcmodel=kernel -mno-red-zone -mno-mmx -mno-sse -msoft-float  -fno-asynchronous-unwind-tables -ffreestanding -fwrapv -fstack-protector -Wall -Wredundant-decls -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wcast-qual -Wundef -Wno-pointer-sign -D__printf__=__freebsd_kprintf__ -Wmissing-include-dirs -fdiagnostics-show-option -Wno-unknown-pragmas -Wno-error=tautological-compare -Wno-error=empty-body -Wno-error=parentheses-equality -Wno-error=unused-function -Wno-error=pointer-sign -Wno-error=shift-negative-value -Wno-address-of-packed-member -Wno-error=unused-but-set-variable -Wno-format-zero-length   -mno-aes -mno-avx  -std=iso9899:1999 -c /usr/home/rz/drm-kmod/drivers/dma-buf/dma-buf.c -o dma-buf.o
In file included from /usr/home/rz/drm-kmod/drivers/dma-buf/dma-buf.c:53:
In file included from /usr/home/rz/drm-kmod/linuxkpi/gplv2/include/linux/dma-buf.h:34:
/usr/home/rz/drm-kmod/linuxkpi/gplv2/include/linux/dma-fence.h:87:3: error: unknown type name 'ktime_t'; did you mean 'time_t'?
                ktime_t timestamp;
                ^~~~~~~
                time_t
/usr/src/sys/sys/types.h:216:18: note: 'time_t' declared here
typedef __time_t        time_t;
                        ^
1 error generated.
*** Error code 1

Stop.
make[1]: stopped in /usr/home/rz/drm-kmod/dmabuf
*** Error code 1

Stop.
make: stopped in /usr/home/rz/drm-kmod
```

ivan-volnov, 2022-05-21:
> The master version from repository does not build.

Yep, doesn't build too

ivan-volnov, 2022-05-21:
Update.

```
drm_v5.7.19_4       works with renoir firmware
drm_v5.8            doesn't build
drm_v5.9_2          works with renoir firmware after adding device ids
drm_v5.10_5         loads ok after adding device ids, but cannot allocate memory error
drm_v5.10.113_1     loads ok but cannot allocate memory error
```

Manually added [device ids](https://github.com/freebsd/drm-kmod/commit/44a2eaf653479067ddee2c2db79e68993ec5b4b8)

ivan-volnov, 2022-05-21:
[This commit](https://github.com/freebsd/drm-kmod/commit/983a29c28df30ea7ef39f29e03732c04cbef6afd) still works with manually added device ids

Excluded unnecessary modules from build:

```
KMODS="dmabuf linuxkpi ttm drm amd" make -j16
```

So, the issue commit range is https://github.com/freebsd/drm-kmod/compare/983a29c28df30ea7ef39f29e03732c04cbef6afd...drm_v5.10_5

ivan-volnov, 2022-05-21:
Tried another commit with some changes to build

[drm_v5.10~300 commit](https://github.com/freebsd/drm-kmod/commit/ed91f467087fd5f10baf2c7c02ee6925387a1551) cannot allocate memory error with renoir firmware

So, the issue commit range is https://github.com/freebsd/drm-kmod/compare/drm_v5.10~626...drm_v5.10~300

ivan-volnov, 2022-05-21:
The issue commit range is https://github.com/freebsd/drm-kmod/compare/drm_v5.10~550...drm_v5.10~450

ivan-volnov, 2022-05-21:
The issue commit range is https://github.com/freebsd/drm-kmod/compare/drm_v5.10~550...drm_v5.10~525

ivan-volnov, 2022-05-21:
The issue commit range is https://github.com/freebsd/drm-kmod/compare/drm_v5.10~531...drm_v5.10~528

ivan-volnov, 2022-05-21:
@evadot @wulf7 Hope it's enough to figure it out.

ivan-volnov, 2022-05-21:
Can't stop

The issue commit range is https://github.com/freebsd/drm-kmod/compare/drm_v5.10~530...drm_v5.10~528

ivan-volnov, 2022-05-21:
So, that's the problematic commit
https://github.com/freebsd/drm-kmod/commit/ec91e61f5007daeb1a45a14700949b08c77d59c0

evadot, 2022-05-23:
Does this mean that taking 5.10-lts and git revert https://github.com/freebsd/drm-kmod/commit/ec91e61f5007daeb1a45a14700949b08c77d59c0 works for you ?

ivan-volnov, 2022-05-23:
> Does this mean that taking 5.10-lts and git revert [ec91e61](https://github.com/freebsd/drm-kmod/commit/ec91e61f5007daeb1a45a14700949b08c77d59c0) works for you ?

It's a bit difficult to revert on 5.10-lts due to git conflicts. This commit adds a renoir firmware but the branch moved to green_sardine firmware for my GPU.

I hope you can figure it out how to fix the bug knowing the issue commit.

May be some parts of HDCP are missing or just not working in 5.10?

Or may be HDCP just doesn't work on FreeBSD and we need to remove it from drivers?

ivan-volnov, 2022-05-23:
@evadot Ok, that's the PR. Tested.

ivan-volnov, 2022-05-26:
Until the PR is accepted everyone who want to make it working can install the drivers from port as listed below:

```sh
cd /usr/ports/graphics/gpu-firmware-amd-kmod
FLAVOR=green_sardine make install clean
cd /usr/ports/graphics/drm-510-kmod
make extract
vim work/drm-kmod-drm_v5.10.113_1/drivers/gpu/drm/amd/amdgpu/psp_v12_0.c
make install clean
```

pointcheck, 2022-06-01:
Still does not build on 13.1.

```
/usr/home/rz/drm-kmod/linuxkpi/gplv2/include/linux/dma-fence.h:87:3: error: unknown type name 'ktime_t'; did you mean 'time_t'?
                ktime_t timestamp;
                ^~~~~~~
                time_t
/usr/src/sys/sys/types.h:216:18: note: 'time_t' declared here
typedef __time_t        time_t;
                        ^
1 error generated.
*** Error code 1
```

evadot, 2022-06-01:
master branch doesn't support 13.1, 5.10-lts does.

pointcheck, 2022-06-01:
May I know what is the purpose of master branch than ? Does it build on 14-CURRENT ?

evadot, 2022-06-02:
Well of course it build on 14-CURRENT.