freebsd/drm-kmod #168: Kernel panic when load of radeonkms is followed by load of amdgpu or i915kms

grahamperrin, 2022-05-19:
**Describe the bug**

Examples: 

* panic on real hardware that requires `i915kms`, after booting from a mobile hard disk drive that was preconfigured for `radeonkms`
* panics in a Virtualbox guest, when (for reproduction purposes) a combination of modules is loaded.

**FreeBSD version**

FreeBSD 14.0-CURRENT and 13.1-RELEASE.

**PCI Info**

For the HP notebook, which I booted from the mobile hard disk drive (14.0-CURRENT), I can't get details at the moment. (It's not my computer.)

Details below are for the VirtualBox guest (13.1-RELEASE). 

<details>
  <summary>pciconf -lv</summary>
```text
    hostb0@pci0:0:0:0:	class=0x060000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x1237 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = '440FX - 82441FX PMC [Natoma]'
        class      = bridge
        subclass   = HOST-PCI
    isab0@pci0:0:1:0:	class=0x060100 rev=0x00 hdr=0x00 vendor=0x8086 device=0x7000 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = '82371SB PIIX3 ISA [Natoma/Triton II]'
        class      = bridge
        subclass   = PCI-ISA
    vgapci0@pci0:0:2:0:	class=0x030000 rev=0x00 hdr=0x00 vendor=0x80ee device=0xbeef subvendor=0x15ad subdevice=0x0405
        vendor     = 'InnoTek Systemberatung GmbH'
        device     = 'VirtualBox Graphics Adapter'
        class      = display
        subclass   = VGA
    em0@pci0:0:3:0:	class=0x020000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x100e subvendor=0x8086 subdevice=0x001e
        vendor     = 'Intel Corporation'
        device     = '82540EM Gigabit Ethernet Controller'
        class      = network
        subclass   = ethernet
    none0@pci0:0:4:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x80ee device=0xcafe subvendor=0x0000 subdevice=0x0000
        vendor     = 'InnoTek Systemberatung GmbH'
        device     = 'VirtualBox Guest Service'
        class      = base peripheral
    pcm0@pci0:0:5:0:	class=0x040100 rev=0x01 hdr=0x00 vendor=0x8086 device=0x2415 subvendor=0x8086 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = '82801AA AC'97 Audio Controller'
        class      = multimedia
        subclass   = audio
    ohci0@pci0:0:6:0:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x106b device=0x003f subvendor=0x0000 subdevice=0x0000
        vendor     = 'Apple Inc.'
        device     = 'KeyLargo/Intrepid USB'
        class      = serial bus
        subclass   = USB
    intsmb0@pci0:0:7:0:	class=0x068000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x7113 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = '82371AB/EB/MB PIIX4 ACPI'
        class      = bridge
    ahci0@pci0:0:13:0:	class=0x010601 rev=0x02 hdr=0x00 vendor=0x8086 device=0x2829 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = '82801HM/HEM (ICH8M/ICH8M-E) SATA Controller [AHCI mode]'
        class      = mass storage
        subclass   = SATA
```
</details>

**DRM KMOD version**

drm-510-kmod 5.10.113

**To Reproduce**

Boot with either of these listings in `/etc/rc.conf`: 

* `kld_list=radeonkms i915kms`
* `kld_list=radeonkms amdgpu`

For the first of those two: 

```text
Dump header from device: /dev/gpt/freebsd-swap
  Architecture: amd64
  Architecture Version: 2
  Dump Length: 191676416
  Blocksize: 512
  Compression: none
  Dumptime: 2022-05-16 19:31:03 -0400
  Hostname: freebsd
  Magic: FreeBSD Kernel Dump
  Version String: FreeBSD 13.1-RELEASE releng/13.1-n250148-fc952ac2212 GENERIC
  Panic String: page fault
  Dump Parity: 4205894177
  Bounds: 8
  Dump Status: good
```

[vmcore.8.txt](https://github.com/freebsd/drm-kmod/files/8732795/vmcore.8.txt)

**Additional context**

From the thread (missing from archives) at <https://gitter.im/FreeBSDDesktop/Lobby/archives/2022/05/16?at=6282929961f5ca3fee04c20f>: 

> Whilst `kld_list=amdgpu radeonkms` does not panic the system, `kldload` can not load `radeonkms`.
> 
> Reportedly "module already loaded or in kernel" and `kldstat | grep radeon` does not list `radeonkms`.
> 
> Playing with fire (?) I can unload `amdgpu` then load `radeonkms`. 
> 
> Subsequent unload of `radeonkms` succeeds, subsequent load of `i915kms` causes a panic with (extraordinarily) no prompt; no option to continue: 
> 
> 
[image]

wulf7, 2022-05-31:
Probably, there is a fix: https://github.com/dumbbell/drm-kmod/commit/0ebda877d01f4b97c48e9e49365829bd66da1a0e

grahamperrin, 2023-06-10:
See also: FreeBSD bug [271897 – graphics/drm-510-kmod: loading amdgpu and i915kms kernel modules simultaneously causes the system to crash](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=271897) (_assuming_ drm-510-kmod).

octal-illumination, 2023-06-12:
Facing similar issue. I have installed drm-kmod, not sure if its drm-510-kmod. But in my case, setting kld_list="i915kms amdgpu" causes the system to not boot/panic/crash. Is there anything I should add to help solve the issue?.  [Posted this comment at the bugs.freesd.org too].

grahamperrin, 2023-06-13:
> # … when load of radeonkms is followed by load of amdgpu or i915kms

The 13.2-RELEASE kernel panic at <https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=271897#c15> was: 

* load of `i915kms` by [rc(8)](https://man.freebsd.org/cgi/man.cgi?query=rc&sektion=8&manpath=freebsd-release)
* followed by `kldload amdgpu`.

```text
root@fuji:~ # sysrc kld_list
kld_list: i915kms fusefs
root@fuji:~ # 
```

I'm inclined to treat 271897 as overlapping with (or the same as) this issue #168, and change the title here … @evadot please, would you agree?