freebsd/drm-kmod #173: Backlight doesn't work on drm-kmod 5.10 or 5/26 master

neelchauhan, 2022-05-27:
**Describe the bug**

On an Intel TigerLake system, I am unable to use the backlight on drm-kmod 5.10. Older drm-kmod releases, until v5.9_2 worked fine.

**FreeBSD version**

    neel@concorde:~ % uname -a
FreeBSD concorde.neelc.org 14.0-CURRENT FreeBSD 14.0-CURRENT #1 main-n255824-20b2b464ebb: Thu May 26 00:03:21 PDT 2022     neel@concorde.neelc.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
    hostb0@pci0:0:0:0:    class=0x060000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a14 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = '11th Gen Core Processor Host Bridge/DRAM Registers'
        class      = bridge
        subclass   = HOST-PCI
    vgapci0@pci0:0:2:0:    class=0x030000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a49 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'TigerLake-LP GT2 [Iris Xe Graphics]'
        class      = display
        subclass   = VGA
    none0@pci0:0:4:0:    class=0x118000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a03 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'TigerLake-LP Dynamic Tuning Processor Participant'
        class      = dasp
    pcib1@pci0:0:7:0:    class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a23 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    pcib2@pci0:0:7:1:    class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x9a25 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    none1@pci0:0:8:0:    class=0x088000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a11 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'GNA Scoring Accelerator module'
        class      = base peripheral
    xhci0@pci0:0:13:0:    class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a13 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 USB Controller'
        class      = serial bus
        subclass   = USB
    none2@pci0:0:13:2:    class=0x0c0340 rev=0x01 hdr=0x00 vendor=0x8086 device=0x9a1b subvendor=0x2222 subdevice=0x1111
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Thunderbolt 4 NHI'
        class      = serial bus
        subclass   = USB
    pcib3@pci0:0:14:0:    class=0x010400 rev=0x00 hdr=0x00 vendor=0x8086 device=0x9a0b subvendor=0x8086 subdevice=0x0000
        vendor     = 'Intel Corporation'
        device     = 'Volume Management Device NVMe RAID Controller'
        class      = mass storage
        subclass   = RAID
    none3@pci0:0:18:0:    class=0x070000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0fc subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Integrated Sensor Hub'
        class      = simple comms
        subclass   = UART
    xhci1@pci0:0:20:0:    class=0x0c0330 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0ed subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller'
        class      = serial bus
        subclass   = USB
    none4@pci0:0:20:2:    class=0x050000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0ef subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Shared SRAM'
        class      = memory
        subclass   = RAM
    iwlwifi0@pci0:0:20:3:    class=0x028000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0f0 subvendor=0x8086 subdevice=0x0074
        vendor     = 'Intel Corporation'
        device     = 'Wi-Fi 6 AX201'
        class      = network
    ig4iic0@pci0:0:21:0:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e8 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Serial IO I2C Controller'
        class      = serial bus
    ig4iic1@pci0:0:21:1:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e9 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Serial IO I2C Controller'
        class      = serial bus
    none5@pci0:0:22:0:    class=0x078000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0e0 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Management Engine Interface'
        class      = simple comms
    pcib6@pci0:0:28:0:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0bd subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tigerlake PCH-LP PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    none6@pci0:0:29:0:    class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x09ab subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        class      = base peripheral
    isab0@pci0:0:31:0:    class=0x060100 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa082 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP LPC Controller'
        class      = bridge
        subclass   = PCI-ISA
    hdac0@pci0:0:31:3:    class=0x040100 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0c8 subvendor=0x103c subdevice=0x87f6
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP Smart Sound Technology Audio Controller'
        class      = multimedia
        subclass   = audio
    ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0a3 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP SMBus Controller'
        class      = serial bus
        subclass   = SMBus
    none7@pci0:0:31:5:    class=0x0c8000 rev=0x20 hdr=0x00 vendor=0x8086 device=0xa0a4 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP SPI Controller'
        class      = serial bus
    pcib4@pci65532:224:29:0:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0b0 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        device     = 'Tiger Lake-LP PCI Express Root Port'
        class      = bridge
        subclass   = PCI-PCI
    pcib5@pci65532:224:29:2:    class=0x060400 rev=0x20 hdr=0x01 vendor=0x8086 device=0xa0b2 subvendor=0x103c subdevice=0x87f7
        vendor     = 'Intel Corporation'
        class      = bridge
        subclass   = PCI-PCI
    nvme0@pci65532:225:0:0:    class=0x010802 rev=0x03 hdr=0x00 vendor=0x8086 device=0x0975 subvendor=0x8086 subdevice=0x8410
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM
    nvme1@pci65532:226:0:0:    class=0x010802 rev=0x00 hdr=0x00 vendor=0x8086 device=0x0975 subvendor=0x8086 subdevice=0x8510
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM
    rtsx0@pci0:87:0:0:    class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x103c subdevice=0x87f7
        vendor     = 'Realtek Semiconductor Co., Ltd.'
        device     = 'RTS525A PCI Express Card Reader'

</details>

**DRM KMOD version**

Both `drm-510-kmod 5.10.113` and `e229eef88b2dea3eab966b1ca52009682733ef3` as of 5/26.

**To Reproduce**
Steps to reproduce the behavior:

Take a Intel TigerLake system using DPCD backlight

The TigerLake HP Spectre x360 14 is a good example as I use this system. I can't speak for the newer AlderLake model, however.

**Screenshots**

N/A

**Additional context**

System: HP Spectre x360 14-ea0023dx. This system requires the DPCD backlight. Backlight got broken with the drm-kmod 5.10 update. Don't know if it's DPCD or not, everything else I have (personal and work) uses 8th Gen.

evadot, 2022-06-01:
Could you bisect ?

neelchauhan, 2022-06-19:
Sorry for taking so long to respond. I cannot bisect as 5.9 does not compile on recent CURRENTs.

uqs, 2022-12-09:
I too have non-working acpi_video after upgrading my 13-STABLE system overnight.

/boot/kernel.old is from Aug 20th and I was running the fbsd13-kmod thing from Aug20th as well. Now I got a kernel from today and drm-510-kmod-5.10.113_8

This is on a Thinkpad T490

evadot, 2022-12-09:
> 

Use backlight(8)

uqs, 2022-12-09:
Perfect, thank you!

evadot, 2023-03-29:
@neelchauhan do you still have this problem ?

evadot, 2024-01-05:
Closing as we have no news from original reporter.

closed 2024-01-05 by evadot