freebsd/drm-kmod #175: amdgpu: Acceleration disabled by VT switch

Math2, 2022-06-07:
**Describe the bug**
After switching to a text mode VT and switching back to X's VT, OpenGL acceleration no longer works and programs are falling back to software rendering.

glxgears and other programs will show:
```
libGL error: failed to authenticate magic 1
libGL error: failed to load driver: radeonsi
```

No kernel messages.

Restarting Xorg restores acceleration.

**FreeBSD version**
13.1-RELEASE amd64

**PCI Info**
```
vgapci0@pci0:9:0:0:     class=0x030000 rev=0xcf hdr=0x00 vendor=0x1002 device=0x67ff subvendor=0x1462 subdevice=0x8a91
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Baffin [Radeon RX 550 640SP / RX 560/560X]'
    class      = display
    subclass   = VGA
hdac0@pci0:9:0:1:       class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0xaae0 subvendor=0x1462 subdevice=0xaae0
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Baffin HDMI/DP Audio [Radeon RX 550 640SP / RX 560/560X]'
    class      = multimedia
    subclass   = HDA
```

**DRM KMOD version**
Recent drm-kmod/5.10-lts (f7cc9dd58ca097a428ffcacad1686d970ca633a8).

evadot, 2022-06-07:
Can't reproduce here on the same card but FreeBSD 14-CURRENT.
Also I have https://reviews.freebsd.org/D35269
Mind testing if that changes something for you ?

Math2, 2022-06-07:
> Can't reproduce here on the same card but FreeBSD 14-CURRENT. Also I have https://reviews.freebsd.org/D35269 Mind testing if that changes something for you ?

No luck. I was running with no xf86-video-amdgpu at all at first. Then I installed 19.1.0_2 when I noticed the problem in case it might help. Then tried this 22.0.0 update. No noticeable difference (but the new version seems to work just fine apart from that one issue).

I tried starting X with twm instead of XFCE too to see if it might be related to the WM. No difference there either.

I guess I could try downgrading to drm-kmod/5.4-lts.

I wonder if there might be something setup weird with this computer but I can't think of anything.

jwp, 2022-12-17:
I'm seeing the same thing on a recent 13.1-STABLE build(tracking stable/13, amd64).

drm-510-kmod 5.10.113_8
drm-kmod 20220907_1
Ellesmere RX 580 (polaris10).

lantw44, 2022-12-18:
I saw the same problem on FreeBSD 13.1-RELEASE-p5. Switching VT caused `glxgears` to show:
```
libGL error: failed to authenticate magic 1
libGL error: failed to load driver: i965
```

Downgrading from drm-510-kmod-5.10.113_8 to drm-54-kmod-5.4.191_1 fixed the problem. This machine uses the integrated graphics of Intel i5-650.

O1d, 2023-01-18:
Same issue here with intel hd graphics 4600 on freshly built stable/13 and fresh drm-510-kmod from ports (5.10.163, but the behavior was the same on 5.10.113 as well)
after X started, glxinfo -B shows the correct device, acceleration and works just fine:
```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel (0x8086)
    Device: Mesa Intel(R) HD Graphics 4600 (HSW GT2) (0x416)
    Version: 22.3.3
    Accelerated: yes
```

But after simply switching to console and back it gives the same magic error and falls back to llvmpipe:
```
libGL error: failed to authenticate magic 1
libGL error: failed to load driver: i965
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa/X.org (0xffffffff)
    Device: llvmpipe (LLVM 15.0.7, 256 bits) (0xffffffff)
    Version: 22.3.3
    Accelerated: no
```
Luckily, the applications which are already using acceleration during the VT switch, appear to continue working and getting acceleration just fine, but anything started after the switch uses llvmpipe instead

Partmedia, 2023-03-08:
Same issue here on 13.1-RELEASE (amd64) running drm-510-kmod-5.10.163_2 with Intel(R) UHD Graphics 630 (CFL GT2).

TheBoctor, 2023-05-01:
Exact same outputs from glxgears/glxinfo here, with i915kms on both 13.2-RELEASE and on 14.0-CURRENT. Tested on a ThinkPad X230 and a Dell Latitude 7490 (3rd and 8th gen Intel, respectively). Acceleration is perfectly behaved until changing VTs, then gone the moment you change back, for all new X clients.

Disabling the automatic VT switch when entering C3/suspend in `loader.conf` works around the issue in my use case:
`kern.vt.suspendswitch=0`

But this certainly doesn't seem like intended behavior. Is there a piece of documentation I've missed, or a vital change I should be making to a "kern" or "vt" tunable?

grahamperrin, 2023-07-31:
```text
% date ; glxinfo -B | grep Accelerated
Mon 31 Jul 2023 03:57:56 BST
    Accelerated: yes
% date ; glxinfo -B | grep Accelerated
Mon 31 Jul 2023 03:58:01 BST
libGL error: failed to authenticate magic 1
libGL error: failed to load driver: r600
    Accelerated: no
% 
```

- a switch between the two runs. 

[AMD Thames [Radeon HD 7550M/7570M/7650M]](https://bsd-hardware.info/?probe=2619fadb11#pci:1002-6841-103c-17a9), FreeBSD 14.0-CURRENT, drm-510-kmod. 

xf86-video-amdgpu not installed, because re: [the description](https://www.freshports.org/x11-drivers/xf86-video-amdgpu/#description), this hardware does not use amdgpu.

grahamperrin, 2024-01-10:
Cross-reference: 

- FreeBSD bug [267915 – x11-wm/mutter: after graphics/mesa-* 21.3.8 -> 22.2.3 update OpenGL compositing fails on Intel Iris Pro](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=267915)

jwp, 2024-05-19:
Not seeing this anymore with 14-STABLE.

```
FreeBSD stable/14-965f92a2829: Sun Mar 10 21:56:32 UTC 2024
AMD Radeon RX 580 Series (radeonsi, polaris10, LLVM 15.0.7, DRM 3.42, 14.0-STABLE)
drm-515-kmod-5.15.118_4
drm-kmod-20220907_3
```

glxgears, glxinfo, and vkcube-xcb are working here after a VT switch.

OlCe2, 2024-09-02:
I never had this problem with my RX 580 running drm-510-kmod.

However, I can confirm I'm still seeing it with a UHD Graphics 620, and drm-515-kmod or drm-61-kmod.

evadot, 2025-06-19:
> Short story, the following patch should fix the problem.
> 
> [drm-check-if-fpriv-is-the-current-master-of-the-devi.txt](https://github.com/user-attachments/files/20819923/drm-check-if-fpriv-is-the-current-master-of-the-devi.txt)
> 
> Long story, the issue is reproducible with the modesetting drivers in Xorg, not with the outdated xf86-video-intel or xf86-video-amdgpu. So proposing to add a section to Xorg to use "Intel" driver instead of the modesetting driver has nothing to do with the bug itself.
> 
> When switching to VT, modesetting_drv calls drmDropMaster and drmGetMaster when switching back to X.
> 
> Without the additional check proposed in the patch, the magic map of the real master of the device (which is Xorg) is cleaned up, programs using DRI were getting a different magic_map for a different master, and thus the magic_map of authenticated clients of the real master Xorg remains free, and so all clients get authentication error, loosing hardware acceleration. You can trace the idr_alloc/idr_find in drm_getmagic/drm_authmagic to see the problem yourself.
> 
> I've tested with drm-61-kmod on amd based system Thinkpad T495 and intel one X270 (without the amdgpu or intel xorg drivers obviously, using only modesetting_drv) On FreeBSD 14.3-RELEASE
> 
> PS: Linux upstream commit:
> 
> 0aae5920a84469: drm: Clear up master tracking booleans

Thanks a lot for fixing this, could you open a MR please ?

PR #359 (2025-06-19): drm: check if fpriv is the current master of the device
Fix incomplete check for the current master of the device, which breaks modesetting_drv when switching to VT and back to X.

This should be also cherry-picked to older drm code (drm-61, drm-515 and drm-510)

Fixes #175
commit e3b289de (2025-06-19): drm: check if fpriv is the current master of the device

Fixes #175
commit 030e65e4 (2025-06-19): drm: check if fpriv is the current master of the device

Fixes #175

(cherry picked from commit e3b289de36619d4830a88770041042ee7221e620)
commit acae77b9 (2025-06-19): drm: check if fpriv is the current master of the device

Fixes #175

(cherry picked from commit e3b289de36619d4830a88770041042ee7221e620)
closed 2025-06-19 by evadot