freebsd/drm-kmod #176: Fails to build on powerpc64* after starting to use DC_FP_START

pkubaj, 2022-06-13:
**Describe the bug**
There are a couple of issues.
https://github.com/freebsd/drm-kmod/blob/master/kconfig.mk#L55= is incorrect. That line doesn't do anything. I found that the correct check is:
`.if ${MACHINE_ARCH:Mpowerpc64*} != ""`
That causes defines to be added. The same is also probably relevant to riscv*.
```
Next `DC_FP_START` and `DC_FP_END` is defined in https://github.com/freebsd/drm-kmod/blob/master/drivers/gpu/drm/amd/display/dc/os_types.h#L53=, behind `CONFIG_DRM_AMD_DC_DCN`, but that is not enabled on powerpc64*.

Adding that causes plenty of errors for Linux-specific code:
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'cpu_has_feature' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                DC_FP_START();
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:62:6: note: expanded from macro 'DC_FP_START'
        if (cpu_has_feature(CPU_FTR_VSX_COMP)) { \
            ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: use of undeclared identifier 'CPU_FTR_VSX_COMP'
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:62:22: note: expanded from macro 'DC_FP_START'
        if (cpu_has_feature(CPU_FTR_VSX_COMP)) { \
                            ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'preempt_disable' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:63:3: note: expanded from macro 'DC_FP_START'
                preempt_disable(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'enable_kernel_vsx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:64:3: note: expanded from macro 'DC_FP_START'
                enable_kernel_vsx(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: use of undeclared identifier 'CPU_FTR_ALTIVEC_COMP'
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:65:29: note: expanded from macro 'DC_FP_START'
        } else if (cpu_has_feature(CPU_FTR_ALTIVEC_COMP)) { \
                                   ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'preempt_disable' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:66:3: note: expanded from macro 'DC_FP_START'
                preempt_disable(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'enable_kernel_altivec' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:67:3: note: expanded from macro 'DC_FP_START'
                enable_kernel_altivec(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: use of undeclared identifier 'CPU_FTR_FPU_UNAVAILABLE'
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:68:30: note: expanded from macro 'DC_FP_START'
        } else if (!cpu_has_feature(CPU_FTR_FPU_UNAVAILABLE)) { \
                                    ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'preempt_disable' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:69:3: note: expanded from macro 'DC_FP_START'
                preempt_disable(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:741:3: error: implicit declaration of function 'enable_kernel_fp' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:70:3: note: expanded from macro 'DC_FP_START'
                enable_kernel_fp(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:743:3: error: implicit declaration of function 'c
pu_has_feature' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                DC_FP_END();
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:74:6: note: expanded from macro 'DC_FP_END'
        if (cpu_has_feature(CPU_FTR_VSX_COMP)) { \
            ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:743:3: error: use of undeclared identifier 'CPU_FTR_VSX_COMP'
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:74:22: note: expanded from macro 'DC_FP_END'
        if (cpu_has_feature(CPU_FTR_VSX_COMP)) { \
                            ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:743:3: error: implicit declaration of function 'disable_kernel_vsx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:75:3: note: expanded from macro 'DC_FP_END'
                disable_kernel_vsx(); \
                ^
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/core/dc.c:743:3: error: implicit declaration of function 'preempt_enable' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
/usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_2/drivers/gpu/drm/amd/display/dc/os_types.h:76:3: note: expanded from macro 'DC_FP_END'
                preempt_enable(); \
                ^
[... 48 lines cut ...]
0003:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
0003:01:00.0 USB controller: Texas Instruments TUSB73x0 SuperSpeed USB 3.0 xHCI Host Controller (rev 02)
0004:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
0004:01:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM5719 Gigabit Ethernet PCIe
0004:01:00.1 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM5719 Gigabit Ethernet PCIe
0005:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
0005:01:00.0 PCI bridge: ASPEED Technology, Inc. AST1150 PCI-to-PCI Bridge (rev 04)
0005:02:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 41)
0030:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
0031:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
0031:01:00.0 Audio device: Creative Labs Sound Core3D [Sound Blaster Recon3D / Z-Series] (rev 01)
0032:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
0033:00:00.0 PCI bridge: IBM POWER9 Host Bridge (PHB4)
```

**DRM KMOD version**
drm-510-kmod 5.10.113_1

**To Reproduce**
make -C /usr/ports/graphics/drm-510-kmod

**Screenshots**

**Additional context**
I'm opening only an issue, since this is up for discussion.

unrelentingtech, 2022-06-17:
Yeah, `DC_FP_*` is used without ifdefs now, see #177 for my "aarch64 fix" that would allow building without DCN again.

For now, to get everything working without DCN, changing `DC_FP_*` to dummy impls is definitely enough. But it would be good to look into implementing `cpu_has_feature` / `(dis|en)able_kernel_<ppc fpu thingy>` soon.

evadot, 2022-12-13:
Is it still relevant ?

pkubaj, 2022-12-13:
Not really, thanks.