freebsd/drm-kmod #179: Loading i915kms.ko built with kernel (LOCAL_MODULES) would hang on Lenovo P51

delphij, 2022-06-19:
**Describe the bug**
Loading i915kms would cause screen go blank, and the kernel seems to stop responding after loading the kernel module.

This only happens when building with `LOCAL_MODULES` (by cloning the `drm-kmod` and `drm-kmod-firmware` under `/usr/local/sys/modules` and build/install kernel).

The module worked correctly when built separately.

Alternatively, reverting the following commits would also work correctly:

 - 07b2b0aefdb2798864d4b0ed7e3ce860c9471514
 - 08434ecbb433056fb16a3162ce591e64074f398c
 - 54c93a09f6657c8ca4453f8ea8927640a75576e8

**FreeBSD version**
FreeBSD p51.home.us.delphij.net 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n256199-0c7d914bcc1f: Sun Jun 19 01:05:00 PDT 2022     delphij@p51.home.us.delphij.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64

The kernel have some local modification which are unrelated

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x05 hdr=0x00 vendor=0x8086 device=0x5918 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:	class=0x060400 rev=0x05 hdr=0x01 vendor=0x8086 device=0x1901 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '6th-10th Gen Core Processor PCIe Controller (x16)'
    class      = bridge
    subclass   = PCI-PCI
vgapci1@pci0:0:2:0:	class=0x030000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x591d subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics P630'
    class      = display
    subclass   = VGA
none0@pci0:0:8:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa12f subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:	class=0x118000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa131 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Thermal Subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:	class=0x118000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa160 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Serial IO I2C Controller'
    class      = dasp
none1@pci0:0:22:0:	class=0x078000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa13a subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family MEI Controller'
    class      = simple comms
ahci0@pci0:0:23:0:	class=0x010601 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa102 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
pcib2@pci0:0:28:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa110 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:2:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa112 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:28:4:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa114 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:29:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa118 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:29:4:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa11c subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa154 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'CM238 Chipset LPC/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none2@pci0:0:31:2:	class=0x058000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa121 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Power Management Controller'
    class      = memory
hdac0@pci0:0:31:3:	class=0x040300 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa171 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'CM238 HD Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa123 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family SMBus'
    class      = serial bus
    subclass   = SMBus
em0@pci0:0:31:6:	class=0x020000 rev=0x31 hdr=0x00 vendor=0x8086 device=0x15e3 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (5) I219-LM'
    class      = network
    subclass   = ethernet
vgapci0@pci0:1:0:0:	class=0x030200 rev=0xa1 hdr=0x00 vendor=0x10de device=0x1436 subvendor=0x17aa subdevice=0x224d
    vendor     = 'NVIDIA Corporation'
    device     = 'GM206GLM [Quadro M2200 Mobile]'
    class      = display
    subclass   = 3D
ppt0@pci0:4:0:0:	class=0x028000 rev=0x78 hdr=0x00 vendor=0x8086 device=0x24fd subvendor=0x8086 subdevice=0x0010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 8265 / 8275'
    class      = network
rtsx0@pci0:63:0:0:	class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x17aa subdevice=0x224d
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS525A PCI Express Card Reader'

</details>

**DRM KMOD version**
revision 71ddbe8afd9b2027c37ea23f69113b6b0ce1002e

**To Reproduce**
Steps to reproduce the behavior:

 - Clone freebsd/drm-kmod and freebsd/drm-kmod-firmware into `/usr/local/sys/modules`
 - Build `GENERIC` kernel

**Screenshots**
The modules built with kernel LOCAL_MODULES appears to have more verbose outputs, but the screen goes blank very soon after loading, so I wasn't able to capture a legible screenshot.

**Additional context**
The kernel module built separately appears to work fine.

When all three commits mentioned in the bug description were reverted, kernel module built with LOCAL_MODULES would work correctly. When any of them were not reverted, the issue would reoccur.  It seems that I had to remove `/usr/obj/usr/src/amd64.amd64/sys/GENERIC/` in order to reliably build the module; if I didn't, the issue sometimes would not show up until after applying some additional commits.

evadot, 2022-06-21:
I don't use this features so it's unlikely that I will look at this bug.
That being said does that works with GENERIC-NODEBUG ?

delphij, 2022-06-26:
@evadot I can confirm that the unmodified driver as of a4948e1e5245b68749a8e2858612945d5d49bb4b worked fine when building with `GENERIC-NODEBUG`.