freebsd/drm-kmod #180: resume fails with drm-510-kmod (5.10.113_1 from pkg) on 14.0-CURRENT

dch, 2022-06-20:
**Describe the bug**

- resume doesn't resume

**FreeBSD version**

FreeBSD akai.skunkwerks.at 14.0-CURRENT FreeBSD 14.0-CURRENT main-n256170-0fb87a1ae221 GENERIC-NODEBUG amd64

based off 
[fc2e4f15a9047bbf546cd675ed590b88e54362bd
](https://cgit.freebsd.org/src/commit/?id=fc2e4f15a9047bbf546cd675ed590b88e54362bd)
 with a few unrelated commits on top
**PCI Info**

<details>
  <summary>pciconf -lv</summary>
hostb0@pci0:0:0:0:      class=0x060000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x5904 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x06 hdr=0x00 vendor=0x8086 device=0x5926 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Iris Plus Graphics 640'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:       class=0x118000 rev=0x03 hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d2f subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:  class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d31 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Thermal subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:    class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d60 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
ig4iic1@pci0:0:21:1:    class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d61 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
none1@pci0:0:22:0:      class=0x078000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3a subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP CSME HECI'
    class      = simple comms
pcib1@pci0:0:28:0:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d10 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:4:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d14 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:5:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d15 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:29:0:      class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d18 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:      class=0x060100 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d4e subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point LPC Controller/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none2@pci0:0:31:2:      class=0x058000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d21 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PMC'
    class      = memory
hdac0@pci0:0:31:3:      class=0x040380 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d71 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP HD Audio'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:    class=0x0c0500 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d23 subvendor=0x1028 subdevice=0x075b
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP SMBus'
    class      = serial bus
    subclass   = SMBus
iwm0@pci0:58:0:0:       class=0x028000 rev=0x78 hdr=0x00 vendor=0x8086 device=0x24fd subvendor=0x8086 subdevice=0x0010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 8265 / 8275'
    class      = network
rtsx0@pci0:59:0:0:      class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x1028 subdevice=0x075b
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS525A PCI Express Card Reader'
nvme0@pci0:60:0:0:      class=0x010802 rev=0x01 hdr=0x00 vendor=0x1179 device=0x0115 subvendor=0x1179 subdevice=0x0001
    vendor     = 'Toshiba Corporation'
    device     = 'XG4 NVMe SSD Controller'
    class      = mass storage
    subclass   = NVM
</details>

**DRM KMOD version**

drm-510-kmod 5.10.113_1
drm-kmod 20220501

**To Reproduce**

suspend. power light on laptop goes off.
press power button again, screen flickers briefly, but nothing else happens.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Has always works perfectly under 12.x, but not in 13.0/13.1, and not currently in CURRENT.

Further info/logs in https://gist.github.com/dch/463caaaf723eabf84cf678b618b2d206/revisions including details from 12.x boots.

dch, 2022-06-20:
after switching to 20220617 GENERIC snapshot & removing all config/tunables,
we have resume from/to console.

dmesg etc https://git.sr.ht/~dch/xps13/tree/14.0-CURRENT-GENERIC-20220617

evadot, 2022-06-20:
Closing this as it's unlikely that this is caused by drm-kmod.

closed 2022-06-20 by evadot

dch, 2022-06-20:
this is resolved by removing `hw.acpi.reset_video=1` from sysctl.conf. Now works on 13.1-RELEASE as well as CURRENT. Thanks @evadot for helping!