freebsd/drm-kmod #181: graphics/drm-510-kmod panic with amdgpu

jownit, 2022-06-21:
**Describe the bug**
Kernel panics when loading amdgpu and external monitor attached via USB-C dock

**FreeBSD version**
FreeBSD joxan 14.0-CURRENT FreeBSD 14.0-CURRENT #38 main-n256222-3cbf6518d6e-dirty: Tue Jun 21 16:29:47 CEST 2022     root@xxx:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d0 subvendor=0x1022 subdevice=0x15d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d1 subvendor=0x1022 subdevice=0x15d1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:1:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:1:7:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15db subvendor=0x5126 subdevice=0x17aa
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb3@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 0'
    class      = bridge
[... 339 lines cut ...]
vpanic() at vpanic+0xf0/frame 0xfffffe00b476a560
panic() at panic+0x43/frame 0xfffffe00b476a5c0
trap_fatal() at trap_fatal+0x387/frame 0xfffffe00b476a620
trap_pfault() at trap_pfault+0x4f/frame 0xfffffe00b476a680
calltrap() at calltrap+0x8/frame 0xfffffe00b476a680
--- trap 0xc, rip = 0xffffffff80bf4cb8, rsp = 0xfffffe00b476a750, rbp = 0xfffffe00b476a7f0 ---
_sx_xlock_hard() at _sx_xlock_hard+0x178/frame 0xfffffe00b476a7f0
compute_mst_dsc_configs_for_state() at compute_mst_dsc_configs_for_state+0x14f/frame 0xfffffe00b476aad0
amdgpu_dm_atomic_check() at amdgpu_dm_atomic_check+0x9a6/frame 0xfffffe00b476ab50
drm_atomic_check_only() at drm_atomic_check_only+0x3fc/frame 0xfffffe00b476abd0
drm_atomic_commit() at drm_atomic_commit+0x13/frame 0xfffffe00b476abf0
drm_client_modeset_commit_atomic() at drm_client_modeset_commit_atomic+0x158/frame 0xfffffe00b476ac60
drm_client_modeset_commit_locked() at drm_client_modeset_commit_locked+0x74/frame 0xfffffe00b476aca0
drm_client_modeset_commit() at drm_client_modeset_commit+0x21/frame 0xfffffe00b476acc0
drm_fb_helper_restore_fbdev_mode_unlocked() at drm_fb_helper_restore_fbdev_mode_unlocked+0x81/frame 0xfffffe00b476acf0
drm_fb_helper_hotplug_event() at drm_fb_helper_hotplug_event+0x133/frame 0xfffffe00b476ad10
drm_kms_helper_hotplug_event() at drm_kms_helper_hotplug_event+0x23/frame 0xfffffe00b476ad30
drm_dp_mst_link_probe_work() at drm_dp_mst_link_probe_work+0x2bd/frame 0xfffffe00b476ade0
linux_work_fn() at linux_work_fn+0xe3/frame 0xfffffe00b476ae40
taskqueue_run_locked() at taskqueue_run_locked+0x191/frame 0xfffffe00b476aec0
taskqueue_thread_loop() at taskqueue_thread_loop+0xc3/frame 0xfffffe00b476aef0
fork_exit() at fork_exit+0x7e/frame 0xfffffe00b476af30
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe00b476af30
--- trap 0xb8fcd0d4, rip = 0xb0bbd87bbcfbd87a, rsp = 0x9fac43229adc4336, rbp = 0x23b36f5537f36f18 ---
KDB: enter: panic

jownit, 2022-06-29:
I just updated my laptop and thought I could try drm-510-kmod again just for the sake of it.
To my surprise it works now...
drm-510-kmod-5.10.113_1 on 14.0-CURRENT #41 main-n256399-454adf0fa4d-dirty

dmesg:
[drm] amdgpu kernel modesetting enabled.
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] initializing kernel modesetting (RAVEN 0x1002:0x15D8 0x17AA:0x5126 0xD2).
drmn0: Trusted Memory Zone (TMZ) feature disabled as experimental (default)
[drm] register mmio base: 0xD0500000
[drm] register mmio size: 524288
[drm] add ip block number 0 <soc15_common>
[drm] add ip block number 1 <gmc_v9_0>
[drm] add ip block number 2 <vega10_ih>
[drm] add ip block number 3 <psp>
[drm] add ip block number 4 <gfx_v9_0>
[drm] add ip block number 5 <sdma_v4_0>
[drm] add ip block number 6 <powerplay>
[drm] add ip block number 7 <dm>
[drm] add ip block number 8 <vcn_v1_0>
drmn0: successfully loaded firmware image 'amdgpu/picasso_gpu_info.bin'
drmn0: Fetched VBIOS from VFCT
amdgpu: ATOM BIOS: 113-PICASSO-114
drmn0: successfully loaded firmware image 'amdgpu/picasso_sdma.bin'
[drm] VCN decode is enabled in VM mode
[drm] VCN encode is enabled in VM mode
[drm] JPEG decode is enabled in VM mode
[drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
drmn0: VRAM: 2048M 0x000000F400000000 - 0x000000F47FFFFFFF (2048M used)
drmn0: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF
drmn0: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
[drm] Detected VRAM RAM=2048M, BAR=2048M
[drm] RAM width 128bits DDR4
[TTM] Zone  kernel: Available graphics memory: 7259850 KiB
[TTM] Zone   dma32: Available graphics memory: 2097152 KiB
[TTM] Initializing pool allocator
[drm] amdgpu: 2048M of VRAM memory ready
[drm] amdgpu: 3072M of GTT memory ready.
[drm] GART: num cpu pages 262144, num gpu pages 262144
[drm] PCIE GART of 1024M enabled (table at 0x000000F400900000).
drmn0: successfully loaded firmware image 'amdgpu/picasso_asd.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_ta.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_pfp.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_me.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_ce.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_rlc.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_mec.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_mec2.bin'
amdgpu: hwmgr_sw_init smu backed is smu10_smu
drmn0: successfully loaded firmware image 'amdgpu/raven_dmcu.bin'
drmn0: successfully loaded firmware image 'amdgpu/picasso_vcn.bin'
[drm] Found VCN firmware Version ENC: 1.12 DEC: 2 VEP: 0 Revision: 1
drmn0: Will use PSP to load VCN firmware
[drm] reserve 0x400000 from 0xf47fc00000 for PSP TMR
drmn0: RAS: optional ras ta ucode is not available
drmn0: RAP: optional rap ta ucode is not available
[drm] kiq ring mec 2 pipe 1 q 0
[drm] DM_PPLIB: values for F clock
[drm] DM_PPLIB:  400000 in kHz, 2924 in mV
[drm] DM_PPLIB:  933000 in kHz, 3249 in mV
[drm] DM_PPLIB:  1067000 in kHz, 3924 in mV
[drm] DM_PPLIB:  1200000 in kHz, 4074 in mV
[drm] DM_PPLIB: values for DCF clock
[drm] DM_PPLIB:  300000 in kHz, 2924 in mV
[drm] DM_PPLIB:  600000 in kHz, 3249 in mV
[drm] DM_PPLIB:  626000 in kHz, 3924 in mV
[drm] DM_PPLIB:  654000 in kHz, 4074 in mV
[drm] Display Core initialized with v3.2.104!
[... 147 lines cut ...]
uaudio0: Record[0]: 16000 Hz, 1 ch, 16-bit S-LE PCM format, 2x2ms buffer.
uaudio0: No MIDI sequencer.
pcm4: <USB audio> on uaudio0
uaudio0: No HID volume keys found.
uaudio1 on uhub5
uaudio1: <ThinkPad Dock USB Audio> on usbus1
uaudio1: Play[0]: 48000 Hz, 2 ch, 24-bit S-LE PCM format, 2x2ms buffer.
uaudio1: Play[0]: 44100 Hz, 2 ch, 24-bit S-LE PCM format, 2x2ms buffer.
uaudio1: Record[0]: 48000 Hz, 2 ch, 24-bit S-LE PCM format, 2x2ms buffer.
uaudio1: Record[0]: 44100 Hz, 2 ch, 24-bit S-LE PCM format, 2x2ms buffer.
uaudio1: No MIDI sequencer.
pcm5: <USB audio> on uaudio1
uaudio1: No HID volume keys found.
uslcom0 on uhub5
uslcom0: <CP2102 USB to UART Bridge Controller> on usbus1
Security policy loaded: MAC/ntpd (mac_ntpd)
bridge0: promiscuous mode enabled
ngeth0: Ethernet address: 58:9c:fc:10:dd:0b
ngeth0: link state changed to UP
ngeth0: changing name to 'ng0_pyze'
lo0: link state changed to UP
[drm] pstate TEST_DEBUG_DATA: 0x3FFE0000
[drm] pstate TEST_DEBUG_DATA: 0x3FFE0000
[drm] pstate took longer than expected ~78us
[drm] pstate took longer than expected ~93us

evadot, 2022-06-29:
Mhm ok, maybe there is a race somewhere, does it works every time ?

jownit, 2022-06-29:
I've rebooted a few times now (I have 'kld_list="amdgpu"' in /etc/rc.conf)
I rebooted both with it docked and undocked and docking after booting
I tried cold booting, both docked and undocked
I tried undocking and redocking while running.
No more panics so far..
I feels stable, more stable than drm-5.4 even, which would occasionally panic while booting

evadot, 2022-06-29:
Ok seems good then, let's close this and if you encounter it again in the futur just re-open (if it's another panic please open a new bug).

closed 2022-06-29 by evadot