freebsd/drm-kmod #188: graphics/drm-510-kmod panic with amdgpu and dual external monitors

jownit, 2022-07-06:
**Describe the bug**
Kernel panics when loading amdgpu.ko and two external monitors connected via USB-C dock
The connected monitors have different resolutions; one 4K (3840x2160) and one 1920x1200

**FreeBSD version**
FreeBSD joxan 14.0-CURRENT FreeBSD 14.0-CURRENT #43 main-n256532-a4d6d3b8910-dirty: Mon Jul  4 12:35:37 CEST 2022     root@joxan:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d0 subvendor=0x1022 subdevice=0x15d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d1 subvendor=0x1022 subdevice=0x15d1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:1:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:1:7:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15db subvendor=0x5126 subdevice=0x17aa
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x17aa subdevice=0x5126
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb3@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 0'
[... 324 lines cut ...]
drm_client_modeset_commit() at drm_client_modeset_commit+0x21/frame 0xfffffe00b475bcc0
drm_fb_helper_restore_fbdev_mode_unlocked() at drm_fb_helper_restore_fbdev_mode_unlocked+0x81/frame 0xfffffe00b475bcf0
drm_fb_helper_hotplug_event() at drm_fb_helper_hotplug_event+0x133/frame 0xfffffe00b475bd10
drm_kms_helper_hotplug_event() at drm_kms_helper_hotplug_event+0x23/frame 0xfffffe00b475bd30
drm_dp_mst_link_probe_work() at drm_dp_mst_link_probe_work+0x2bd/frame 0xfffffe00b475bde0
linux_work_fn() at linux_work_fn+0xe3/frame 0xfffffe00b475be40
taskqueue_run_locked() at taskqueue_run_locked+0x191/frame 0xfffffe00b475bec0
taskqueue_thread_loop() at taskqueue_thread_loop+0xc3/frame 0xfffffe00b475bef0
fork_exit() at fork_exit+0x7e/frame 0xfffffe00b475bf30
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe00b475bf30
--- trap 0x59f9812f, rip = 0xd09eac85eefaa893, rsp = 0xe30b730f79c97bcb, rbp = 0xebb71fe5e7f71f61 ---
KDB: enter: panic

**Additional context**
If only one monitor (the 4K one) is connected, it works.
If the second monitor is plugged in after amdgpu is loaded, it does not panic and seems to works just fine, but with these messages:
fbd0 on drmn0
device_attach: fbd0 attach returned 6
lkpi_iic8: <LinuxKPI I2C> on drmn0
iicbus8: <Philips I2C bus> on lkpi_iic8
iic8: <I2C generic I/O> on iicbus8
iicsmb8: <SMBus over I2C bridge> on iicbus8
smbus9: <System Management Bus> on iicsmb8
smb9: <SMBus generic I/O> on smbus9
[drm ERROR :drm_dp_send_dpcd_read] mstb 0xfffff80023c95600 port 2: DPCD read on addr 0x60 for 1 bytes NAKed

evadot, 2022-07-25:
There is probably a race somewhere I guess then, but no idea where to start looking for now ...
Does it work with 5.4 ?

jownit, 2022-08-15:
> There is probably a race somewhere I guess then, but no idea where to start looking for now ... Does it work with 5.4 ?

drm-54-kmod does not build on -current.

Alix82, 2022-09-15:
I believe I'm hitting the same issue, Thinkpad T495 on a USB-C dock with 2 external monitors but on FreeBSD 13.1 release, drm-510-kmod 5.10.113 crashes immediately on kldload when the USB-C dock is plugged. If I load amdgpu before plugging the USB-C dock it loads okay, but as soon as I plug the dock with 2 external monitors connected, it crashes immediately.

Fortunately I still have drm-fbsd13-kmod-5.4.144.g20220223 code in /usr/local/sys/modules, building and using that instead solves the issue (with drm-fbsd13-kmod-5.4.144 I never had this issue before).

vmcore dump when loading drm-510-kmod shows:

Fatal trap 12: page fault while in kernel mode
cpuid = 6; apic id = 06
fault virtual address	= 0x2f
fault code		= supervisor read data, page not present
instruction pointer	= 0x20:0xffffffff8037bb8d
stack pointer	        = 0x28:0xfffffe00ff8c5c40
frame pointer	        = 0x28:0xfffffe00ff8c5cb0
code segment		= base 0x0, limit 0xfffff, type 0x1b
			= DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags	= interrupt enabled, resume, IOPL = 0
current process		= 79001 (devd)
trap number		= 12
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /usr/local/sys/modules/drm-kmod-drm_v5.10.113_7/drivers/gpu/drm/drm_atomic_helper.c:619
...
kernel trap 22 with interrupts disabled
                            WARNING acrtc_attach->pflip_status != AMDGPU_FLIP_NONE failed at /usr/local/sys/modules/drm-kmod-drm_v5.10.113_7/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c:736
...

The rest is of the backtrace is not very useful, it is the usual syscallenter->sys_fork->fork1->do_fork->trap_pfault sequence.

arvedarved, 2022-11-01:
Me Too.
Thinkpad E14 with Lenovo USB-C dock and two screens with different resolution.
`
FreeBSD lenovo 14.0-CURRENT FreeBSD 14.0-CURRENT #0 334fc46f0: Sun Oct 30 21:53:42 CET 2022     root@lenovo:/usr/obj/usr/src/amd64.amd64/sys/LENOVO amd64
`
DRM is drm-510-kmod-5.10.113_7

I don't have a coredump, but the ddb trace also at compute_mst_dsc_configs_for_state()

arvedarved, 2023-03-08:
I have tried with CURRENT from today with drm from GIT,
```
#13 _sx_xlock_hard (sx=sx@entry=0xfffff8024ccf3828, x=<optimized out>, 
    opts=opts@entry=0, 
    file=file@entry=0xffffffff83da6f59 "/usr/home/arved/GIT/drm-kmod/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_mst_types.c", line=line@entry=985)
    at /usr/src/sys/kern/kern_sx.c:686
#14 0xffffffff80bdb325 in _sx_xlock (sx=0xfffff8024ccf3828, opts=0, 
    file=0xffffffff83da6f59 "/usr/home/arved/GIT/drm-kmod/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_mst_types.c", line=985)
    at /usr/src/sys/kern/kern_sx.c:332
#15 0xffffffff83b8119d in compute_mst_dsc_configs_for_state ()
   from /boot/modules/amdgpu.ko
#16 0xffffffff83b8b953 in amdgpu_dm_atomic_check ()
```
I have a core file and can try stuff

arvedarved, 2023-03-08:
[display.txt](https://github.com/freebsd/drm-kmod/files/10925184/display.txt)

I checked with upstream and they removed the mutex locking around this function  (see https://lore.kernel.org/lkml/20221114221754.385090-5-lyude@redhat.com/ )

I did the same and no more panics and everythings seems to work.

Alix82, 2023-03-09:
I can also confirm that applying the patch to remove the mutex locking in compute_mst_dsc_configs_for_state solves the issue for me using drm-510-kmod-5.10.163_2 on FreeBSD 13.1

jownit, 2023-03-09:
+1
This works for me on 14-current

**AND** it made drm-515-kmod-5.15.25 work, where previously  it would crash with even one external monitor.

evadot, 2024-01-05:
Closing as it's fixed in master and 6.1-lts

closed 2024-01-05 by evadot