freebsd/drm-kmod #192: Regression: ascii85 removal makes Xorg applications significantly slower

delphij, 2022-07-28:
**Describe the bug**
It seems that ignoring ascii85 errors (698df7ccaf4e9b95204cd06d84ac67f2b1c23fbf and ec67438c45244ca830c122ed55a5055397fa5a74) made Xorg applications significantly slower.

**FreeBSD version**
`FreeBSD p51.home.us.delphij.net 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n257009-23a61626f172: Thu Jul 28 00:05:41 PDT 2022     delphij@p51.home.us.delphij.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64`

The kernel is derived from freebsd/freebsd-src@309a263fe6484658827ca28a922fad70a2b42499 with some unrelated changes.

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x05 hdr=0x00 vendor=0x8086 device=0x5918 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:	class=0x060400 rev=0x05 hdr=0x01 vendor=0x8086 device=0x1901 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '6th-10th Gen Core Processor PCIe Controller (x16)'
    class      = bridge
    subclass   = PCI-PCI
vgapci1@pci0:0:2:0:	class=0x030000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x591d subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'HD Graphics P630'
    class      = display
    subclass   = VGA
none0@pci0:0:8:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa12f subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:	class=0x118000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa131 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Thermal Subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:	class=0x118000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa160 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Serial IO I2C Controller'
    class      = dasp
none1@pci0:0:22:0:	class=0x078000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa13a subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family MEI Controller'
    class      = simple comms
ahci0@pci0:0:23:0:	class=0x010601 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa102 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode]'
    class      = mass storage
    subclass   = SATA
pcib2@pci0:0:28:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa110 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:2:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa112 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:28:4:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa114 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:29:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa118 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:29:4:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0xa11c subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa154 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'CM238 Chipset LPC/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none2@pci0:0:31:2:	class=0x058000 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa121 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family Power Management Controller'
    class      = memory
hdac0@pci0:0:31:3:	class=0x040300 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa171 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'CM238 HD Audio Controller'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x31 hdr=0x00 vendor=0x8086 device=0xa123 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = '100 Series/C230 Series Chipset Family SMBus'
    class      = serial bus
    subclass   = SMBus
em0@pci0:0:31:6:	class=0x020000 rev=0x31 hdr=0x00 vendor=0x8086 device=0x15e3 subvendor=0x17aa subdevice=0x224d
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (5) I219-LM'
    class      = network
    subclass   = ethernet
vgapci0@pci0:1:0:0:	class=0x030200 rev=0xa1 hdr=0x00 vendor=0x10de device=0x1436 subvendor=0x17aa subdevice=0x224d
    vendor     = 'NVIDIA Corporation'
    device     = 'GM206GLM [Quadro M2200 Mobile]'
    class      = display
    subclass   = 3D
ppt0@pci0:4:0:0:	class=0x028000 rev=0x78 hdr=0x00 vendor=0x8086 device=0x24fd subvendor=0x8086 subdevice=0x0010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 8265 / 8275'
    class      = network
rtsx0@pci0:63:0:0:	class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x525a subvendor=0x17aa subdevice=0x224d
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS525A PCI Express Card Reader'

</details>

**DRM KMOD version**
Not applicable; kmod from git 833208195d4880b95ce4fb01a09e48af50813ab1 .

**To Reproduce**
Steps to reproduce the behavior:
build and install kernel (drm and drm-firmware source installed to /usr/local/sys/modules/ and picked up by `make buildkernel`) and reboot, start Xorg, attempt to start `mate-terminal` and `chrome`.

**Additional context**

Currently, I had to revert the following revisions to make kernel work (it's standard GENERIC kernel from -CURRENT with debugging options enabled):

 - 07b2b0aefdb2798864d4b0ed7e3ce860c9471514.
 - 08434ecbb433056fb16a3162ce591e64074f398c.
 - 54c93a09f6657c8ca4453f8ea8927640a75576e8.
 - ec67438c45244ca830c122ed55a5055397fa5a74. (new)
 - 698df7ccaf4e9b95204cd06d84ac67f2b1c23fbf. (new)

To be honest, the two revision looks quite harmless to me, they don't modify internal data structures of drm, etc. and I can't seem to be able to understand why they were causing issues. However, after several clean builds of kernel by reverting and re-instating the two revisions, it seems that they were indeed the culprit.

delphij, 2022-07-31:
It seems that this was related to `gnome-keyring-40.0_3`; removing the package makes Xorg to behave fine.  I've restored the ascii85 changes on my laptop for now.

evadot, 2022-08-01:
Ok that make more sense, feel free to close the bug if that was the case.

delphij, 2022-08-03:
Any idea why the ascii85 changes didn't play nice with gnome-keyring though? (gnome-keyring was happy with these reverted, so it's still a regression in my opinion...)

evadot, 2022-08-03:
> 

Ah ok I didn't understood you.
No I don't have any idea, are you sure it's really this ? This doesn't make any sense ...

evadot, 2022-12-13:
@delphij did you had another look at this by any chance ?

delphij, 2022-12-13:
Yes, the issues still exist; if I don't revert these changes locally, the screen will go blank after loading the kernel module.  I wonder if there is some timing races, as the changes looked pretty harmless to me.

delphij, 2022-12-14:
To be exact: cfad481295a0f2243de935cd9f3304fe4663dfbd is broken; I'm currently using fa49fe56504273d8bc8fce7caf3de296debf6172 with the reverts.

evadot, 2024-04-17:
@delphij did you ever find what was really the problem ?

delphij, 2024-04-17:
No, the recent versions worked fine for me, let's close this as not reproducible for now.

closed 2024-04-17 by delphij