freebsd/drm-kmod #193: Can't build in 14.0-CURRENT

itisshayan, 2022-07-30:
Hi, I've to say, thanks a lot for this driver. I used this driver for years. I fresh reinstalled FreeBSD-14.0-CURRENT on my notebook. the probem is, I can't build 5.7-stable at all: 
[image]

I also tried latest 5.10 and 5.10-lts branches. they successfully built. but, did not work. here is my output, when I ran kldload amdgpu (on 5.10 and 5.10-lts branches):

[image]

What exactly should I do? thanks.

Lenovo Ideapad Gaming 3
Ryzen 5 4600H with Radeon Vega 6 Graphics Driver and GTX-1650Ti 
FreeBSD-14.0-CURRENT

Alexander88207, 2022-07-30:
You are missing gpu-firmware-amd-kmod-renoir

itisshayan, 2022-07-30:
That problem kinda solved. thanks anyway. I installed FreeBSD 13.1-STABLE and successfully installed drm-510-kmod from the ports.
but It seems that I have very limited access. for example, I can't change my display's brightness. or, I can't change resolution or scale or even refresh rate. 
[image]
[image]

**output of kldstat:** 

```
shayan@freebsd:~ $ kldstat
Id Refs Address                Size Name
 1   73 0xffffffff80200000  1f38bc0 kernel
 2    1 0xffffffff82c00000   414220 amdgpu.ko
 3    2 0xffffffff82af9000    739e0 drm.ko
 4    3 0xffffffff82b6d000     5220 linuxkpi_gplv2.ko
 5    4 0xffffffff82b73000     72d8 dmabuf.ko
 6    1 0xffffffff82b7b000     c758 ttm.ko
 7    1 0xffffffff82b88000     3378 acpi_wmi.ko
 8    1 0xffffffff82b8c000     3218 intpm.ko
 9    1 0xffffffff82b90000     2180 smbus.ko
10    1 0xffffffff82b93000     2340 uhid.ko
11    1 0xffffffff82b96000     4350 ums.ko
12    1 0xffffffff82b9b000     3380 usbhid.ko
13    1 0xffffffff82b9f000     31f8 hidbus.ko
14    1 0xffffffff82ba3000     4d00 ng_ubt.ko
15    6 0xffffffff82ba8000     aac8 netgraph.ko
16    2 0xffffffff82bb3000     a238 ng_hci.ko
17    4 0xffffffff82bbe000     25a8 ng_bluetooth.ko
18    1 0xffffffff82bc1000     e250 ng_l2cap.ko
19    1 0xffffffff82bd0000    1bee8 ng_btsocket.ko
20    1 0xffffffff82bec000     39c0 ng_socket.ko
22    1 0xffffffff82bf5000     3380 if_urndis.ko
23    1 0xffffffff82bf9000     3178 uether.ko
24    1 0xffffffff82bf0000     4350 acpi_video.ko
```

I tried reloading acpi_video module, no chance. can I do anything? sorry for opening this issue anyway, thanks.

closed 2022-07-30 by itisshayan