freebsd/drm-kmod #194: Intel GVT-g virtual machine guest with FreeBSD-14.0-CURRENT fails

markshank, 2022-07-31:
**Describe the bug**
Intel GVT-g virtual machine guest with FreeBSD-14.0-CURRENT fails

**FreeBSD version**
mds@freebsd14:~ $ uname -a
FreeBSD freebsd14 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n256882-8f733dabcc3: Fri Jul 22 08:31:37 UTC 2022     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
mds@freebsd14:~ $

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
mds@freebsd14:~ $ pciconf -vl
hostb0@pci0:0:0:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x29c0 subvendor=0x1af4 subdevice=0x1100
    vendor     = 'Intel Corporation'
    device     = '82G33/G31/P35/P31 Express DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:2:0:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:2:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:2:2:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:2:3:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:2:4:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:2:5:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:2:6:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:2:7:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib9@pci0:0:3:0:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib10@pci0:0:3:1:      class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
    vendor     = 'Red Hat, Inc.'
    device     = 'QEMU PCIe Root port'
    class      = bridge
    subclass   = PCI-PCI
pcib11@pci0:0:3:2:      class=0x060400 rev=0x00 hdr=0x01 vendor=0x1b36 device=0x000c subvendor=0x1b36 subdevice=0x0000
[... 293 lines cut ...]

[  4349.263] (EE) open /dev/dri/card0: No such file or directory
[  4349.263] (WW) Falling back to old probe method for modesetting
[  4349.263] (EE) open /dev/dri/card0: No such file or directory
[  4349.263] (WW) Falling back to old probe method for scfb
[  4349.263] scfb trace: probe start
[  4349.263] (II) scfb(1): using default device
[  4349.263] scfb trace: probe done
[  4349.263] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[  4349.263] (EE) Screen 0 deleted because of no matching config section.
[  4349.263] (II) UnloadModule: "modesetting"
[  4349.263] (EE) 
Fatal server error:
[  4349.263] (EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devices
[  4349.263] (EE) 
[  4349.263] (EE) 
Please consult the The [X.Org](http://x.org/) Foundation support 
         at http://wiki.x.org/
 for help. 
[  4349.263] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[  4349.263] (EE) 
[  4349.263] (EE) Server terminated with error (1). Closing log file.
root@freebsd14:/var/log #

</details>

markshank, 2022-07-31:
### GVTg roadmap

https://github.com/intel/gvt-linux/wiki/GVTg-New-Architecture-Introduction-Update#collaboration-between-different-kernel-subsystems-and-vendors

evadot, 2022-08-01:
We do not enable CONFIG_DRM_I915_GVT right now, not sure if anyone ever attempted (even if that would be good to have this working).

markshank, 2022-08-01:
OK, thank you.  That explains all the error messages.  It was truly acting like it had the physical device.