freebsd/drm-kmod #196: Support for AST VGA chips possible?

georghuber, 2022-08-07:
Hi everyone, and sorry, for I am not technically reporting a bug here but I'd like to know whether it would be possible to include support for the AST VGA chips in drm-kmod.

AST VGA chips are used on many server mainboards.

From what I think I understand, drm-kmod relies on the linux drm. I have found on github a drm linux driver for the AST chips (https://github.com/torvalds/linux/tree/master/drivers/gpu/drm/ast). Would that mean that this driver could also be "portet" to FreeBSD via the drm-kmod package?

Please forgive me, it is probably really naive, but it sucks to have the console driver VGA stuck at 640x480, and my old server (with radeonkms) could easily do 1024x768 or higher.

Is there a way this could be implemented?

Many thanks for your answer,

Georg

**FreeBSD version**
FreeBSD BSD.local 13.1-RELEASE FreeBSD 13.1-RELEASE releng/13.1-n250148-fc952ac2212 GENERIC amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
pcib5@pci0:4:0:0:       class=0x060400 rev=0x02 hdr=0x01 vendor=0x1a03 device=0x1150 subvendor=0x1a03 subdevice=0x1150
    vendor     = 'ASPEED Technology, Inc.'
    device     = 'AST1150 PCI-to-PCI Bridge'
    class      = bridge
    subclass   = PCI-PCI
vgapci0@pci0:5:0:0:     class=0x030000 rev=0x21 hdr=0x00 vendor=0x1a03 device=0x2000 subvendor=0x1043 subdevice=0x84eb
    vendor     = 'ASPEED Technology, Inc.'
    device     = 'ASPEED Graphics Family'
    class      = display
    subclass   = VGA

</details>

wulf7, 2022-09-03:
There is 2 years old ast branch in the repo. You may try to test it. Not sure if it ever worked or even compiles now

DanielO, 2022-09-14:
I tested this branch in July 2021, I had to patch it (https://gist.github.com/DanielO/3a8cfd1e955e4e399a51bb75f1d6f1dd) to avoid a panic (https://gist.github.com/DanielO/a3aa3440120c1cee260d66233d23219b).

After that my notes say it loaded and X started but lightdm just showed a black screen (mouse moved though).

I haven't tried it since.

DanielO, 2022-09-14:
Note that the Xorg driver does work for this chipset (now it's been patched to fix an infinite loop).

Performance is pretty terrible though, so make sure you disable composition :)