freebsd/drm-kmod #198: AMD Ryzen 5 4500U (Renoir) not working

hurzl, 2022-08-14:
I can start X, mouse and keyboard are working, but I only get a black screen with some randomly colored squares (and mouse pointer)

FreeBSD 13.1-STABLE or 14.0-CURRENT
drm-kmod: both 5.10-lts and master branch
Options in xorg.conf don't matter, except switching off Accel or AccelMethod. 
Linux works fine since at least `(DRM 3.39.0, 5.9.0-3-amd64, LLVM 11.0.0)`

```
vgapci0@pci0:4:0:0:	class=0x030000 rev=0xc3 hdr=0x00 vendor=0x1002 device=0x1636 subvendor=0x17aa subdevice=0x507f
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Renoir'
    class      = display
    subclass   = VGA
```

```[ 73139.668] 
X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
[ 73139.668] Current Operating System: FreeBSD thinkpad 13.1-STABLE FreeBSD 13.1-STABLE #15 stable/13-n252092-d1cd71da2b3: Wed Aug 10 21:21:37 CEST 2022     root@thinkpad:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
[ 73139.668]  
[ 73139.668] Current version of pixman: 0.40.0
[ 73139.668] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[ 73139.668] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 73139.669] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Aug 12 22:25:59 2022
[ 73139.671] (==) Using config file: "/etc/X11/xorg.conf"
[ 73139.671] (==) Using config directory: "/usr/local/etc/X11/xorg.conf.d"
[ 73139.671] (==) Using system config directory "/usr/local/share/X11/xorg.conf.d"
[ 73139.672] (==) No Layout section.  Using the first Screen section.
[ 73139.672] (==) No screen section available. Using defaults.
[ 73139.672] (**) |-->Screen "Default Screen Section" (0)
[ 73139.672] (**) |   |-->Monitor "<default monitor>"
[ 73139.672] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[ 73139.672] (**) |   |-->Device "Card0"
[ 73139.672] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[ 73139.673] (==) Automatically adding devices
[ 73139.673] (==) Automatically enabling devices
[ 73139.673] (==) Not automatically adding GPU devices
[ 73139.673] (==) Automatically binding GPU devices
[ 73139.673] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 73139.676] (==) FontPath set to:
	/usr/local/share/fonts/misc/,
	/usr/local/share/fonts/TTF/,
	/usr/local/share/fonts/OTF/,
	/usr/local/share/fonts/Type1/,
	/usr/local/share/fonts/100dpi/,
	/usr/local/share/fonts/75dpi/,
	catalogue:/usr/local/etc/X11/fontpath.d
[ 73139.676] (==) ModulePath set to "/usr/local/lib/xorg/modules"
[ 73139.676] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[ 73139.676] (II) Module ABI versions:
[ 73139.676] 	X.Org ANSI C Emulation: 0.4
[ 73139.676] 	X.Org Video Driver: 25.2
[ 73139.676] 	X.Org XInput driver : 24.4
[ 73139.676] 	X.Org Server Extension : 10.0
[ 73139.676] (--) PCI:*(4@0:0:0) 1002:1636:17aa:507f rev 195, Mem @ 0x460000000/268435456, 0x470000000/2097152, 0xfd300000/524288, I/O @ 0x00001000/256, BIOS @ 0x????????/65536
[ 73139.677] (II) LoadModule: "glx"
[ 73139.679] (II) Loading /usr/local/lib/xorg/modules/extensions/libglx.so
[ 73139.681] (II) Module glx: vendor="X.Org Foundation"
[ 73139.681] 	compiled for 1.21.1.4, module version = 1.0.0
[ 73139.681] 	ABI class: X.Org Server Extension, version 10.0
[ 73139.681] (II) LoadModule: "amdgpu"
[ 73139.681] (II) Loading /usr/local/lib/xorg/modules/drivers/amdgpu_drv.so
[ 73139.682] (II) Module amdgpu: vendor="X.Org Foundation"
[ 73139.682] 	compiled for 1.21.1.4, module version = 22.0.0
[... 215 lines cut ...]
VT: Replacing driver "efifb" with new "fb".
start FB_INFO:
type=11 height=1080 width=1920 depth=32
pbase=0x3b0ce0000 vbase=0xfffffe0118e76000
name=drmn0 flags=0x0 stride=7680 bpp=32
end FB_INFO
drmn0: ring gfx uses VM inv eng 0 on hub 0
drmn0: ring comp_1.0.0 uses VM inv eng 1 on hub 0
drmn0: ring comp_1.1.0 uses VM inv eng 4 on hub 0
drmn0: ring comp_1.2.0 uses VM inv eng 5 on hub 0
drmn0: ring comp_1.3.0 uses VM inv eng 6 on hub 0
drmn0: ring comp_1.0.1 uses VM inv eng 7 on hub 0
drmn0: ring comp_1.1.1 uses VM inv eng 8 on hub 0
drmn0: ring comp_1.2.1 uses VM inv eng 9 on hub 0
drmn0: ring comp_1.3.1 uses VM inv eng 10 on hub 0
drmn0: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
drmn0: ring sdma0 uses VM inv eng 0 on hub 1
drmn0: ring vcn_dec uses VM inv eng 1 on hub 1
drmn0: ring vcn_enc0 uses VM inv eng 4 on hub 1
drmn0: ring vcn_enc1 uses VM inv eng 5 on hub 1
drmn0: ring jpeg_dec uses VM inv eng 6 on hub 1
vgapci0: child drmn0 requested pci_get_powerstate
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
lkpi_iic3: <LinuxKPI I2C> on drm1
```

evadot, 2022-08-16:
Is it the full Xorg.log ?
Can you share the output of pkg info | grep xf86-video ?
And please share a Xorg.log without any xorg.conf
Also this doesn't seems to be the full dmesg, you're missing a line that looks something like "[drm] Initialized amdgpu 3.40.0 20150101 for drmn0 on minor 0"

hurzl, 2022-08-16:
```
>  pkg info | grep xf86-video
xf86-video-amdgpu-22.0.0_1     X.Org amdgpu display driver
xf86-video-dummy-0.3.8_4       X.Org dummy display driver
xf86-video-intel-2.99.917.916_2,1 X.Org legacy driver for Intel integrated graphics chipsets
xf86-video-mach64-6.9.7        X.Org mach64 display driver
xf86-video-nv-2.1.21_5         X.Org nv display driver
xf86-video-r128-6.12.0_3       X.Org r128 display driver
xf86-video-vesa-2.5.0_1        X.Org vesa display driver
```

```
>  dmesg | grep amdgpu
[drm] amdgpu kernel modesetting enabled.
amdgpu: ATOM BIOS: 113-RENOIR-025
drmn0: successfully loaded firmware image 'amdgpu/renoir_sdma.bin'
[drm] amdgpu: 2048M of VRAM memory ready
[drm] amdgpu: 3072M of GTT memory ready.
drmn0: successfully loaded firmware image 'amdgpu/renoir_asd.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_pfp.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_me.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_ce.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_rlc.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_mec.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_mec2.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_dmcub.bin'
drmn0: successfully loaded firmware image 'amdgpu/renoir_vcn.bin'
[drm] Initialized amdgpu 3.40.0 20150101 for drmn0 on minor 0

```

I thought the keyboard/mouse stuff wouldn't be necessary. 
Here's the complete Xorg.log **without** config (which gives the same picture)

```
[ 64728.241] 
X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
[ 64728.241] Current Operating System: FreeBSD thinkpad 13.1-STABLE FreeBSD 13.1-STABLE #15 stable/13-n252092-d1cd71da2b3: Wed Aug 10 21:21:37 CEST 2022     root@thinkpad:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
[ 64728.241]  
[ 64728.241] Current version of pixman: 0.40.0
[ 64728.242] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[ 64728.242] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 64728.242] (==) Log file: "/var/log/Xorg.1.log", Time: Tue Aug 16 14:36:15 2022
[ 64728.242] (==) Using config file: "/etc/X11/xorg.conf"
[ 64728.242] (==) Using system config directory "/usr/local/share/X11/xorg.conf.d"
[ 64728.243] (==) No Layout section.  Using the first Screen section.
[ 64728.243] (==) No screen section available. Using defaults.
[ 64728.243] (**) |-->Screen "Default Screen Section" (0)
[ 64728.243] (**) |   |-->Monitor "<default monitor>"
[ 64728.243] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[ 64728.243] (==) Automatically adding devices
[ 64728.243] (==) Automatically enabling devices
[ 64728.243] (==) Not automatically adding GPU devices
[ 64728.243] (==) Automatically binding GPU devices
[ 64728.243] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 64728.243] (==) FontPath set to:
	/usr/local/share/fonts/misc/,
	/usr/local/share/fonts/TTF/,
	/usr/local/share/fonts/OTF/,
	/usr/local/share/fonts/Type1/,
	/usr/local/share/fonts/100dpi/,
	/usr/local/share/fonts/75dpi/,
	catalogue:/usr/local/etc/X11/fontpath.d
[ 64728.243] (==) ModulePath set to "/usr/local/lib/xorg/modules"
[ 64728.243] (II) The server relies on udev to provide the list of input devices.
[... 367 lines cut ...]
[ 64728.458] (II) event8  - ACPI video extension: device is a keyboard
[ 64728.459] (II) event8  - ACPI video extension: device removed
[ 64728.459] (**) Option "config_info" "udev:/dev/input/event8"
[ 64728.459] (II) XINPUT: Adding extended input device "ACPI video extension" (type: KEYBOARD, id 12)
[ 64728.459] (**) Option "xkb_rules" "evdev"
[ 64728.460] (II) event8  - ACPI video extension: is tagged by udev as: Keyboard
[ 64728.460] (II) event8  - ACPI video extension: device is a keyboard
[ 64728.653] (II) modeset(0): Disabling kernel dirty updates, not required.
[ 64737.334] (II) event0  - System mouse: device removed
[ 64737.334] (II) event1  - System keyboard multiplexer: device removed
[ 64737.335] (II) event2  - Power Button: device removed
[ 64737.335] (II) event4  - Sleep Button: device removed
[ 64737.335] (II) event5  - AT keyboard: device removed
[ 64737.335] (II) event7  - ETPS/2 Elantech TrackPoint: device removed
[ 64737.335] (II) event8  - ACPI video extension: device removed
[ 64737.335] (II) AIGLX: Suspending AIGLX clients for VT switch
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.897] (II) UnloadModule: "libinput"
[ 64741.899] (II) Server terminated successfully (0). Closing log file.
```

evadot, 2022-08-16:
What's in [ 65958.065] (==) Using config file: "/etc/X11/xorg.conf" ?

hurzl, 2022-08-16:
Ok sorry but that's all commented lines. No content.

hurzl, 2022-08-16:
I get the same picture but without mouse pointer , instead a moving colored rectangle

evadot, 2022-08-16:
> I get the same picture but without mouse pointer , instead a moving colored rectangle

Then it's likely a drm related problem and not an xorg one but I have no idea how to help more right now.

hurzl, 2022-08-16:
So is there a difference to the Linux drm, since the same versions (DRM 3.40) behave differently, or where else?

hurzl, 2022-08-17:
By accident I came across a hint to `/boot/device.hints`, and removed it because it contained some very outdated things like hint.fd and hint.sc.
Now I have to kldload amdgpu by hand after boot, but the problem with X is gone. 
Haven't found out where the file came from ... is it a default file created when installing FreeBSD?

closed 2022-08-17 by hurzl