freebsd/drm-kmod #204: Build failure on 14-Current

robbak, 2022-09-12:
graphics/drm-510-kmod is not building on my 14-C machine. The error is the same as on this recent pkg-fallout message:

https://www.mail-archive.com/freebsd-pkg-fallout@freebsd.org/msg1990172.html

usr/ports/graphics/drm-510-kmod/work/drm-kmod-drm_v5.10.113_6/drivers/dma-buf/sync_file.c:394:3: error: implicit declaration of function 'cpu_relax' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                cpu_relax();
                ^

My version details (the error is the same as the pkg-fallout log)

uname -a -U
FreeBSD XXXX 14.0-CURRENT FreeBSD 14.0-CURRENT #5 main-n256715-ac19e54390a: Sun Jul 17 13:37:34 AEST 2022     root@XXXXX:/usr/obj/usr/src/amd64.amd64/sys/XXXXX amd64 1400063

robbak, 2022-09-14:
Updated my source tree and it built. So it was just because I had too old a source tree. Do we need another if ${OSVERSION} line in the makefile to alert users that have outdated kernel source?

evadot, 2022-09-14:
> Updated my source tree and it built. So it was just because I had too old a source tree. Do we need another if ${OSVERSION} line in the makefile to alert users that have outdated kernel source?

You had a really old tree, anything older than ~1 month for CURRENT isn't really supported.

closed 2022-09-14 by evadot