freebsd/drm-kmod #205: Add support few AMD devices

rozhuk-im, 2022-09-13:
**Describe the bug**
Add support 5750G APU.
Known issue: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=264305
Work good for me more than 2 month: web browser, games, video.

**FreeBSD version**
FreeBSD rimwks.local 13.1-STABLE FreeBSD 13.1-STABLE RIM_WKS amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
vgapci0@pci0:10:0:0:	class=0x030000 rev=0xd8 hdr=0x00 vendor=0x1002 device=0x1638 subvendor=0x1002 subdevice=0x1636
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Cezanne'
    class      = display
    subclass   = VGA

</details>

```
--- drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c	2022-02-23 16:59:52.000000000 +0300
+++ drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c	2022-03-01 02:46:20.449902000 +0300
@@ -1034,7 +1034,10 @@
 	{0x1002, 0x734F, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_NAVI14},
 
 	/* Renoir */
+	{0x1002, 0x15E7, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},
 	{0x1002, 0x1636, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU|AMD_EXP_HW_SUPPORT},
+	{0x1002, 0x1638, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},
+	{0x1002, 0x164C, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_RENOIR|AMD_IS_APU},
 
 	/* Navi12 */
 	{0x1002, 0x7360, PCI_ANY_ID, PCI_ANY_ID, 0, 0, CHIP_NAVI12|AMD_EXP_HW_SUPPORT},
```

rozhuk-im, 2022-09-13:
https://github.com/freebsd/drm-kmod/issues/118 - looks already includes 0x1638 into 5.7 but probably miss later.

evadot, 2022-09-13:
Upstream commits : 
https://github.com/torvalds/linux/commit/8bf0835132c19437e1530621b730dd4f29fe938e
https://github.com/torvalds/linux/commit/278cdb6834901658a81a1e22f5799aa15dca5029
https://github.com/torvalds/linux/commit/775da83005cb61d4c213c636df9337da05714ff1

The first two are in the 5.10 lts branch : https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c?h=linux-5.10.y#n1081

Which ones did you tested exactly ?

I'll probably take time to update the 5.10-lts branch to be in sync with the 5.10.142 upstream tag.

wulf7, 2022-09-13:
I have already imported these commits in to 5.10-lts. See https://github.com/freebsd/drm-kmod/blob/5.10-lts/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c#L1100

rozhuk-im, 2022-09-13:
```
vgapci0@pci0:10:0:0: class=0x030000 rev=0xd8 hdr=0x00 vendor=0x1002 device=0x1638 subvendor=0x1002 subdevice=0x1636
```

evadot, 2022-09-13:
> I have already imported these commits in to 5.10-lts. See https://github.com/freebsd/drm-kmod/blob/5.10-lts/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c#L1100

Meh, I didn't even checked our branch :)
So what's this issue about then ?

rozhuk-im, 2022-09-13:
Oops, sorry, l miss migrating drm-fbsd13-kmod -> drm-510-kmod.

closed 2022-09-13 by rozhuk-im