freebsd/drm-kmod #209: drm-510-kmod: Linuxulator GL_RENDERER reverts to llvmpipe after waking from suspend

hunter2k01, 2022-10-01:
**Describe the bug**
On my system using an RX 580 GPU, applications running inside the linux-c7 environment will be accelerated by the graphics card until waking up from suspend which changes the GL Renderer to llvmpipe.

**FreeBSD version**
FreeBSD desktop 13.1-RELEASE-p2 FreeBSD 13.1-RELEASE-p2 GENERIC amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1480 subvendor=0x1022 subdevice=0x1480
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1481 subvendor=0x1022 subdevice=0x1481
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib4@pci0:0:3:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:5:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:7:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:7:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1484 subvendor=0x1022 subdevice=0x1484
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B]'
    class      = bridge
    subclass   = PCI-PCI
hostb7@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib6@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1484 subvendor=0x1022 subdevice=0x1484
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B]'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x1458 subdevice=0x5001
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x1458 subdevice=0x5001
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb8@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1440 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse/Vermeer Data Fabric: Device 18h; Function 0'
    class      = bridge
    subclass   = HOST-PCI
hostb9@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1441 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse/Vermeer Data Fabric: Device 18h; Function 1'
    class      = bridge
    subclass   = HOST-PCI
hostb10@pci0:0:24:2:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1442 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse/Vermeer Data Fabric: Device 18h; Function 2'
    class      = bridge
    subclass   = HOST-PCI
hostb11@pci0:0:24:3:	class=0x060000 re
```
</details>

**DRM KMOD version**
drm-510-kmod 5.10.113_6

**To Reproduce**
Steps to reproduce the behavior:
* Have the linux-c7 metaport as well as the aforementioned setup with drm-510-kmod
* Make sure the Linuxulator points are mounted and running, I used the same mount parameters as the Handbook section
* Run `/compat/linux/bin/glxgears -info` and look for GL_RENDERER which should show the proper AMD card being used for acceleration
* Suspend with `zzz`
* Wake the system and run `/compat/linux/bin/glxgears -info` again which will now show the GL_RENDERER is llvmpipe

**Additional context**
I did not realize this was occurring until I noticed games would become slow and choppy in games/linux-steam-utils but only after suspending. If you logout (which should restart X) and log back into a graphical environment, it will fix the issue until the next time you suspend.

grahamperrin, 2022-10-02:
Originally reported as <https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=266745>

hunter2k01, 2022-10-02:
> Originally reported as https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=266745 with a little more detail, including _xf86-video-amdgpu_ …

I updated this one with that information. When he led me here I thought it was specific to drm-kmod, but worth mentioning anyway I suppose

hunter2k01, 2022-10-23:
Interestingly, in IceWM when I exit (which _does_ restart X) and come back the Linuxulator is still using llvmpipe. I have no idea what MATE did differently (from which I wrote this issue).

O1d, 2023-01-18:
Try switching to vt console (alt+ctrl+f1) and back (alt+f9), the result should be the same (llvmpipe as renderer), as reported in #175
So I believe this is the same bug