freebsd/drm-kmod #210: drm bug on Lenovo Thinkpad T490

evilgausus, 2022-10-08:
**Describe the bug**
There is an error while loading the module:

drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
drmn0: [drm] *ERROR* bridge device not found
drmn0: 0xfffffe010fee97c8Vdrmn0: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.device_attach: drmn0 attach returned 5

**FreeBSD version**
FreeBSD fenris 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n258503-ce9e720af2f: Sat Oct  8 10:11:54 CEST 2022     root@fenris:/usr/obj/usr/src/amd64.amd64/sys/FENRIS amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
root@fenris:~ # pciconf -lv
hostb0@pci0:0:0:0:      class=0x060000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x3e34 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Coffee Lake HOST and DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x3ea0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'WhiskeyLake-U GT2 [UHD Graphics 620]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:       class=0x118000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
none1@pci0:0:8:0:       class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
pchtherm0@pci0:0:18:0:  class=0x118000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df9 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Thermal Controller'
    class      = dasp
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9ded subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP USB 3.1 xHCI Controller'
    class      = serial bus
    subclass   = USB
none2@pci0:0:20:2:      class=0x050000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9def subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Shared SRAM'
    class      = memory
    subclass   = RAM
iwm0@pci0:0:20:3:       class=0x028000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df0 subvendor=0x8086 subdevice=0x0030
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP CNVi [Wireless-AC]'
    class      = network
ig4iic0@pci0:0:21:0:    class=0x0c8000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de8 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Serial IO I2C Controller'
    class      = serial bus
none3@pci0:0:22:0:      class=0x078000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP MEI Controller'
    class      = simple comms
none4@pci0:0:22:3:      class=0x070002 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de3 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Keyboard and Text (KT) Redirection'
    class      = simple comms
    subclass   = UART
pcib1@pci0:0:28:0:      class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db8 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:4:      class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9dbc subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:29:0:      class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:29:4:      class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db4 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:      class=0x060100 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9d84 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP LPC Controller'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:      class=0x040380 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9dc8 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
none5@pci0:0:31:4:      class=0x0c0500 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9da3 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP SMBus Controller'
    class      = serial bus
    subclass   = SMBus
none6@pci0:0:31:5:      class=0x0c8000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9da4 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP SPI Controller'
    class      = serial bus
em0@pci0:0:31:6:        class=0x020000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x15bd subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (6) I219-LM'
    class      = network
    subclass   = ethernet
rtsx0@pci0:1:0:0:       class=0xff0000 rev=0x01 hdr=0x00 vendor=0x10ec device=0x522a subvendor=0x17aa subdevice=0x2279
    vendor     = 'Realtek Semiconductor Co., Ltd.'
    device     = 'RTS522A PCI Express Card Reader'
pcib3@pci0:2:0:0:       class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:3:0:0:       class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:3:1:0:       class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:3:2:0:       class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
none7@pci0:4:0:0:       class=0x088000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x15bf subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 NHI (Low Power) [Alpine Ridge LP 2016]'
    class      = base peripheral
xhci1@pci0:58:0:0:      class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x15c1 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 USB 3.1 Controller (Low Power) [Alpine Ridge LP 2016]'
    class      = serial bus
    subclass   = USB
nvme0@pci0:61:0:0:      class=0x010802 rev=0x00 hdr=0x00 vendor=0x15b7 device=0x5006 subvendor=0x15b7 subdevice=0x5006
    vendor     = 'Sandisk Corp'
    device     = 'WD Black SN750 / PC SN730 NVMe SSD'
    class      = mass storage
    subclass   = NVM

</details>

**DRM KMOD version**

drm-510-kmod 5.10.113_7
drm-kmod 20220907_1

**To Reproduce**
 kldload /boot/modules/i915kms.ko

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

grahamperrin, 2022-10-08:
https://github.com/freebsd/drm-kmod/blob/1b8a46a045ec5f041319c69a2a7360ca051ac112/drivers/gpu/drm/i915/i915_drv.c#L91-L102

https://github.com/freebsd/drm-kmod/blob/master/drivers/gpu/drm/i915/i915_drv.c

grahamperrin, 2022-10-08:
<https://discord.com/channels/727023752348434432/831066226074976267/1028323447723528272>

```text
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x3ea0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'WhiskeyLake-U GT2 [UHD Graphics 620]'
    class      = display
    subclass   = VGA
```

----

For `PCI 8086:3ea0`, _WhiskeyLake-U GT2 [UHD Graphics 620]_: 

<https://bsd-hardware.info/?id=pci:8086-3ea0-17aa-2279?d=FreeBSD>

evadot, 2022-10-10:
Can you share your FENRIS kernel config please ?

uqs, 2022-12-09:
Hi, I have a T490 as well, I was running 13-STABLE and the previous fbsd13 kmod just fine for months. I upgraded to the latest stable last night, also had to move from the "fbsd13" port to `graphics/drm-510-kmod` and now acpi_video no longer provides any backlight knobs in sysctl. But anyway, at least Xorg works fine.

```
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
[drm] Got stolen memory base 0x7b800000, size 0x4000000
drmn0: could not load firmware image 'i915/kbl_dmc_ver1_04.bin'
drmn0: [drm] Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.
drmn0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915lkpi_iic0: <LinuxKPI I2C> on drmn0
iicbus0: <Philips I2C bus> on lkpi_iic0
iic0: <I2C generic I/O> on iicbus0
lkpi_iic1: <LinuxKPI I2C> on drmn0
iicbus1: <Philips I2C bus> on lkpi_iic1
iic1: <I2C generic I/O> on iicbus1
lkpi_iic2: <LinuxKPI I2C> on drmn0
iicbus2: <Philips I2C bus> on lkpi_iic2
iic2: <I2C generic I/O> on iicbus2
lkpi_iic3: <LinuxKPI I2C> on drmn0
iicbus3: <Philips I2C bus> on lkpi_iic3
iic3: <I2C generic I/O> on iicbus3
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
lkpi_iic4: <LinuxKPI I2C> on drm1
iicbus4: <Philips I2C bus> on lkpi_iic4
iic4: <I2C generic I/O> on iicbus4
lkpi_iic5: <LinuxKPI I2C> on drm2
iicbus5: <Philips I2C bus> on lkpi_iic5
iic5: <I2C generic I/O> on iicbus5
lkpi_iic6: <LinuxKPI I2C> on drm4
iicbus6: <Philips I2C bus> on lkpi_iic6
iic6: <I2C generic I/O> on iicbus6
[drm] Initialized i915 1.6.0 20200917 for drmn0 on minor 0
VT: Replacing driver "efifb" with new "fb".                                                                                                                     
start FB_INFO:
type=11 height=1080 width=1920 depth=32
pbase=0x80000000 vbase=0xfffff80080000000
name=drmn0 flags=0x0 stride=7680 bpp=32
end FB_INFO
acpi_video0: <ACPI video extension> on vgapci0
ig4iic0: <Intel Cannon Lake-LP I2C Controller-2> mem 0xa2745000-0xa2745fff at device 21.0 on pci0
ig4iic0: Using MSI
iicbus7: <Philips I2C bus (ACPI-hinted)> on ig4iic0
ugen0.2: <VIA Labs, Inc. USB2.0 Hub> at usbus0
uhub1 on uhub0
uhub1: <VIA Labs, Inc. USB2.0 Hub, class 9/0, rev 2.10/b.e0, addr 1> on usbus0

```

evadot, 2022-12-09:
Closing this bug, no news from original reporter for two months.

closed 2022-12-09 by evadot