freebsd/drm-kmod #212: Panic in amdgpu_get_vblank_counter_kms when resuming monitors from standby

franrogers, 2022-10-24:
**Describe the bug**
`drm-kmod` mostly works fine for me, but when I'm away from the computer, come back and move the mouse to wake up the monitors, frequently the machine will either immediately reboot, or scroll kernel panic messages endlessly until I hard reboot the system.

**FreeBSD version**
`FreeBSD 13.1-RELEASE-p2 GENERIC amd64`

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x1022 subdevice=0x1630
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
amdiommu0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x1022 subdevice=0x1631
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x1458 subdevice=0x5001
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x1458 subdevice=0x5001
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb4@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x166a subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Cezanne Data Fabric; Function 0'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x166b subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Cezanne Data Fabric; Function 1'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:24:2:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x166c subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Cezanne Data Fabric; Function 2'
[... 112 lines cut ...]
#1 0xffffffff80c1bb5f at vpanic+0x17f
#2 0xffffffff80c1b9d3 at panic+0x43
#3 0xffffffff80f4f678 at vm_fault+0x1298
#4 0xffffffff80f4e310 at vm_fault_trap+0xb0
#5 0xffffffff810afff3 at trap_pfault+0x1f3
#6 0xffffffff81087578 at calltrap+0x8
#7 0xffffffff83629350 at amdgpu_get_vblank_counter_kms+0x50
#8 0xffffffff8345f82b at drm_update_vblank_count+0x7b
#9 0xffffffff83462afc at drm_handle_vblank+0xac
#10 0xffffffff837437e0 at dm_crtc_high_irq+0xd0
#11 0xffffffff83734788 at amdgpu_dm_irq_handler+0x88
#12 0xffffffff83627492 at amdgpu_irq_dispatch+0xb2
#13 0xffffffff83626ac0 at amdgpu_ih_process+0xb0
#14 0xffffffff83626c70 at amdgpu_irq_handler+0x20
#15 0xffffffff80e60f47 at linux_irq_handler+0x27
#16 0xffffffff80bdbcfa at ithread_loop+0x25a
#17 0xffffffff80bd8a9e at fork_exit+0x7e
```

`xorg` versions:
```
xorg 7.7_3
xorg-drivers 7.7_7
xorg-server 21.1.4_1,1
```

ivan-volnov, 2022-10-31:
I also have this issue sometimes after waking up from suspend to ram.

[image]

It happen after updating drm-510-kmod from 5.10.113_1 to 5.10.113_6.

@evadot Do you have any ideas why?

evadot, 2024-04-17:
Is this still happening with 515 or 61 ?

grahamperrin, 2024-08-17:
Also, if reproducible: 

> … kernel panic messages endlessly …

There may be a surprisingly long wait before an automated reboot; some minutes. Be (very) patient.