freebsd/drm-kmod #216: i915 drm-510-kmod: hangs whenever external screen is attached

mkzol, 2022-11-20:
**Describe the bug**
Whenever an external screen is attached, the system hangs

**FreeBSD version**
FreeBSD t590 13.1-STABLE FreeBSD 13.1-STABLE #8 patch-bug-252288-c3f1ecb0d8: Fri Nov 18 22:35:23 CET 2022     root@t590:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
a.k.a patch https://reviews.freebsd.org/D36699 rebased on commit d3b97a1ea0 

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x3e34 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Coffee Lake HOST and DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x3ea0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'WhiskeyLake-U GT2 [UHD Graphics 620]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
none1@pci0:0:8:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
none2@pci0:0:18:0:	class=0x118000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df9 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Thermal Controller'
    class      = dasp
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9ded subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP USB 3.1 xHCI Controller'
    class      = serial bus
    subclass   = USB
none3@pci0:0:20:2:	class=0x050000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9def subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Shared SRAM'
    class      = memory
    subclass   = RAM
iwm0@pci0:0:20:3:	class=0x028000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df0 subvendor=0x8086 subdevice=0x0034
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP CNVi [Wireless-AC]'
    class      = network
none4@pci0:0:21:0:	class=0x0c8000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de8 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Serial IO I2C Controller'
    class      = serial bus
none5@pci0:0:22:0:	class=0x078000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP MEI Controller'
    class      = simple comms
pcib1@pci0:0:28:0:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db8 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:4:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9dbc subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:29:0:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:29:4:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db4 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9d84 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP LPC Controller'
    class      = bridge
    subclass   = PCI-ISA
hdac0@pci0:0:31:3:	class=0x040380 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9dc8 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
none6@pci0:0:31:4:	class=0x0c0500 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9da3 subvendor=0x17aa subdevice=0x2279
    device     = 'Cannon Point-LP SMBus Controller'
    class      = serial bus
    subclass   = SMBus
none7@pci0:0:31:5:	class=0x0c8000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9da4 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP SPI Controller'
    class      = serial bus
em0@pci0:0:31:6:	class=0x020000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x15be subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (6) I219-V'
    class      = network
    subclass   = ethernet
sdhci_pci0@pci0:1:0:0:	class=0x080501 rev=0x01 hdr=0x00 vendor=0x17a0 device=0x9750 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Genesys Logic, Inc'
    device     = 'GL9750 SD Host Controller'
    class      = base peripheral
    subclass   = SD host controller
pcib3@pci0:2:0:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:3:0:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:3:1:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:3:2:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
none8@pci0:4:0:0:	class=0x088000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x15bf subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 NHI (Low Power) [Alpine Ridge LP 2016]'
    class      = base peripheral
xhci1@pci0:58:0:0:	class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x15c1 subvendor=0x17aa subdevice=0x2279
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 USB 3.1 Controller (Low Power) [Alpine Ridge LP 2016]'
    class      = serial bus
    subclass   = USB
nvme0@pci0:61:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x144d device=0xa808 subvendor=0x144d subdevice=0xa801
    vendor     = 'Samsung Electronics Co Ltd'
    device     = 'NVMe SSD Controller SM981/PM981/PM983'
    class      = mass storage
    subclass   = NVM
root@t590:/usr/src # 

</details>

**DRM KMOD version**
drm-510-kmod 5.10.113_5
drm-510-kmod 5.10.113_6
drm-510-kmod 5.10.113_7
modules below drm-510-kmod 5.10.113_5 does not compile

**To Reproduce**
Steps to reproduce the behavior:
hook an external display

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
reverting binary modules to 5.4.191.g20220604 solves the problem. Can't compile these modules anymore.

mkzol, 2022-11-20:
[](url)

mkzol, 2022-11-20:
//    https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=267878

evadot, 2024-04-17:
Is this still happening with 515 or 61 ?