freebsd/drm-kmod #219: Are there any plans to support Alder/Raptor Lake?

ChomperT, 2022-12-06:
Hello, I found out that OpenBSD is already supported and many laptops on the market now use 12th/13th generation CPUs, is there any plan to support it in the near future?

oliver-giersch, 2022-12-06:
For what's it worth, I tried installing 14-CURRENT on a machine with an Alderlake i5 12400 with UHD 730 GPU but the DRM driver does not work. I also tried installing drm-kmod built from the master branch of this repo to no avail.

evadot, 2022-12-13:
AFAIK nobody (including me) is working on updating drm-kmod right now.

lin72h, 2023-01-02:
Happy 2023! Someone has been [working on it](https://github.com/christian-moerz/drm-kmod)

orbitz, 2023-02-04:
Hello!  As a recent owner of a 12th Gen Framework Laptop I'm highly motivated in supporting any work that leads to updating the graphics drivers.  I see the existing 5.14 work by @dumbbell , is there any effort for 5.15?  Is there any way to help if I don't have the software knowledge to make code changes?

dumbbell, 2023-02-04:
I would like the 5.13 (#224) and 5.14 (#226) PRs to be merged before starting 5.15 (which is planned as I need the driver myself).

Without coding, testing the current PRs is invaluable if you can, like using FreeBSD on your workstation on a daily basis. Non-obvious issues appear only after several hours of uptime sometimes.

orbitz, 2023-02-04:
@dumbbell Great!  I'll get this going on my setup.  I understand this branch is a bit buggy but I at least want to try it out so I can be ready for follow ups.  I'm already in a flow where I'm compiling the system by hand so I think I'm in a good position try run the changes.

And thank you for this hard work, I wish I could help out more.

dante-robinson, 2023-02-06:
OpenBSD supports up to 5.15 already in 7.2 and supposedly 6.1 in the next release if you wanted to try a different BSD in the mean time.

dumbbell, 2023-02-08:
The Linux 5.15 backport work in progress can be followed in #230.

dumbbell, 2023-02-16:
Updates to 5.15 and 5.16 are merged into `master`.

The Linux 5.17 backport work in progress can be followed in #236.

This version should bring stable support for Alder Lake P iGPUs.

dumbbell, 2023-08-24:
> For my test it doesn't support i7-1260P GPU.
> 
> drm-kmod on freebsd 14.0 alpha 2.

Hi!

i7-1260P is supported by `drm-515-kmod`. I don't think this is the package installed by the `drm-kmod` metaport. You probably need to install `drm-515-kmod` explicitly.

ykla, 2023-08-24:
> > For my test it doesn't support i7-1260P GPU.
> > drm-kmod on freebsd 14.0 alpha 2.
> 
> Hi!
> 
> i7-1260P is supported by `drm-515-kmod`. I don't think this is the package installed by the `drm-kmod` metaport. You probably need to install `drm-515-kmod` explicitly.

Actually, I pulled this GitHub project and compiled it for installation. However, I also installed `drm-515-kmod` according to your method, but the results are the same: when i915kms is loaded, the screen goes black, and the monitor displays "No HDMI signal input."

dumbbell, 2023-08-24:
Could you please share the output of dmesg(8) after loading the driver?

ykla, 2023-08-24:
> Could you please share the output of dmesg(8) after loading the driver?

```
root@ykla:/home/ykla # dmesg 
---<<BOOT>>---
Copyright (c) 1992-2023 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 14.0-ALPHA2 amd64 1400094 #0 main-n264841-77013f29d048: Fri Aug 18 08:13:49 UTC 2023
    root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
FreeBSD clang version 16.0.6 (https://github.com/llvm/llvm-project.git llvmorg-16.0.6-0-g7cbf1a259152)
WARNING: WITNESS option enabled, expect reduced performance.
VT(efifb): resolution 1920x1080
CPU: 12th Gen Intel(R) Core(TM) i7-1260P (2496.00-MHz K8-class CPU)
  Origin="GenuineIntel"  Id=0x906a3  Family=0x6  Model=0x9a  Stepping=3
  Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
  Features2=0x7ffafbff<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,SMX,EST,TM2,SSSE3,SDBG,FMA,CX16,xTPR,PDCM,PCID,SSE4.1,SSE4.2,x2APIC,MOVBE,POPCNT,TSCDLT,AESNI,XSAVE,OSXSAVE,AVX,F16C,RDRAND>
  AMD Features=0x2c100800<SYSCALL,NX,Page1GB,RDTSCP,LM>
  AMD Features2=0x121<LAHF,ABM,Prefetch>
  Structured Extended Features=0x239c27eb<FSGSBASE,TSCADJ,BMI1,AVX2,FDPEXC,SMEP,BMI2,ERMS,INVPCID,NFPUSG,RDSEED,ADX,SMAP,CLFLUSHOPT,CLWB,PROCTRACE,SHA>
  Structured Extended Features2=0x984007bc<UMIP,PKU,OSPKE,WAITPKG,GFNI,VAES,VPCLMULQDQ,RDPID,MOVDIRI,MOVDIR64B>
  Structured Extended Features3=0xfc18c410<FSRM,MD_CLEAR,IBT,IBPB,STIBP,L1DFL,ARCH_CAP,CORE_CAP,SSBD>
  XSAVE Features=0xf<XSAVEOPT,XSAVEC,XINUSE,XSAVES>
  IA32_ARCH_CAPS=0x88fd6b<RDCL_NO,IBRS_ALL,SKIP_L1DFL_VME,MDS_NO,TAA_NO>
  VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID,VID,PostIntr
  TSC: P-state invariant, performance statistics
real memory  = 17179869184 (16384 MB)
avail memory = 16287182848 (15532 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <ALASKA A M I >
FreeBSD/SMP: Multiprocessor System Detected: 16 CPUs
FreeBSD/SMP: Non-uniform topology
random: registering fast source Intel Secure Key RNG
random: fast provider: "Intel Secure Key RNG"
random: unblocking device.
ioapic0 <Version 2.0> irqs 0-119
Launching APs: 1 4 3 13 14 15 7 10 6 9 8 11 12 5 2
random: entropy device external interface
kbd0 at kbdmux0
efirtc0: <EFI Realtime Clock>
efirtc0: registered as a time-of-day clock, resolution 1.000000s
smbios0: <System Management BIOS> at iomem 0x45cbd000-0x45cbd01e
smbios0: Version: 3.6, BCD Revision: 3.6
aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS,SHA1,SHA256>
acpi0: <ALASKA A M I >
Firmware Error (ACPI): Could not resolve symbol [\134_SB.PC00.TXHC.RHUB.SS01], AE_NOT_FOUND (20221020/dswload2-315)
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20221020/psobject-372)
Firmware Error (ACPI): Could not resolve symbol [\134_SB.PC00.TXHC.RHUB.SS02], AE_NOT_FOUND (20221020/dswload2-315)
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20221020/psobject-372)
Firmware Error (ACPI): Could not resolve symbol [\134_SB.PC00.TXHC.RHUB.SS03], AE_NOT_FOUND (20221020/dswload2-315)
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20221020/psobject-372)
Firmware Error (ACPI): Could not resolve symbol [\134_SB.PC00.TXHC.RHUB.SS04], AE_NOT_FOUND (20221020/dswload2-315)
ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20221020/psobject-372)
acpi0: Power Button (fixed)
hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Timecounter "HPET" frequency 19200000 Hz quality 950
Event timer "HPET" frequency 19200000 Hz quality 350
Event timer "HPET1" frequency 19200000 Hz quality 340
Event timer "HPET2" frequency 19200000 Hz quality 340
Event timer "HPET3" frequency 19200000 Hz quality 340
Event timer "HPET4" frequency 19200000 Hz quality 340
Event timer "HPET5" frequency 19200000 Hz quality 340
Event timer "HPET6" frequency 19200000 Hz quality 340
Event timer "HPET7" frequency 19200000 Hz quality 340
atrtc1: <AT realtime clock> on acpi0
atrtc1: Warning: Couldn't map I/O.
atrtc1: registered as a time-of-day clock, resolution 1.000000s
Event timer "RTC" frequency 32768 Hz quality 0
attimer0: <AT timer> port 0x40-0x43,0x50-0x53 irq 0 on acpi0
[... 180 lines cut ...]
iicbus10: <Philips I2C bus (ACPI-hinted)> on ig4iic0
drmn0: [drm] Cannot find any crtc or sizes
iicbus10: <unknown card> at addr 0x2c
iic10: <I2C generic I/O> on iicbus10
ichsmb0: <Intel Alder Lake SMBus controller> port 0xefa0-0xefbf mem 0x6001118000-0x60011180ff irq 16 at device 31.4 on pci0
smbus0: <System Management Bus> on ichsmb0
iwlwifi0: <iwlwifi> mem 0x50600000-0x50603fff irq 17 at device 0.0 on pci3
iwlwifi0: successfully loaded firmware image 'iwlwifi-cc-a0-73.ucode'
iwlwifi0: api flags index 2 larger than supported by driver
iwlwifi0: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
iwlwifi0: loaded firmware version 73.35c0a2c6.0 cc-a0-73.ucode op_mode iwlmvm
iwlwifi0: Detected Intel(R) Wi-Fi 6 AX200 160MHz, REV=0x340
iwlwifi0: Detected RF HR B3, rfid=0x10a100
iwlwifi0: base HW address: 34:13:e8:b6:b4:3f
lo0: link state changed to UP
re1: link state changed to UP
ums0 on uhub0
ums0: <PixArt USB Optical Mouse, class 0/0, rev 2.00/1.00, addr 1> on usbus0
ums0: 3 buttons and [XYZ] coordinates ID=0
uhid0 on uhub0
uhid0: <E-Signal USB Gaming Keyboard, class 0/0, rev 2.00/1.00, addr 2> on usbus0
ubt0 on uhub0
ubt0: <vendor 0x8087 product 0x0029, class 224/1, rev 2.01/0.01, addr 3> on usbus0
Security policy loaded: MAC/ntpd (mac_ntpd)
```

dumbbell, 2023-08-24:
The `master` branch of this project is incompatible with Intel 12th gen. Only `update-to-v5.17` and `5.15-lts` are.

* How do you load the `i915kms` driver?
* Did you configure kernel dumps? Having a swap partition + `debug.debugger_on_panic=0`  in your `/etc/sysctl.conf` should be enough on FreeBSD -CURRENT. You can try if this works by running `sysctl debug.kdb.panic=1`, this will trigger a panic.

ykla, 2023-08-24:
> 

I followed your instructions, but it seems to have no effect. The system is still stuck in the same place. I used `kld_list="i915kms"` written to `/etc/rc.conf` for loading.

huanghwh, 2023-12-29:
Does master branch support Raptor Lake now ?

takawata, 2024-01-05:
I tried on ThinkPad X13 Gen 4 (Raptor Lake) and works fine. Thanks.

closed 2024-01-05 by evadot