freebsd/drm-kmod #221: Loading i915kms causes the screen to go blank on Jasper Lake CPU

mkohanim, 2022-12-27:
**Describe the bug**
Loading i915kms causes the screen to go blank on Jasper Lake CPU. 

**FreeBSD version**
FreeBSD 13.1-RELEASE-p5 FreeBSD 13.1-RELEASE-p5 releng/13.1-n250174-753d65a19a55 GENERIC amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
 [pciconf.bad.gz](https://github.com/freebsd/drm-kmod/files/10306500/pciconf.bad.gz)
</details>

**DRM KMOD version**
drm-510-kmod 5.10.113_8
drm-kmod 20220907_1

**To Reproduce**
sudo kldload i915kms

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
This is running on a NUC: Intel(R) Celeron(R) N5105 @ 2.00GHz, with 8GB RAM and 60GB eMMC.

evadot, 2022-12-27:
Please add more details.
Do you have access to the machine via ssh or is it frozen ?
If you have access please add hw.dri.__drm_debug=0x1FF to /boot/loader.conf, reboot, kldload i915kms and post dmesg content.
Also try with different screen if possible.

Thanks.

mkohanim, 2022-12-27:
Hi Emmanuel, thank you for getting back so quickly. I do have access; except for the monitor going blank, everything else works perfectly and therefore I do have access. I only have the same type of monitor here in the lab (HP 32"). But, this has been also tested with multiple monitors and resolutions in the factory. Here's the output of dmesg:
[drm] Initialized
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
drmn0: intel_wopcm_init_early: WOPCM: 2048K
drmn0: __confirm_options: enable_guc=0 (guc:no submission:no huc:no)
drmn0: i915_gem_init__contexts: fake context support initialized
drmn0: intel_pch_type: Found Jasper Lake PCH
drmn0: get_allowed_dc_mask: Allowed DC state mask 08
drmn0: intel_uncore_init_mmio: unclaimed mmio detected on uncore init, clearing
drmn0: init_engine_mask: vdbox enable: 0001, instances: 0001
drmn0: init_engine_mask: vebox enable: 0001, instances: 0001
drmn0: intel_device_info_runtime_init: rawclk rate: 19200 kHz
drmn0: intel_device_info_runtime_init: CS timestamp wraparound in 111669ms
drmn0: ggtt_probe_hw: GGTT size = 4096M
drmn0: ggtt_probe_hw: GMADR size = 256M
drmn0: ggtt_probe_hw: DSM size = 0M
[drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
[drm] Got stolen memory base 0x0, size 0x0
drmn0: wa_init_mcr: L3 fuse = c
drmn0: wa_init_mcr: MCR slice/subslice = 1000000
[drm] Initialized 6 GT workarounds on global
drmn0: intel_opregion_setup: graphic opregion physical addr: 0x727ed018
drmn0: intel_opregion_setup: ACPI OpRegion version 2.1.0
drmn0: intel_opregion_setup: Public ACPI methods supported
drmn0: intel_opregion_setup: SWSCI supported
drmn0: swsci_setup: SWSCI GBDA callbacks 00000cb3, SBCB callbacks 00300483
drmn0: intel_opregion_setup: ASLE supported
drmn0: intel_opregion_setup: ASLE extension supported
drmn0: intel_opregion_setup: Found valid VBT in ACPI OpRegion (RVDA)
drmn0: skl_get_dram_info: DRAM type: LPDDR4
drmn0: skl_dram_get_dimm_info: CH0 DIMM L size: 4 GB, width: X16, ranks: 2, 16Gb DIMMs: no
drmn0: skl_dram_get_dimm_info: CH0 DIMM S size: 4 GB, width: X32, ranks: 3, 16Gb DIMMs: no
drmn0: skl_dram_get_channel_info: CH0 ranks: 2, 16Gb DIMMs: no
drmn0: skl_dram_get_dimm_info: CH1 DIMM L size: 0 GB, width: X0, ranks: 0, 16Gb DIMMs: no
drmn0: skl_dram_get_dimm_info: CH1 DIMM S size: 0 GB, width: X0, ranks: 0, 16Gb DIMMs: no
drmn0: skl_dram_get_channel_info: CH1 not populated
drmn0: skl_dram_get_channels_info: Memory configuration is symmetric? no
drmn0: intel_dram_detect: DRAM bandwidth: 23466672 kBps, channels: 1
drmn0: intel_dram_detect: DRAM ranks: 2, 16Gb DIMMs: no
drmn0: icl_get_qgv_points: QGV 0: DCLK=128 tRP=20 tRDPRE=8 tRAS=45 tRCD=20 tRC=65
drmn0: icl_get_qgv_points: QGV 1: DCLK=144 tRP=22 tRDPRE=9 tRAS=51 tRCD=22 tRC=73
drmn0: icl_get_qgv_points: QGV 2: DCLK=176 tRP=27 tRDPRE=11 tRAS=62 tRCD=27 tRC=89
drmn0: icl_get_bw_info: BW0 / QGV 0: num_planes=2 deratedbw=4725
drmn0: icl_get_bw_info: BW0 / QGV 1: num_planes=2 deratedbw=5073
drmn0: icl_get_bw_info: BW0 / QGV 2: num_planes=2 deratedbw=5585
drmn0: icl_get_bw_info: BW1 / QGV 0: num_planes=1 deratedbw=5851
drmn0: icl_get_bw_info: BW1 / QGV 1: num_planes=1 deratedbw=6392
drmn0: icl_get_bw_info: BW1 / QGV 2: num_planes=1 deratedbw=7306
drmn0: init_vbt_defaults: Set default to SSC at 120000 kHz
drmn0: intel_bios_init: VBT signature "$VBT JASPERLAKE     ", BDB version 236
drmn0: parse_general_features: BDB_GENERAL_FEATURES int_tv_support 0 int_crt_support 0 lvds_use_ssc 0 lvds_ssc_freq 120000 display_clock_mode 1 fdi_rx_polarity_inverted 0
drmn0: parse_general_definitions: crt_ddc_bus_pin: 2
drmn0: parse_general_definitions: Expected child device config size for VBT version 236 not known; assuming 39
drmn0: parse_general_definitions: Found VBT child device with type 0x60d2
drmn0: parse_general_definitions: Found VBT child device with type 0x60d2
drmn0: parse_general_definitions: Found VBT child device with type 0x60d6
drmn0: intel_opregion_get_panel_type: Ignoring OpRegion panel type (0)
drmn0: parse_panel_options: Panel type: 2 (VBT)
drmn0: parse_panel_options: DRRS supported mode is seamless
drmn0: parse_lfp_backlight: VBT backlight PWM modulation frequency 200 Hz, active high, min brightness 6, level 255, controller 0
drmn0: parse_sdvo_device_mapping: Skipping SDVO device mapping
drmn0: parse_ddi_port: Port B VBT info: CRT:0 DVI:1 HDMI:1 DP:0 eDP:0 LSPCON:0 USB-Type-C:0 TBT:0 DSC:0
drmn0: parse_ddi_port: VBT HDMI level shift for port B: 0
drmn0: parse_ddi_port: VBT DP max link rate for port B: 810000
drmn0: parse_ddi_port: Port C VBT info: CRT:0 DVI:1 HDMI:1 DP:0 eDP:0 LSPCON:0 USB-Type-C:0 TBT:0 DSC:0
drmn0: parse_ddi_port: VBT HDMI level shift for port C: 0
drmn0: parse_ddi_port: VBT DP max link rate for port C: 810000
drmn0: parse_ddi_port: Port D VBT info: CRT:0 DVI:1 HDMI:1 DP:1 eDP:0 LSPCON:0 USB-Type-C:0 TBT:0 DSC:0
[... 1176 lines cut ...]
[drm] Added [PLANE:201:plane 7C] 0xfffff801b4351300 state to 0xfffff801e5845800
[drm] Set [NOFB] for [PLANE:201:plane 7C] state 0xfffff801b4351300
[drm] Added [PLANE:209:cursor C] 0xfffff801b41a3d80 state to 0xfffff801e5845800
[drm] Set [NOFB] for [PLANE:209:cursor C] state 0xfffff801b41a3d80
[drm] Added [CRTC:91:pipe A] 0xfffff801e525b000 state to 0xfffff801e5845800
[drm] Set [NOMODE] for [CRTC:91:pipe A] state 0xfffff801e525b000
[drm] Set [NOFB] for [PLANE:31:plane 1A] state 0xfffff801adb30600
[... cut ...]
[drm] Adding all current connectors for [CRTC:213:pipe C] to 0xfffff801e5845800
[drm] checking 0xfffff801e5845800
drmn0: intel_atomic_get_global_obj_state: Added new global object 0xfffffe00c89ed260 state 0xfffff801e5656500 to 0xfffff801e5845800
drmn0: intel_atomic_get_global_obj_state: Added new global object 0xfffffe00c89eecf8 state 0xfffff801dcb34c00 to 0xfffff801e5845800
[drm] committing 0xfffff801e5845800
[drm] Clearing atomic state 0xfffff801e5845800
[drm] Freeing atomic state 0xfffff801e5845800

With kind regards,
Michel

mkohanim, 2023-01-11:
UEFI settings were the issue. Fixed by updating the BIOS.

closed 2023-01-11 by mkohanim