freebsd/drm-kmod #223: graphics/drm-510-kmod panic with display via usb-c dock

MATPOCKuH, 2022-12-28:
**Describe the bug**
A kernel panic occurred after upgrade to drm-510-kmod-5.10.113_8

**FreeBSD version**
FreeBSD cat 13.1-STABLE FreeBSD 13.1-STABLE #6 f08254503: Tue Dec 27 03:25:09 MSK 2022     root@cat:/usr/obj/usr/src/amd64.amd64/sys/desktop amd64

**PCI Info**

<details>
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x1028 subdevice=0x0815
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
    subclass   = VGA
</details>

**DRM KMOD version**
drm-510-kmod 5.10.113_8
drm-kmod 20220907_1

**To Reproduce**
Two ways:
1. Attach a display via usb-c dock station and then kldload i915kms
2. kldload i915kms first and then attach a display to dock station or dock station to laptop

[core.txt](https://github.com/freebsd/drm-kmod/files/10313715/core.txt)
**Additional context**
This problem may be related to issue #162.

MATPOCKuH, 2023-02-03:
The problem still exists in drm-510-kmod-5.10.163

MATPOCKuH, 2023-02-15:
I can provide a vmcore file for crash dump analysis.
Also I can try to apply some patches.

evadot, 2024-04-17:
Is this still happening with 515 or 61 ?

MATPOCKuH, 2024-04-24:
I'm upgrades system to 14.0-stable and currently using drm-515-kmod-5.15.118_2 without this problem.