freebsd/drm-kmod #227: i915kms: artifacts with overlays on top of video playback in chromium+youtube

erikarn, 2023-01-28:
**Describe the bug**

There are artifacts with overlays on top of video playback - with youtube, when watching videos it's fine, but when you mouse over the playback window to manipulate the controls, you end up with horizontal banding.

This didn't happen with the -HEAD install I had here from a few months ago. (Yes, if requested I can actually go grab the ports tree and FreeBSD-HEAD version as I still have the installation media..)

**FreeBSD version**

it's -HEAD, ff06a8dbb677a04 (Aug 19, 2022 commit from jhibbits, but  it's from the tree @ Jan 22 2023)

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
(It's a Haswell-ULT - vendor 0x8086, device 0x0a26, subvendor 0x106b, subdevice 0x011b)

</details>

**DRM KMOD version**
drm-510-kmod 5.10.113_8

**To Reproduce**
* fire up xorg
* fire up chromium
* watch youtube :-)
* 
**Screenshots**
[image]
[image]

**Additional context**

This is on both a macbookair6,1 and macbookair6,2, which are similar era computers.

evadot, 2023-01-28:
Is this with libva-intel-driver installed ?
As said on irc this kind of regression is hard to track as it might be from drm or mesa or something else ...

evadot, 2023-03-29:
Ping for the question about libva

evadot, 2024-04-17:
Closing as no news.

closed 2024-04-17 by evadot