freebsd/drm-kmod #228: Uncaught kernel crash reboots the machine, unable to get debugger or stacktrace

ngortheone, 2023-02-04:
FreeBSD 14-Current most recent main branch build.
CPU Ryzen 9 5950x
GPU:  AMD Radeon Pro W5700 (navi10), `amdgpu` kernel module
WM: Sway, wayland, DPI scaling 2 (4k screen)

Running a slightly buggy vulkan application results in a crash and reboot. I was unable to get the stacktrace, 
even with the following sysctls set the the computer hangs for a second or two and then reboots.

```
debug.debugger_on_trap: 1
debug.debugger_on_recursive_panic: 1
debug.debugger_on_panic: 1
```

The way to reproduce:
In project https://github.com/Yatekii/imgui-wgpu-rs
run `cargo run --example hello-world` (rust is needed)

The crash happens with a chance of ~50/50, sometimes application starts normally, but I can trigger the crash if I run the app a few times in a row.

I suspect that this bug on Linux results in a crash on FreeBSD, but I have no hard evidence of that, it may be something else.
https://github.com/Yatekii/imgui-wgpu-rs/issues/77

Please advice on further debugging steps

P.S. I haven't tested this on X (and I don't have a good setup to do so), and I am not sure if it is relevant...

---------------------------------------------------------------
Details:

```
~ » uname -a
FreeBSD zen.hq 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n260536-81471049650f: Fri Feb  3 17:43:47 PST 2023     root@zen.hq:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
```

Packages
```
# pkg info | grep drm
drm-510-kmod-5.10.163          DRM drivers modules
libdrm-2.4.114,1               Userspace interface to kernel Direct Rendering Module services
```

```
 # kldstat
Id Refs Address                Size Name
 1  131 0xffffffff80200000  215b5d0 kernel
 2    1 0xffffffff8235c000     7c90 cryptodev.ko
 4    1 0xffffffff82369000   6c4948 zfs.ko
 5    1 0xffffffff82a2e000     a478 cuse.ko
 6    1 0xffffffff83400000   412990 amdgpu.ko
 7    2 0xffffffff83210000    72d78 drm.ko
 8    1 0xffffffff83283000     22a8 iic.ko
 9    3 0xffffffff83286000     30d8 linuxkpi_gplv2.ko
10    4 0xffffffff8328a000     6320 dmabuf.ko
11    1 0xffffffff83291000     c768 ttm.ko
12    1 0xffffffff8329e000    2f198 amdgpu_navi10_sos_bin.ko
13    1 0xffffffff832ce000    292d8 amdgpu_navi10_asd_bin.ko
14    1 0xffffffff832f8000     a3d8 amdgpu_navi10_ta_bin.ko
15    1 0xffffffff83303000    437a0 amdgpu_navi10_smc_bin.ko
16    1 0xffffffff83347000    425d8 amdgpu_navi10_pfp_bin.ko
17    1 0xffffffff8338a000    425d8 amdgpu_navi10_me_bin.ko
18    1 0xffffffff83813000    42558 amdgpu_navi10_ce_bin.ko
19    1 0xffffffff833cd000     cc88 amdgpu_navi10_rlc_bin.ko
20    1 0xffffffff83856000    43a08 amdgpu_navi10_mec_bin.ko
21    1 0xffffffff8389a000    43a08 amdgpu_navi10_mec2_bin.ko
22    1 0xffffffff833da000     a4d8 amdgpu_navi10_sdma_bin.ko
23    1 0xffffffff833e5000     a4d8 amdgpu_navi10_sdma1_bin.ko
24    1 0xffffffff838de000    63f18 amdgpu_navi10_vcn_bin.ko
25    1 0xffffffff833f0000     3370 acpi_wmi.ko
26    1 0xffffffff83942000    86190 if_iwlwifi.ko
[... 199 lines cut ...]
    device     = 'Navi 10 USB'
    class      = serial bus
none2@pci0:46:0:0:	class=0x130000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x148a subvendor=0x1022 subdevice=0x148a
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Function'
    class      = non-essential instrumentation
none3@pci0:47:0:0:	class=0x130000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1485 subvendor=0x1022 subdevice=0x1485
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Reserved SPP'
    class      = non-essential instrumentation
none4@pci0:47:0:1:	class=0x108000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1486 subvendor=0x1022 subdevice=0x1486
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Cryptographic Coprocessor PSPCPP'
    class      = encrypt/decrypt
xhci2@pci0:47:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x149c subvendor=0x1022 subdevice=0x148c
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse USB 3.0 Host Controller'
    class      = serial bus
    subclass   = USB
hdac1@pci0:47:0:4:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1487 subvendor=0x1462 subdevice=0x9c56
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse HD Audio Controller'
    class      = multimedia
    subclass   = HDA
```

ngortheone, 2023-02-04:
I can rule out my`sysctl.conf` - crash happens with empty `sysctl.conf`

ngortheone, 2023-02-04:
changing display scale factor 2->1 prevents the issue from happening (or makes it less deterministic), this must be caused by floating point operations

ngortheone, 2023-02-07:
It looks like update to 5.13 fixes this, so far I didn't have a single crash

evadot, 2023-03-29:
> It looks like update to 5.13 fixes this, so far I didn't have a single crash

Can you test if there is no regression between 5.15 and the drm-515-kmod port ?
Thanks.

ngortheone, 2023-08-17:
@evadot I've been running 5.15 for server months now and I didn't have a single crash.

evadot, 2024-04-17:
Closing as it's fixed

closed 2024-04-17 by evadot