freebsd/drm-kmod #238: i915 5.10.163 loosing external HDMI screen after a time, problem gets worse

mathiasp, 2023-02-23:
First, it's recognized as usual HDMI-1, but after a while, faster, if the screen goes into powersave, the external screen vanishes from xrandr output, but mostly stays sort of operational. 

I mainly use keyboard-driven UI, and it's mostly noticeable  using mouse-driven menus, which pop up on the internal screen, quite annoying.
Since I have to use a lot of mouse-driven apps currently it starts to really slow down my work.

xrandr output after the screen "vanished":

```
mathiasp:~% xrandr
Screen 0: minimum 8 x 8, current 8128 x 3132, maximum 32767 x 32767
eDP-1 connected primary 2560x1440+0+0 (normal left inverted right x axis y axis) 310mm x 170mm
   2560x1440     60.01*+  59.95  
   1920x1440     60.00  
   1856x1392     60.01  
   1792x1344     60.01  
   2048x1152     60.00    59.90    59.91  
   1920x1200     59.88    59.95  
   1920x1080     59.96    60.00    59.93  
   1600x1200     60.00  
   1680x1050     59.95    59.88  
   1400x1050     59.98  
   1600x900      60.00    59.95    59.82  
   1280x1024     60.02  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1368x768      60.00    59.88    59.85  
   1280x800      59.81    59.91  
   1280x720      59.86    60.00    59.74  
   1024x768      60.00  
   1024x576      60.00    59.90    59.82  
   960x540       60.00    59.63    59.82  
   800x600       60.32    56.25  
   864x486       60.00    59.92    59.57  
   640x480       59.94  
   720x405       59.51    60.00    58.99  
   640x360       59.84    59.32    60.00  
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-1 disconnected 5568x3132+2560+0 (normal left inverted right x axis y axis) 0mm x 0mm
HDMI-2 disconnected (normal left inverted right x axis y axis)
VIRTUAL1 disconnected (normal left inverted right x axis y axis)
  3840x2160 (0x7c8) 297.000MHz +HSync +VSync
        h: width  3840 start 4016 end 4104 total 4400 skew    0 clock  67.50KHz
        v: height 2160 start 2168 end 2178 total 2250           clock  30.00Hz

```

As seen, the virtual screen is also shown as disconnected but works and still includes the external monitor and I can use it, just every menu pops up on the internal screen.

** History **

Since this mainly shows in popups, which I really use rarely, I cannot say exactly, but it startet probably around March 2022. Yeah, I'm late, but it took me a while to even figure out what was going on, and then I got used to it ;)

**FreeBSD version**
FreeBSD danton 13.2-PRERELEASE FreeBSD 13.2-PRERELEASE #0 stable/13-n254200-8497b431f19e: Mon Jan 30 12:56:25 CET 2023     mathiasp@danton:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:      class=0x060000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x5914 subvendor=0x17aa subdevice=0x2259
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x17aa subdevice=0x2259
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
[... 688 lines cut ...]
[    45.308] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.309] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.310] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.310] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.311] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.311] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.311] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.313] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.313] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.313] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.315] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.315] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.315] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.332] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.334] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.568] (II) intel(0): resizing framebuffer to 5568x3132
[    45.615] (II) intel(0): switch to mode 3840x2160@30.0 on HDMI-1 using pipe 1, position (0, 0), rotation normal, reflection none
[    45.712] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.719] (II) intel(0): resizing framebuffer to 8128x3132
[    45.799] (II) intel(0): switch to mode 3840x2160@30.0 on HDMI-1 using pipe 1, position (2560, 0), rotation normal, reflection none
[    45.887] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    45.934] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    46.388] (--) intel(0): HDMI max TMDS frequency 300000KHz
[    46.389] (--) intel(0): HDMI max TMDS frequency 300000KHz
[  1004.147] (II) event14 - La-VIEW Technology SteelSeries, class 0/0, rev 2.00/0.00, addr 7: SYN_DROPPED event - some input events have been lost.

mathiasp, 2023-02-25:
xrandr in a freshly started system:

```
mathiasp:~% xrandr
Screen 0: minimum 8 x 8, current 8128 x 3132, maximum 32767 x 32767
eDP-1 connected primary 2560x1440+0+0 (normal left inverted right x axis y axis) 310mm x 170mm
   2560x1440     60.01*+  59.95  
   1920x1440     60.00  
   1856x1392     60.01  
   1792x1344     60.01  
   2048x1152     60.00    59.90    59.91  
   1920x1200     59.88    59.95  
   1920x1080     59.96    60.00    59.93  
   1600x1200     60.00  
   1680x1050     59.95    59.88  
   1400x1050     59.98  
   1600x900      60.00    59.95    59.82  
   1280x1024     60.02  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1368x768      60.00    59.88    59.85  
   1280x800      59.81    59.91  
   1280x720      59.86    60.00    59.74  
   1024x768      60.00  
   1024x576      60.00    59.90    59.82  
   960x540       60.00    59.63    59.82  
   800x600       60.32    56.25  
   864x486       60.00    59.92    59.57  
   640x480       59.94  
   720x405       59.51    60.00    58.99  
   640x360       59.84    59.32    60.00  
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-1 connected 5568x3132+2560+0 (normal left inverted right x axis y axis) 600mm x 340mm
   3840x2160     30.00*+  25.00    24.00    29.97    23.98  
   2560x1440     59.95  
   1920x1080     60.00    59.94    30.00    29.97  
   1600x900      60.00  
   1280x1024     60.02  
   1280x800      59.91  
   1280x720      60.00    59.94  
   1024x768      60.00  
   800x600       60.32  
   720x480       60.00    59.94  
   640x480       60.00    59.94  
HDMI-2 disconnected (normal left inverted right x axis y axis)
VIRTUAL1 disconnected (normal left inverted right x axis y axis)

```

evadot, 2023-03-29:
What is this VIRTUAL1 thing ? Are you using something like RDP or something ?

mathiasp, 2023-03-29:
> What is this VIRTUAL1 thing ? Are you using something like RDP 
> or something ?

No, that’s xorgs name for the combined screen (laptop & external 
screen).

rosorio, 2023-08-26:
I have a similar issue on my thinkpad X1 with i915 5.10.163, who has 2 external HDMI/DP ports.
HDMI-1 is properly recognized but not the HDMI-2 which is reported disconnected.
From time to time HDMI-2 comes back, but I found some black screen issues I can solve by switching to
in console mode [Ctrl+Alt+F1] then comes back to  xorg.

Screen 0: minimum 320 x 200, current 3840 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 308mm x 173mm
   1920x1080     60.00*+  59.97    59.96    59.93
   1680x1050     59.95    59.88
   1400x1050     59.98
   1600x900      59.99    59.94    59.95    59.82
   1280x1024     60.02
   1400x900      59.96    59.88
   1280x960      60.00
   1440x810      60.00    59.97
   1368x768      59.88    59.85
   1280x800      59.99    59.97    59.81    59.91
   1280x720      60.00    59.99    59.86    59.74
   1024x768      60.04    60.00
   960x720       60.00
   928x696       60.05
   896x672       60.01
   1024x576      59.95    59.96    59.90    59.82
   960x600       59.93    60.00
   960x540       59.96    59.99    59.63    59.82
   800x600       60.00    60.32    56.25
   840x525       60.01    59.88
   864x486       59.92    59.57
   700x525       59.98
   800x450       59.95    59.82
   640x512       60.02
   700x450       59.96    59.88
   640x480       60.00    59.94
   720x405       59.51    58.99
   684x384       59.88    59.85
   640x400       59.88    59.98
   640x360       59.86    59.83    59.84    59.32
   512x384       60.00
   512x288       60.00    59.92
   480x270       59.63    59.82
   400x300       60.32    56.34
   432x243       59.92    59.57
   320x240       60.05
   360x202       59.51    59.13
   320x180       59.84    59.32
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-1 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 531mm x 298mm
   1920x1080     60.00*+  50.00    59.94
   1280x1024     75.02    60.02
   1152x864      75.00
   1280x720      60.00    50.00    59.94
   1024x768     119.99    75.03    60.00
   800x600      119.97    75.00    60.32
   720x576       50.00
   720x480       60.00    59.94
   640x480       75.00    60.00    59.94
   720x400       70.08
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)